letscontrolit/ESPEasy #5177: [FR]

fly74, 2024-12-14:
The use of the blk#value in a rule only submit the value from the last device interval. Can I force the reading without changing the device interval?
```
on Rules#Timer=2 do
Publish Live/%sysname%/Temperature,[blk#Temperature]
Publish Live/%sysname%/Pressure,[blk#Pressure]
Publish Live/%sysname%/Humidity,[blk#Humidity]
timerSet,2,6
endon
```

TD-er, 2024-12-14:
You can call `taskrun` to trigger a read of that task.

TD-er, 2024-12-14:
I use it myself on a mobile unit which acts on the event sent by GPS to trigger reading of each sensor/task every X meters travelled.

fly74, 2024-12-14:
But will this fire the controller from the Device Task?

TD-er, 2024-12-14:
Yep, the `taskrun` command is like an override of the interval.
It will trigger a `PLUGIN_READ` which collects sensor data and sends it to the controller(s) of that task.

fly74, 2024-12-14:
That's bad. I want to publish from the Task with another interval as the rule publish interval.

TD-er, 2024-12-14:
I can explain how I do it on my mobile nodes, as that's probably what you also need to have...

- GPS task sends out event every <interval> or <meters travelled>, whichever comes first (is feature of the GPS plugin)
- Rules call `taskrun` on the sensor tasks, which send their data to the cache controller to store in to flash
- events of new sensor data are dealt with in the rules which copy this data to 'copies' of those but then using `dummy` plugin set as task. These have their own interval and controller setup which sends data to a controller if connected to the network (thus not while moving)

fly74, 2024-12-14:
The short way would be maybe. No controller in the device and 2 timer in the rule with taskrun.

TD-er, 2024-12-14:
Yep, only one possible drawback when sending lots of values to the same controller at the same time is that you may fill the controller queue before they are all sent. This may cause data loss or a system stall.

fly74, 2024-12-14:
> I can explain how I do it on my mobile nodes, as that's probably what you also need to have...
> 
> * GPS task sends out event every  or , whichever comes first (is feature of the GPS plugin)
> * Rules call `taskrun` on the sensor tasks, which send their data to the cache controller to store in to flash
> * events of new sensor data are dealt with in the rules which copy this data to 'copies' of those but then using `dummy` plugin set as task. These have their own interval and controller setup which sends data to a controller if connected to the network (thus not while moving)

As I read your's, that's was I used before with the ikea dust sensor. I'll try it.

fly74, 2024-12-14:
I had to use the 2 timer solution, because I need a different MQTT topic for the live data.

closed 2024-12-14 by fly74

tonhuisman, 2024-12-14:
@fly74 You do know we have a [support forum](https://www.letscontrolit.com/forum) for this type of questions right?
And if such question ends up being a feature request (or bug 🫨), _then_ the GH Issue system is the nest step to take.

TIA