letscontrolit/ESPEasy #5179: [BUG] Using substring for MQTT controller seems to fail

thalesmaoanz, 2024-12-15:
Hi there

I have multiple temperature sensors and I want to group in the same publish topic.

```
espeasy/t/1
espeasy/t/2
espeasy/t/3
espeasy/t/4
```

Since I can't use the same name for devices and cant use numbers for values, I created as follows.

[image]

My ideia is to manipulate publish with substrings, however it is not working for openHab MQTT controller.

```
%sysname%/{substring:1:4:%tskname%}/%valname%
```
It uses %tskname% as a string, which returns "tsk".

[image]

It this the expected behaviour?

tonhuisman, 2024-12-15:
> It this the expected behaviour?

The current way of variable replacement is to first replace the 'global' variables and functions, like `%sysname%` and `{substring:1:4:<str>}`, and in a second step, for each task & variable the `%tskname%` and `%valname%` are replaced. That explains why you get the `tsk` substring result.

I'm working on the MQTT controllers, so I can consider changing this behavior.

As an alternative you could: 
a) use a fixed value for the tskname part
b) change multiple tasks to have the same name (discouraged, but should work, may have minor side-effects)
c) use some mapping on the broker or openHAB side (not sure what's possible there, though)

thalesmaoanz, 2024-12-15:
I was using fixed tskname part as first suggested. But I was considering to add pressure sensor now, which lead to a new problem. I will try to understand the code. If I find something that I could help, I will open a PR.

Thanks.

closed 2024-12-15 by thalesmaoanz

tonhuisman, 2024-12-15:
> If I find something that I could help, I will open a PR.

I've added an improvement for this issue already.

thalesmaoanz, 2024-12-15:
Perfect! I just checked the commit and changed the line order in `StringHelper.cpp`.

This works!
```
%sysname%/{substring:0:2:%tskname%}/%valname%
```

This also
```
%sysname%/{substring:0:2:%tskname%}/{substring:0:10:%valname%}
```

But this one doesn't.
```
%sysname%/{substring:0:2:%tskname%}/{substring:1:10:%valname%}
```

TD-er, 2024-12-15:
Do you know what is the result of that last one?
And what is you use 
```
%sysname%/{substring:0:2:%tskname%}/{substring:1:10:%vname%}
```
?

That's what is searched for in `parseEventVariables`

thalesmaoanz, 2024-12-15:
I'm not sure if I follow your question. There is a typo? You replaced `%valname%` by `%vname%`.

If so, I do know the expected results, but it is treating %valname% as string. Same as before. 
In other words, it is working for `%tskname%`, but not for `%valname%`.

tonhuisman, 2024-12-16:
> In other words, it is working for `%tskname%`, but not for `%valname%`.

Yes, I was aware of that. The somewhat 'global' variables like `%tskname%` are processed only once, and `%valname%` is processed for each task value, starting with that preprocessed part, where `substring` has already been processed.

To change this will have more impact, and will cause minor headaches when merging, as I've done extensive work on the MQTT controllers, moving that part of the code to a new .cpp file.
I can change the order of processing, of course, and take the merge issues for granted 😅

thalesmaoanz, 2024-12-16:
It is clear now @tonhuisman , thank you.

thalesmaoanz, 2024-12-23:
Sorry about the late reply. I took some time to test it because I was having a lot of issue with my code. 
It does works, thanks! However, user must take care about its usage.

If I build the topic as I use to:

```
espeasy/tc/1
espeasy/tc/2
espeasy/tc/3
espeasy/tc/4
```

Espeasy self recognize as a command and state error of unknown command. I can either change subscribe topic or drop my will of building that way. Any suggestion?

PR #5178 (2024-12-21): Improvements and fixes Dec. 2024
Improvements and fixes:
- **`P021`** Move #defines to .h file to avoid compiler warnings (Arduino compiler doesn't (properly) support #ifdef in .ino files)
- **`P057`** Fix `mprint` command to skip the colon segment if no colon is provided there (from a [forum report](https://www.letscontrolit.com/forum/viewtopic.php?t=10470#p71903))
- **`Controllers`** Replace % variables before processing functions in controller variables like `Topic`
- **`Custom ESP8266`** Apply `Custom.h` settings on clean install (ESP8266 issue only, [see comment](https://github.com/letsco [... cut]