letscontrolit/ESPEasy #5180: [BUG] Serial Server stops working after saving rule

BasvanH, 2024-12-16:
**Describe the bug**
 I notice when saving my rules the Serial Server stops working. I need to go into the Serial Server device and Submit it again and then the AT commands work again. This is the sequence:

```
635206: WD : Uptime 11 ConnectFailures 0 FreeMem 229780 WiFiStatus: WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
636468: HTTP: serialsend,AT           <= AT command works
644196: Upload: START, filename: rules1.txt    <= update a rule
644308: Upload: WRITE, Bytes: 804
644316: Upload: END, Size: 804
644590: ESPEasy console using ESPEasySerial
654551: HTTP: serialsend,AT+CON30451152A5E8    <= AT command no longer works, i do a Submit in the Serial Server device
663148: EVENT: Clock#Time=Tue,11:01
663581: Ser2Net: WiFi server closed
663642: ESPEasy console using ESPEasySerial
663687: Ser2Net: TaskIndex=1 port=4 rxPin=1 txPin=3 BAUDRATE=115200 SERVER_PORT=1234 SERIAL_PROCESSING=1
663713: Ser2Net: WiFi server started at port 1234
663938: EVENT: TaskExit#hm=1,1
664074: EVENT: TaskInit#hm=1,1
664078: EVENT: Rules#Timer=1,11
665206: WD : Uptime 11 ConnectFailures 0 FreeMem 227856 WiFiStatus: WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
668161: HTTP: serialsend,AT+CON30451152A5E8     <= AT command works again
```

**Expected behavior**
When saving rules, the Serial Server should stay online.

**Used platform (please complete the following information):**
 - ESP type: ESP32
 - Build version: Self built from 20240822
 - Build set: Custom with only serial server and HTTP and HTTP Advanced plugins.

**Additional context**
I discvered this bug during my testing with [HM-10 module](https://github.com/letscontrolit/ESPEasy/issues/5158), I now created separate issue for this.

tonhuisman, 2024-12-16:
Ah, you have used Serial0 with GPIO 1 and 3 for your SerialServer configuration, but on the Tools/Advanced page you'll find that the same port is also used for Console logging. When saving files, this part is re-initialized, but not the SerialServer, so the Console now owns that serial port.

Best advice: Use a different serial port with different GPIO pins for your SerialServer.
Second best advice: Disable the Console serial (fallback) setting.

BasvanH, 2024-12-16:
Thanks for the explanation. I've changed the Serial Server to use HW Serial2 (GPIO 16 & 17), and now the Serial Server keeps on running.

closed 2024-12-16 by BasvanH

TD-er, 2024-12-16:
Ehhhh.... that might also not be the best possible pins to use as those will sometimes be polled for presence of PSRAM.
At least on the ESP32-classic.
See: https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#special-notes-on-gpio-6-11-16-17