letscontrolit/ESPEasy #5188: [FR] Add timeout for dummy plugin

chromoxdor, 2024-12-22:
I often use taskvalueset to write incoming data from other nodes to the dummy plugin.
But when these nodes stop sending, the last value is there indefinitely. There is no way to see if there is a problem.
A timeout option would be great for this.

- Select the timeout time and an error value. 
- An individual timer for a task value starts each time data is written. 
- When a timer expires, a given error value (as in P028) is written to that task value.

TD-er, 2024-12-23:
You can also use the command `taskvaluesetandrun` which will trigger an event.
Then in the rules you can start a timer and act on it when it triggers.

That's probably what you mean, right?

uzi18, 2024-12-23:
You can also use p2p to exchange task data between other espeasy hosts

chromoxdor, 2024-12-23:
> You can also use p2p to exchange task data between other espeasy hosts

This is what I did in the beginning, but as my number of sensors and devices has grown exponentially, not being able to choose the position of the plugin makes P2P plugin sharing very inconvenient.
I also don't think there's some sort of `no data input timeout', which is what this is all about.

> You can also use the command `taskvaluesetandrun` which will trigger an event.

This is what I do, so the FR is not that important, but it would be very convenient. 
I send the whole `taskvalueset' command via `sendto' so it would be a 'send and forget' situation with no additional changes on the receiving side.

TD-er, 2024-12-23:
> This is what I did in the beginning, but as my number of sensors and devices has grown exponentially, not being able to choose the position of the plugin makes P2P plugin sharing very inconvenient.

That's already being looked into...
Part of this investigation resulted in the addition of a field to select remote unit ID.
However I have to think this through a bit more as I don't want to make it too complex and also would like to have the option of sending to multiple units.
Also the current implementation of sending to UDP broadcast addresses isn't the best choice when considering scaling.
But using multicast is also rather tricky as not all network switches/accesspoints do handle it in the same way.

So the options I'm considering are hopping between "setup of p2p parameters" in the task or in the controller.

chromoxdor, 2024-12-23:
> Also the current implementation of sending to UDP broadcast addresses isn't the best choice when considering scaling.

what do you mean by that?

> However I have to think this through a bit more as I don't want to make it too complex and also would like to have the option of sending to multiple units.

In my perfect ESPEasy world I would have the possibility in every DevicePlugin to add devices to which I want to send my data.
And on the receiving side, either a dummy DevicePlugin opens automatically or I have to add it manually.
In the background this would mean that the plugin would send something like `SendTo,1,TaskValueSet,3,%taskvalue1%,%taskvalue2%,...` to a device.

And it would be even more perfect if the FR was included :P

[image]

TD-er, 2024-12-23:
> > Also the current implementation of sending to UDP broadcast addresses isn't the best choice when considering scaling.
> 
> what do you mean by that?

Now each node which has the UDP port set is receiving all broadcast traffic for that port.
This does already impact OTA updating if you reach some critical amount of ESPEasy nodes in your network.
So that doesn't scale well, even though it scales somewhat linear with the number of nodes.

Also for IPv6 another approach is needed.

And I do plan on making the p2p sharing as you describe, where you can define what to send where.
However since I also need to keep in mind the required change in approach for IPv6, I need to figure out a simple representation how to configure what task values to send to which end node.

The most elegant approach would be to use multicast as you can 'subscribe' to a multicast IP/port.
This is a bit like MQTT, where you subscribe to a topic and get notified on new messages.
However multicast implementations are quite tricky as some switches/access points only forward packets to subscribers for some time (e.g. 10 minutes on Ubiquity stuff) and work without issues on other (e.g. Mikrotik)
For the most reliable multicast handling, you should either run some querier on your network (typically on the L3-switch with the highest IP in your subnet), but somehow not everyone has 1000+ euro switches in their network. (I know, hard to imagine....)
So I need to find some other way to make p2p scalable and allow for sending data from nodeA, task X to nodeB, task Y (and nodeC...)
Only then I can extend the p2p data structure and add those settings to either the task config or the p2p controller setup.

chromoxdor, 2024-12-23:
Ok.. i see, there is no simple solution to it. Ipv6 was definitely not on my radar...
Thanks for the explanation!