letscontrolit/ESPEasy #5189: [FR] A map function for rules

chromoxdor, 2024-12-22:
It would be very convenient to have a map function in rules.

Something like:

`Map,<variableindex>,<inputvalue>,<fromLow>, <fromHigh>,<toLow>,<toHigh> `
where "map" writes the data to a variable lieke "let"

tonhuisman, 2024-12-22:
I would put this in the `[Math functions](https://espeasy.readthedocs.io/en/latest/Rules/Rules.html#math-functions)` category, and make it compatible with the Arduino `map(value, fromLow, fromHigh, toLow, toHigh)` function (actually, calling that function with the arguments provided 😅), returning the mapped value to apply it where needed, like in a `let` assignment or an `if` condition 😉

tonhuisman, 2024-12-22:
@chromoxdor I've added the `map()` function as a feature, can you please test with some scenarios? Build available from this [Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/12457417517)

NB: You will need quotes around the function when used in a `Let` command, as it currently uses commas as an argument separator.

tonhuisman, 2024-12-22:
Hmm, those quotes make me a bit unhappy, so we might need to use a different argument separator, probably a colon (`:`), like used with the string functions 🤔

tonhuisman, 2024-12-22:
@chromoxdor I've switched a few things:
- Using existing function `mapADCtoFloat()`, similar to Arduino `map()` but supporting `float` values
- Using colon as the separator, as a comma causes too many parsing issues, and (confusing) quotes have to be used

chromoxdor, 2024-12-23:
@tonhuisman  Wow! That was fast. 
I am in a bit of pre-Christmas stress at the moment, but I will happily test the function tonight.
Thank you very much!

chromoxdor, 2024-12-23:
First test where successful. 
A colon as separator is perfect and I really like it!
The closing bracket is not colored correctly, which I will look into ..... probably next year ;)

PR #5190 (2024-12-29): [Functions] Add 'map' function
Resolves #5189 

Features:
- Add Math function `map(value:fromLow:fromHigh:toLow:toHigh)` similar to the Arduino map() function. Using colon as an argument separator to avoid needing wrapping quotes when using this function.
- Updated documentation.
- ~[ColorCode] Reverted and changes~ moved to #5196 
- Moved `mapADCtoFloat()` from `Hardware` to `ESPEasy_math` Helper.

TODO:
- [x] More testing ([confirmed1](https://github.com/letscontrolit/ESPEasy/issues/5189#issuecomment-2560272271))
closed 2024-12-29 by TD-er