letscontrolit/ESPEasy #5191: [BUG]

fly74, 2024-12-23:
Maybe an issue. With the new release the rule event is not firing to set the dummy anymore. Had not an eye on it while the testcycle.

[image]

Rule:
```
On [WZDUST#PM2.5] do
  TaskValueSet 12,1,%eventvalue1%
Endon
```

TD-er, 2024-12-23:
Your rule isn't correct, you should loose the `[]`
```
On WZDUST#PM2.5 do
  TaskValueSet 12,1,%eventvalue1%
Endon
```

fly74, 2024-12-23:
Same result with

```
On WZDUST#PM2.5 do
  TaskValueSet 12,1,%eventvalue1%
Endon
```

TD-er, 2024-12-23:
OK, what task do you expect to fire an event?
Is it task nr 12?

`TaskValueSet` does not "run" the dummy task.
It just sets the value.
`TaskValueSetAndRun` will combine `TaskValueSet` and `TaskRun`.

`TaskRun` will send out an event and feed taskvalues to any connected controller of that task.

I don't see how this might have been a recent change, as this has been like this for years.

Only change (in the last 2 years) is that you cannot use `TaskValueSet` on something other than Dummy tasks anymore.
If you need that old functionality, you can enable it on the Tools->Advanced page to send to tasks other than 'dummy' tasks.

fly74, 2024-12-23:
It worked before the new testing release. Anyway so the Vindriktning has an interval of 8 seconds to get the values. The dummy task has an 5 minute interval to publish MQTT to the controller.

TD-er, 2024-12-23:
OK, can you show your rules, so I do understand what you try to do?
And also a list of the related tasks. (or screenshot of 'devices' page)

TD-er, 2024-12-23:
And what if you change your rules into this:

```
On WZDUST1#PM2.5 do
```

The dummy task doesn't have `WZDUST#PM2.5`
So if you need to act on the dummy task then you need to use this:
`WZDUST#WZDUST`

tonhuisman, 2024-12-23:
🤔 When/where does the Vindriktning send its value to the Dummy device?

fly74, 2024-12-23:
Damn the 1  was missing On WZDUST1#PM2.5, maybe my fault.

Offtopic: the Keepalive is used for Grafana monitoring:

[image]

closed 2024-12-23 by fly74

TD-er, 2024-12-23:
You can also use the looptimer instead of setting the timer again when handling it.
This will eliminate any 'drift' in moments of sending.

fly74, 2024-12-23:
> You can also use the looptimer instead of setting the timer again when handling it. This will eliminate any 'drift' in moments of sending.

Can you give an example ?

TD-er, 2024-12-23:
```
on WiFi#Connected do
  Publish SYSTEM/%sysname%/%cpu_model%/%mac%/%ip%/%sysbuild_filename%,0
  loopTimerSet,1,1
Endon

on Rules#Timer=1 do
  Publish Keepalive/%sysname%,%rssi%
endon
```

fly74, 2024-12-23:
So  that's a duplicate of code, not really a benefit.

fly74, 2024-12-23:
maybe looptimer needs parameter for firing when set.

TD-er, 2024-12-23:
It already has quite a lot of arguments, so maybe better to have an additional command analog to the `TaskValueSetAndRun`:

- LoopTimerSetNow
- LoopTimerSetNow_ms

Or some better name....

fly74, 2024-12-23:
[FR] add LoopTimerSetAndRun :)

fly74, 2024-12-23:
Another thing:
```
00:19:07.923 : (222020) Debug  : Command: Let
00:19:07.924 : (222020) Debug  : Let,1,12231924+6
00:19:07.936 : (221956) Debug  : Par1: 1 Par2: 12231924 Par3: 0 Par4: 0 Par5: 0
00:19:07.943 : (222084) Info   : ACT  : Let,2,12231924
00:19:07.945 : (222100) Debug  : Command: Let
00:19:07.945 : (222100) Debug  : Let,2,12231924
00:19:07.957 : (222036) Debug  : Par1: 2 Par2: 12231924 Par3: 0 Par4: 0 Par5: 0
00:19:07.962 : (222004) Info   : ACT  : TaskValueSet 16,1,12231924+6
00:19:07.964 : (221968) Debug  : Command: TaskValueSet
00:19:07.964 : (222020) Debug  : TaskValueSet 16,1,12231924+6
00:19:07.976 : (221956) Debug  : Par1: 16 Par2: 1 Par3: 12231924 Par4: 0 Par5: 0
00:19:07.980 : (221840) Info   : taskValueSet: TaskValueSet 16,1,12231924+6  taskindex: 15 varNr: 0 result: 12231930.000000 type: 40
00:19:07.997 : (221956) Debug  : timeStringToSeconds: "12231930" --> invalid
00:19:07.998 : (221956) Debug  : timeStringToSeconds: "12231924" --> invalid
00:19:08.011 : (221892) Debug  : conditionMatch: "12231930 > 12231924" --> "12231930" > "12231924" --> false (1223193                                                                  0 > 12231924)
00:19:08.023 : (221940) Debug  : conditionMatchExtended: false "12231930 > 12231924"
00:19:08.023 : (221968) Debug  : Lev.1: [if false]=false
```

Why this results in timeStringToSeconds?

```
TaskValueSet 16,1,[D1#Dummy]+6

If [D2#Dummy] > [D1#Dummy]
```

Task 16 is D2.

TD-er, 2024-12-23:
Not 100% sure why it is calling a time conversion function.
But when you like to calculate things without using the `let` function you should prefix the calculation with an `=`
```
Let,1,=12231924+6
```

I know it is a bit random whether calculations will be performed or not but prefixing it with an = character should make it clear to the rules parser there is some calculation to be done.

TD-er, 2024-12-23:
Hmm I think it is time for me to go to bed or something... you are using "let"...
Well then I don't know yet what the issue is here...

tonhuisman, 2024-12-23:
> Why this results in timeStringToSeconds?

The rules parser first tries to match a date/time value, that fails for numeric values, hence 'invalid'. Next step is to process as numeric.
Don't take Debug log texts too literal, that's _intended_ as developer-only.

Can you share that rule, so we can try to replay?

NB: in the TaskValueSet family of commands you can use the Task & value names, like `TaskValueSet,D2,Dummy,[D1#Dummy]+6`
It took me quite some effort to implement that 😅 with the intention to make your scripts less device dependent.

fly74, 2024-12-23:
It's quite simple there a 2 values of date:hour in format mmddhhmm:

[image]

I want to compare them to make clear the lower value is not farer then 6 minutes away, means the data is not to old.

So I add 6 minutes to the last value and compare to the system date in same format. This should be an integer calculation.

Rule from last try:

```
Let,1,=[MQTT_in3#Value1]+6
Let,2,=%sysmonth%%sysday_0%%syshour_0%%sysmin_0%

TaskValueSet 10,1,[var#1]

If var#2  < var#1
```

tonhuisman, 2024-12-23:
ESP8266 or ESP32? ...

fly74, 2024-12-23:
ESP32-D0WDQ6

fly74, 2024-12-23:
```
00:01:09.545 : (217240) Debug  : timeStringToSeconds: "12232206" --> invalid
00:01:09.546 : (217240) Debug  : timeStringToSeconds: "12232203" --> invalid
00:01:09.560 : (217176) Debug  : conditionMatch: "12232206  > 12232203" --> "12232206" > "12232203" --> false (12232206 > 12232203)
00:01:09.571 : (217224) Debug  : conditionMatchExtended: false "12232206  > 12232203"
00:01:09.572 : (217256) Debug  : Lev.1: [if false]=false
....
00:01:09.741 : (217288) Debug  : EVENT: Rules#Timer=1,1 Processing: 484 ms
```

fly74, 2024-12-23:
This variant

```
Let,1,=[MQTT_in3#Value1]+6
Let,2,=%sysmonth%%sysday_0%%syshour_0%%sysmin_0%

TaskValueSet 10,1,[var#1]

If [var#1]  > [var#2]
```

tonhuisman, 2024-12-23:
Please don't use so many spaces around comparison operators, `if` doesn't like that...

fly74, 2024-12-23:
Same result with "If [var#1]>[var#2]"
Is there a cast funtion to int or similar?

tonhuisman, 2024-12-23:
Another case of float-resolution-overflow. Drop the 12 at the front and it works as intended.

fly74, 2024-12-23:
Same without 12:

```
Debug  : Par1: 2 Par2: 12232213 Par3: 0 Par4: 0 Par5: 0
00:11:43.701 : (217596) Info   : ACT  : TaskValueSet 10,1,232219
00:11:43.702 : (217548) Debug  : Command: TaskValueSet
00:11:43.703 : (217596) Debug  : TaskValueSet 10,1,232219
00:11:43.716 : (217532) Debug  : Par1: 10 Par2: 1 Par3: 232219 Par4: 0 Par5: 0
00:11:43.720 : (217436) Info   : taskValueSet: TaskValueSet 10,1,232219  taskindex: 9 varNr: 0 result: 232219.000000 type: 1
00:11:43.724 : (217544) Debug  : timeStringToSeconds: "232219" --> invalid
00:11:43.736 : (217544) Debug  : timeStringToSeconds: "12232213" --> invalid
00:11:43.739 : (217496) Debug  : conditionMatch: "232219>12232213" --> "232219" > "12232213" --> false (232219 > 12232213)
00:11:43.751 : (217544) Debug  : conditionMatchExtended: false "232219>12232213"
00:11:43.762 : (217592) Debug  : Lev.1: [if false]=false
00:11:43.765 : (217560) Debug  : DEBUG: Formatted String='[7SEG3].7dtext,37'
00:11:43.768 : (217532) Debug  : DEBUG: Formatted String='[7SEG3].7dtext,37^~~39'
00:11:43.782 : (215652) Debug  : Lev.1: [else]=true
00:11:43.785 : (216700) Info   : ACT  : [7SEG3].7dtext,
00:11:43.787 : (216668) Debug  : Command: [7SEG3].7dtext
00:11:43.799 : (216188) Debug  : [7SEG3].7dtext,
00:11:43.801 : (216124) Debug  : Par1: 0 Par2: 0 Par3: 0 Par4: 0 Par5: 0
00:11:43.803 : (216064) Info   : 7DGT : Show Text=
00:11:43.821 : (216120) Debug  : timeStringToSeconds: "37" --> invalid
```

TD-er, 2024-12-23:
That's why we have introduced the seconds of current day system variable.

fly74, 2024-12-23:
```
36.681 : (217656) Debug  : Par1: 2 Par2: 232215 Par3: 0 Par4: 0 Par5: 0
00:13:36.695 : (217576) Info   : ACT  : TaskValueSet 10,1,232221
00:13:36.697 : (217520) Debug  : Command: TaskValueSet
00:13:36.698 : (217576) Debug  : TaskValueSet 10,1,232221
00:13:36.710 : (217512) Debug  : Par1: 10 Par2: 1 Par3: 232221 Par4: 0 Par5: 0
00:13:36.714 : (217396) Info   : taskValueSet: TaskValueSet 10,1,232221  taskindex: 9 varNr: 0 result: 232221.000000 type: 1
00:13:36.730 : (217560) Debug  : timeStringToSeconds: "232221" --> invalid
00:13:36.732 : (217560) Debug  : timeStringToSeconds: "232215" --> invalid
00:13:36.746 : (217504) Debug  : conditionMatch: "232221>232215" --> "232221" > "232215" --> true (232221 > 232215)
00:13:36.757 : (217560) Debug  : conditionMatchExtended: true "232221>232215"
00:13:36.758 : (217576) Debug  : Lev.1: [if true]=true
00:13:36.771 : (217540) Debug  : DEBUG: Formatted String='[7SEG3].7dtext,37'
00:13:36.775 : (217512) Debug  : DEBUG: Formatted S...
```

works!

fly74, 2024-12-23:
Maybe you have an better idea to make sure that values not older than 6 mintues.

TD-er, 2024-12-24:
> Maybe you have an better idea to make sure that values not older than 6 mintues.

Using `%unixday_sec%` is by far the easiest. Just make sure to use the modulo when comparing or else you may see issues around midnight.

Other one could be to set a timer when receiving a value.

fly74, 2024-12-24:
@tonhuisman @TD-er 

Thank U, I'll give it a try. Merry X-mas.

fly74, 2024-12-24:
A lot easier thx:

```
Let,1,=[MQTT_in3#Value1]+360
Let,2,=%unixday_sec%

If [var#1]>[var#2]
```

TD-er, 2024-12-24:
Can even done with 1 line less code :)

```
Let,1,=[MQTT_in3#Value1]+360

If [var#1]>%unixday_sec%
```

fly74, 2024-12-24:
Works too, but now back to the Christmas tree!