letscontrolit/ESPEasy #5200: [BUG] Can't save Rules when an admin password is defined

ericdlr, 2024-12-30:
When an admin password is defined and the system ask for it when saving the "Rules", all I get in the console is:
```
[18:21:06]03:07:21.876Info   : EVENT: Login#Failed
[18:21:06]03:07:21.889Info   : EVENT: Login#Failed
[18:21:12]03:07:28.162Info   : EVENT: Login#Failed
[18:21:12]03:07:28.169Info   : EVENT: Login#Failed
[18:21:12]03:07:28.176Info   : EVENT: Login#Failed
[18:21:12]03:07:28.188Info   : EVENT: Login#Failed
[18:21:16]03:07:32.362Info   : EVENT: Login#Failed
[18:21:16]03:07:32.369Info   : EVENT: Login#Failed
[18:21:16]03:07:32.377Info   : EVENT: Login#Failed
[18:21:16]03:07:32.385Info   : EVENT: Login#Failed
```
Even if I enter the correct login. And the browser is asking again for the password. Sometimes I get multiple login failed entry for a single entry of the password.
In order to save the rules, I have to disable the admin password, save the rules and redefined an admin password.

This happens with an ESP8266 on a NodeMCU v1 ESP-12F board with the following build:
- **ESP_Easy_mega_20241222_normal_ESP8266_4M1M_VCC Dec 22 2024**

And with an ESP-01F with the following build:
- **ESP_Easy_mega_20241222_normal_ESP8266_1M_VCC Dec 22 2024**

TD-er, 2024-12-30:
That's really strange as there has nothing changed regarding this code as far as I know.
At least not on ESP8266.

Have you tried another build to make sure this isn't caused by some update of the browser or some browser plugin which may prevent you from fetching the JavaScript code used to upload the rules file?

Do you have any 'special' characters in your password?

chromoxdor, 2024-12-30:
This must be an older problem. I had the same issues with an much older build until i disabled the password.

TD-er, 2024-12-31:
What browser is used? Safari?

chromoxdor, 2024-12-31:
Not sure what browsers i tested beck then. I will dig a bit deeper when i am home since it seems to be still a thing.

ericdlr, 2024-12-31:
I tried with chrome and it worked. It seems to be a Safari thing. I'm on Sequoia and Safari v18.3.
In Safari's Settings, under the Security tab, the "WebContent: Enable Javascript" box is checked.
So there may be something else blocking JavaScript.

ericdlr, 2024-12-31:
Changing 'chrome' to 'Chrome'

TD-er, 2024-12-31:
Maybe Safari does have some extra protections like "if not HTTPS and login required then no JavaScript" or something like that.

chromoxdor, 2025-01-02:
I start to investigate the issue. 
I think rules are not the only problem.
Uploading a file is also impossible. And this is not limited to Safari (also chrome and firefox). Can sombody confirm?