letscontrolit/ESPEasy #5209: [BUG] ESP_Easy_mega_20241222_normal_ESP32_4M316k Dec 22 2024: Fixed IP ignored after update

juersei, 2025-01-04:
I upgraded OTA using ESP_Easy_mega_20241222_normal_ESP32_4M316k Dec 22 2024, which I did the same way many times with earlier updates.

I use a fixed IP(*.211)  in the WLAN settings.

However, my ESP32 upgrade after reboot pulled a DHCP IP (*.45)  from my wireless router ignoring all the 4 IP settings under WIFI IP settings. I then have told the wireless router as well to assign the *.211 to this ESP32 (via MAC address) and now the ESP32 responds to the correct static IP address.

Regards

JUERSEI

tonhuisman, 2025-01-04:
Hmm, strange, _all_ my nodes have a fixed IP (both ESP8266 and ESP32) and they are all still on their fixed IP address.

What is the brand and model of your WiFi router?

If you remove the router configured IP via MAC (may need to reboot your router to apply the change), check your fixed WiFi config on the ESP (all 4 fields *must* be set!), then reboot the  ESP, is the configured IP address assigned to it?

juersei, 2025-01-04:
I am usings an AVM Fritz!box 7560 and as far as I remember this latest OTA update was the only one where I had to search in the router for the (new) IP address of the ESP32. 
Thouldn´t the ESP32 just follow the settings in the Config page under the WiFi IP settings and just grab a different IP address if the four input fields are empty?
Main were set to
192.168.178.211
192.168.168.1
255.255.255.0
192.168.178.1
and were working like that for ever.

After requesting in the pouter to just give the *.211 to the ESP32 everything was fine. It seems that the ESP32 prefers to follow DHCP from the router before it looks into its own the WiFi IP settings.

I will update another ESP32 on the same WLAN soon and will report / confirm if the same happens to that devive as well.

juersei, 2025-01-04:
2nd ESP device, same problem!

I confirm that the firmware ESP_Easy_mega_20241222_normal_ESP32_4M316k Dec 22 2024 ignores the IP WiFi settings and also ignores the assigned NTP IP address.

Again, my 2nd ESP32 device lost the assigned **.212 IP address and also got the **.45 IP address as the first ESP32. 

Since the 2nd ESP is connected to the same router via a WiFi repeater using a different SSID it is not displayed as a WiFi device in my Fritz!Box router, therefore I can´t easily assign the correct IP address **.212 via the WiFi MAC setting in the Fritz!Box.

I then decided to downgrade the ESP32 to the previous firmware [ESP_Easy_mega_20240414_normal_ESP32_4M316k Apr 14 2024]and all probems disappeared immediately.

Pls. have a look into the WiFi IP configuration settings! 

Otherwise you guys are really doing a great job, I really thank you for that and appreciate that!

Rgds.!

TD-er, 2025-01-04:
I will have a look at the code.
I suspect the DHCP-client may not be turned off completely (I have seen some issues being reported about this a few months ago in the Arduino issues)
I am rewriting the complete WiFi/Network code, so that will be one of the tests.

If you have some spare ESP and/or are willing to move this ESP to using LittleFS, could you also test if this has been fixed in the latest LittleFS build? This is using a newer ESP-IDF5.3 SDK. If it is working in this build, then I know it is not an ESPEasy bug.
Maybe best to use a 'new' ESP for it, so there is no configuration in the FritzBox that could lead to wrong conclusions.

N.B. Make sure your static IPs used are not in the range of the DHCP-pool, or you have at least assigned it as static in the router.
When a device is truly using static IP (thus no DHCP-client is used), the router will not see a DHCP request, so it may assign the same address to any other node.

juersei, 2025-01-04:
I have taken a spare ESP32 and flashed it to "ESP_Easy_mega_20241222_normal_ESP32_4M316k_LittleFS_ETH Dec 22 2024" since it seemed to be the firmware closest to my needs, however, using LittleFS".

After flashing (via cable) and setting the AP WLAN settings the device received **`.46 IP address via DHCP from my WLAN router.

Then I uploaded the config.dat of one of my other ESP32, set the static IP address settings on the config page to **.213 and rebooted the device.

However, also with this LittleFS supporting firmware my static IP WLAN stettings 

192.168.178.213
192.168.178.1
255.255.255.0
192.168.178.1

were ignored, I could only get access to the device on the **.46 IP address.

Then I flashed the device again back to "ESP_Easy_mega_20240414_normal_ESP32_4M316k Apr 14 2024" and repeated all the last steps again and the behavior was really as it should be:

After starting with the **.46 IP-address and uploading the config.dat (which included the **.213 static IP settings) a reboot changed the IP address of the device correctly to the desired **.213, everything was fine.

At least in my case the DEC 24 2024 firmware builds don´t react on my static IP settings as all the earlier builds but ignore them. 

I will need to stick for the time being to the earlier builds!

juersei, 2025-01-05:
More observations:

I downgraded my spare ESP32 to the earlier [ESP_Easy_mega_20240414_normal_ESP32_4M316k_LittleFS Apr 14 2024], uploaded the config.dat and also with that build my WiFi IP settings were ignored (after several reboots) and only the IP address obtained via DHCP from the router is applied.

However: 

I then found if I simply flash the device with the mentioned firmware and don´t upload any config.dat but simply set the WiFi IP settings (and only them at this time) manually the IP changes are accepted immediately after reboot.

As soon as I upload the config.dat things change and the IP settings are ignored. 

It seems that something is wrong with my config.dat, is that a possibility which you could think of as well?

TD-er, 2025-01-05:
Since there are no passwords stored in config.dat, can you zip the file and upload it here or email me the file?

juersei, 2025-01-05:
Here is my config.dat:

[config.zip](https://github.com/user-attachments/files/18310062/config.zip)

TD-er, 2025-01-06:
> Here is my config.dat:
> 
> [config.zip](https://github.com/user-attachments/files/18310062/config.zip)

Just tried your config and indeed I do also get an IP from the DHCP server.
We do have completely different subnets, so it is easy to see:

```

01.150 : (233828) Info   : WIFI : Set WiFi to OFF
01.365 : (257424) Info   : WIFI : Set WiFi to STA
01.488 : (233252) Info   : WIFI : Connecting Lurch3 B8:69:F4:9F:21:FA Ch:1 (0dBm) open '01' ch: 1..14 attempt #0
01.498 : (233304) Info   : IP   : Static IP : 192.168.178.213 GW: 192.168.178.1 SN: 255.255.255.0 DNS: 192.168.178.1 / 0.0.0.0
01.605 : (232356) Info   : WIFI : Arduino wifi status: WL_CONNECTED 3 ESPeasy internal wifi status: DISCONNECTED
01.611 : (227228) Info   : Webserver: start
01.626 : (227180) Info   : EVENT: System#Boot
01.628 : (226884) Info   : WIFI : Connected! AP: Lurch3 (B8:69:F4:9F:21:FA) Ch: 1 Duration: 54 ms
01.630 : (226548) Info   : WIFI : Static IP: 192.168.11.16 (Pool-Controller-3-3) GW: 192.168.10.1 SN: 255.255.254.0 DNS: 192.168.10.1 / 0.0.0.0   duration: 85 ms
01.661 : (224164) Info   : UDP : Start listening on port 8266
01.662 : (224116) Info   : firstLoopConnectionsEstablished
```

juersei, 2025-01-07:
TD-er,

thanks for checking and confirming!

In summary: After also trying the littleFS build from April 2024 (showing the same bug) which we know now:

1. This bug appears only if a config.dat is used to specify the fixed IP settings, it does not occur if the IP settings are instead set just manually in the config menu.
2. This bug is not necessarily new to the Dec. 2024 build(s) but also appears already in the April 2024 little FS build(s). However, I had not recognized it those days in the SPIFFS build(s), which I exclusively used so far.

TD-er, 2025-01-07:
@juersei Can you try from this test build, to see if the static-ip stuff has been firxed? https://github.com/letscontrolit/ESPEasy/actions/runs/12636860167

N.B. only install this on a node that is reachable for re-flashing using serial an don't have important data on it as the PR is absolutely not finished.
Still a lot to do as it is a complete overhaul of the WiFi code which has become way too complex in the last 7 years working around bugs and undocumented behavior of WiFi.

juersei, 2025-01-07:
@TD-er

I tried the [ESP_Easy_mega_20250106_normal_ESP32_4M316k Jan 6 2025] via serial (factory file), the ESPEASY AP started, but I could not get access to the webserver on 192.168.4.1 and therefore could not change anything.

I then flashed back to an older April 2024 firmware (serial flashing) and made an OTA update on the above mentioned firmware (non factory file). Subsequently I uploaded my config.dat file.

Result: Only the DHCP IP address is applied, the manually set IP address is ignored, even after reboots.

TD-er, 2025-01-08:
Thanks for testing.
Back to the drawing board.

juersei, 2025-01-08:
> We may need to analyze what 'broken' setting in that config.dat is causing this behavior, so we can address the root-cause.

Hi Tony,

if there is anything I can do for you (with my very limited coding knowledge) I will be happy to do so!

I really admire and appreciate your incredible work to keep the code for all of us up to date!

I modified the config.dat already slightly to now comply with my other (1st) home WLAN 192.168.0.* instead of 192.168.178.* (in 2nd home previously). However, the static IP failure did not disappear.

I took the spare ESP32 device back (to 1st) home in order to be able to support failure analysis. 

 

Both homes have 2 different AVM Fritz!box devices connected with a permanent VPN. For that reason I use the “Allow all” option in the Client IP filtering field.

 

I now tried already several issues, all on my spare ESP32 with the Dec. 2024 firmware:

 

Under the Config menu:

*	Changing the settings under the Config menu for Client IP filtering: no success! 
*	Entering again all figures under the WiFi IP settings to exclude wrong hidden characters: no success

 

Under the Tools / advances menu (all with no success):

*	Change NTP IP
*	Use last connected AP from RTC
*	Extra Wait WiFi Conect
*	Enable SDK WiFi Auto Reconnect

The only way to receive a different IP address (with the loaded config.dat) is via telling the DHCP server of the WiFi router to allocate a specific address.

It seems that the firmware follows this flow scheme:

1. Check for config.dat
1a. if config.dat loaded just ignore all IP settings from the config page / in the config.dat and request IP address from WiFi DHCP server of WLAN router

1b. if config.dat is missing look into IP settings of config page, otherwise ask DHCP server

Best regards 

Juergen