letscontrolit/ESPEasy #5211: Long latency with Safari on MacOS/iPadOS when selecting the configuration pages.

TheoW11, 2025-01-04:
Hi,

I am running the latest version mega-20241222 Christmas Edition, but also previous versions were having the same long (5.5 s) latency problem before the webpage is updated on a ESP-01 and Wemos mini D1.

I don't have any latency issues on Windows PC with MS-Edge.

### Problems on:
MacBook Pro
- MacOS 14.7.1 
 - Safari 18.1
   
iPAD
- iPadOS 17.7.2
- Safari, when switching off 'Ask for Desktop version', speed improves

This is shown in the Webinfo window of the Safari browser.
[image]

I have found a similar issue reported previously --> [https://github.com/letscontrolit/ESPEasy/issues/4659]

### Summarize of the problem/feature request

Long latency of about 5.5 seconds before one of the configuration web page is shown.

### Expected behavior

Same behaviour as on Windows PC with MS-Edge

### Steps to reproduce

1. Flash a ESP8266 device with the latest version.
2. Take an iPAD/MacBook Pro and start Safari and open one of the configuration/settings pages. Page to load takes about 5-6 seconds.

Kind regards,
Theo.

tonhuisman, 2025-01-04:
We've seen (and fixed) another new issue with Safari, recently. Possibly caused by the removal of backward compatibility features from recent Safari versions, as all seems to work as usual when browsing the ESP from either FireFox or Chrome (using a Mac or iPad).
Can you test using FireFox and/or Chrome (or some other Chromium based browser like Brave or Vivaldi)?

TheoW11, 2025-01-06:
Hi Ton,

I have tested on these devices / browsers:
- iPAD
    - Chrome same long latency
    - Vivaldi same long latency
    - Firefox same long latency
- Mac
    - Chrome working oké
    - Firefox working oké
    - MacOS 15.2 Safari same long latency

On browsers with long latency, if you click again within the latency period or double click, the web page is updated immediately.

Kind regards,
Theo.

GravityRZ, 2025-03-28:
I think I have the same problem on safari
however it is not a latency problem.
it looks like when clicking a menu item(page) it tries to jump to that page, then jumps back to the previous page and after that jumps back to the page selected

what happens is
1 click on config
address changes from 192.168.1.152 to 192.168.152/config
then after 1 second it jumps back to the previous page (192.168.1.152) the main page
then after 3 seconds it jumps to the page you clicked(192.168.152/config)

so it jumps back and forth casing delays

GravityRZ, 2025-03-29:
[image]

this my situation. seems to be the same problem

GravityRZ, 2025-03-29:
did some more testing but I am no html expert

noticed that the pages are all loaded HTTP and the files HTTPS(external)

could it be some kind of security issue that safari is to strict on things

[image]

tonhuisman, 2025-03-29:
> could it be some kind of security issue that safari is to strict on things

Well, that wouldn't explain why it's not slow when double-clicking to activate, I'd expect the delay then to be triggered twice on a double-click 🤔

GravityRZ, 2025-03-29:
@tonhuisman agreed but it must be something stupid simple.

in Chrome(which is working) I see a warning about an html attribute autocomplete which is missing

InSafari(where the problem is) I see 2 errors
these errors are always there regardless of the page loading fast or not but maybe it has something to do with it

[image]

TD-er, 2025-03-29:
Could this hammer npm code be from some plugin/extension of Safari?
I don't remember we needed it.

chromoxdor, 2025-03-29:
The error you see is most likely from a browser extension. hammer.min.js doesn't belong to ESPEasy

chromoxdor, 2025-03-29:
If you haven't already you could also try to disable all browser extensions and test again....

chromoxdor, 2025-03-29:
What is the signal strength of the device? (RSSI value found in /info under WiFi)

GravityRZ, 2025-03-29:
disabled all extensions, cleared cache, closed safari, start safari, same result.

the wifi strength is really good -44 dBm
it is 1.5 meters from the router, that is why I use this espgui  for testing

GravityRZ, 2025-03-30:
the hammer.min error shows up even if the page load fast so that probably is not the problem.

once in a while however I see this error

[image]

TD-er, 2025-03-30:
Hmm that should not happen, as we allow for explicitly loosely set CORS.

Or could it be related to some login checks?
Do you have a password set to access the ESP?
If so, can you disable it to see if it makes a difference?

GravityRZ, 2025-03-30:
no password set, I just type in the ip address and it shows the gui
gonna drop some screen dumps. somehow I could step through the process while it hangs

[image]

[image]

[image]

[image]

GravityRZ, 2025-03-30:
[image]

GravityRZ, 2025-03-30:
maybe you can see where down he line things go wrong

chromoxdor, 2025-03-30:
Looks like jQuery could use an update...

TD-er, 2025-03-30:
https://code.jquery.com/jquery-3.6.4.min.js 
too old?

chromoxdor, 2025-03-31:
> No idea if this breaks stuff though

Me neither. Was more of a wild guess as i am traveling with just my phone and only one ESPEasy device 🙂. So i am unable to help and test until mid April.

GravityRZ, 2025-03-31:
will it help if i download that new version and put it on the local file system of the ESP

i can try that if you like

TD-er, 2025-03-31:
JQuery is only linked via CDN and not searched for on the file system.
So we need to make a different build with the other version.
What build do you need?

GravityRZ, 2025-03-31:
ESP_Easy_mega_xxxx_normal_ESP8266_4M1M

TD-er, 2025-03-31:
Just changed the JQuery URL: https://www.dropbox.com/scl/fi/zd29u7ml43eg1tmivnxpf/ESP_Easy_mega_20250331_normal_ESP8266_4M1M.bin?rlkey=rx3d2px4m5xibuspeisu9ipao&dl=0

GravityRZ, 2025-03-31:
tested. unfortunately it did not solve the problem.

what code could be causing that the esp jumps back to the previous address?

[IMG_2155.zip](https://github.com/user-attachments/files/19534482/IMG_2155.zip)

chromoxdor, 2025-03-31:
> what code could be causing that the esp jumps back to the previous address?

That is really hard to tell without having the hardware and software to test this. (I do but see my last post)

For easyfetch i had some similar behavior with chrome. But there it had to do with the ESPEasy UI embedded in an iFrame. It was caused by chromes preloading feature. I did a bug report back then…

GravityRZ, 2025-03-31:
it is like you need to double click

so it does not matter which tab you click

the tab you click second is loaded instanly

so if you are on  main and click config there is a delay 
if you are on main en double click config it instantly loads config
if you are on main, click on hardware and then on config, config loads instantly.

GravityRZ, 2025-05-03:
is somebody investigating this?

chromoxdor, 2025-05-03:
Thanks for the reminder. It slipped my mind because I was traveling a lot and still not home. I can check it tomorrow evening, if I haven't forgotten it by then.

GravityRZ, 2025-05-04:
thanks for looking into this.
there are lot's of previous problems with the combination safari and the need for double clicks
somehow i think this is related.

chromoxdor, 2025-05-04:
I can confirm the issue. About 5sec load time when changing to any tab. Double click calls the page immediately.
The reason i didn't experience this until today is easyfech.
When using easyfetch, you can open the ESPeasy web UI in an iFrame by clicking the house symbol. 
In the iFrame there is no issue when clicking through the tabs.
I don't know why this is happening and i couldn't find any helpful info in the internet....

TD-er, 2025-05-04:
Just an idea to search for...
Could it be that Safari extensively tries to load Favicon or maybe tries some HTTPS entry which times out?
Like could there be some pending request stalling any other request?
If using EasyFetch, then that page is already running in the browser, so any new request may be dealt with differently as it isn't a full page load, but just some background request issued from the SPA.

chromoxdor, 2025-05-05:
> Could it be that Safari extensively tries to load Favicon or maybe tries some HTTPS entry which times out?
Like could there be some pending request stalling any other request?

No, this is something you would see in the dev console. (At least this is what i think)

> If using EasyFetch, then that page is already running in the browser, so any new request may be dealt with differently as it isn't a full page load, but just some background request issued from the SPA.

It's really weird! 
In the iFrame in Safari it simply calls the link twice. I don't understand why this is happening.

in the iFrame:
[image]
vs:
[image]

<br/>
<ins>Here are the timing statistics (just normal ESPEasy without easyfetch): </ins>

(sorry, my safari speaks german but I think you get the idea)
single click on link:
[image]

 vs double click (and a double click here doesn't mean cliking on the link twice. it it registered as one action ):
[image]

GravityRZ, 2025-05-05:
and don't forget it seems to jump back to the referrer page but still does not load that one either.

chromoxdor, 2025-05-05:
> and don't forget it seems to jump back to the referrer page but still does not load that one either.

this didn't happen to me.. at least not macOS

GravityRZ, 2025-05-05:
i posted a small movie in a zip a couple of posts up(which i posted on pc and can not open on imac)
i can post a mall movie again if you like

TD-er, 2025-05-05:
Hmm I think Safari may keep a connection open, unless it actively cancels a request like in double click or clicking on another link while the first request is still busy.

The ESP doesn't like to process the new request while the other one is still active/pending.
Not sure if you can see anything of this in the timing stats, as I do intend to close the connection when the page rendering is done.

chromoxdor, 2025-05-05:
> Hmm I think Safari may keep a connection open, unless it actively cancels a request like in double click or clicking on another link while the first request is still busy.

That's probably what is happening

I also did a last test for today.. I changed ESPeasys webserver timeouts to 10sec....guess what happens:

[image]

I think the link only opens after the server signals a timeout. 
My knowledge is too limited, though, to make heads or tails of it.

TD-er, 2025-05-05:
Problem is this timeout is also needed when doing an OTA flash.
If I lower this value, OTA updates will most certainly fail.

So I guess there must be some flag somewhere where we can tell Safari not to keep the connection open. Though I thought we already did this.

TD-er, 2025-05-05:
Hmm apparently there is something considered to be 'keep alive' in Chrome too

[image]

chromoxdor, 2025-05-05:
> Problem is this timeout is also needed when doing an OTA flash.
> If I lower this value, OTA updates will most certainly fail.

No, the timeout is fine and should not be changed just to solve this problem.

TD-er, 2025-05-05:
Oh never mind... I need coffee... The "keep-alive" is in the request header, not in the response.
Was already looking through the code where it might have been set.

So I think this is for sure a Safari bug

TD-er, 2025-05-05:
I did have some chat with someone from Espressif.
He asks if you're using CDN to serve the static files, or are these served by the ESP?

So there might actually be a pending call which isn't answered by the ESP for some reason.

chromoxdor, 2025-05-05:
Found it i guess... safari seems to be picky with the base URL
[#5312](https://github.com/letscontrolit/ESPEasy/pull/5312)

chromoxdor, 2025-05-05:
hmmm.. still not quite the fix. It seems to work with the IP but not with the Hostname 🤷‍♂️

chromoxdor, 2025-05-05:
ok... 
other than that i found at least one entry in the whole internet with the same issue back in 2022
https://www.facebook.com/groups/esp8266microcontrollers/posts/1459047317878081/

TD-er, 2025-05-05:
I also found some similar issue reported around the same time in the me-no-dev/AsyncWebServer (I think it was that repo)
That's also the person I am chatting with about this.
However I will not convert it all to using an async webserver (at least for now) as that would also mean a complete overhaul of how web pages are being served.
So first I will have a look at how static stuff can be served inline, or forced to be served from a CDN.

TD-er, 2025-05-05:
@chromoxdor 
Can you replace the `serve_favicon()` function with this and test?

```c++
void serve_favicon() {
  #ifdef WEBSERVER_FAVICON_CDN
  addHtml(F("<link"));
  addHtmlAttribute(F("rel"), F("icon"));
  addHtmlAttribute(F("type"), F("image/x-icon"));
  addHtmlAttribute(F("href"), generate_external_URL(F("favicon.ico"), true));
  addHtml('/', '>');
  #else
  addHtml(F("<link"));
  addHtmlAttribute(F("rel"), F("icon"));
  addHtmlAttribute(F("type"), F("image/svg+xml"));
  addHtmlAttribute(F("href"), F("data:image/svg+xml,%3Csvg xmlns='http://www.w3.org/2000/svg' width='113.392' height='113.392' viewBox='0 0 30.002 30.002'%3E%3Cpath fill='%23f9f9f9' d='M0 24.728V5.272A5.272 5.272 0 0 1 5.272 0h19.456c2.91 0 5.274 2.36 5.274 5.272v19.456a5.276 5.276 0 0 1-5.274 5.274H5.272A5.274 5.274 0 0 1 0 24.728zM30.002 0'/%3E%3Cpath fill='%2300f' stroke='%2300f' stroke-linecap='round' stroke-width='2.636' d='m3.75 11.25 7.5-7.5M3.75 18.75l15-15M3.75 26.25l22.5-22.5M11.25 26.25l15-15'/%3E%3Cpath fill='%2300f' d='M27.374 24.377a2.999 2.999 0 1 0-5.999 0 2.999 2.999 0 0 0 3.002 2.997 2.997 2.997 0 0 0 2.997-2.997zm-2.997 0'/%3E%3C/svg%3E"));
  addHtml('/', '>');

/*
  <link rel="icon" type="image/svg+xml"
      href="data:image/svg+xml,%3Csvg xmlns='http://www.w3.org/2000/svg' width='113.392' height='113.392' viewBox='0 0 30.002 30.002'%3E%3Cpath fill='%23f9f9f9' d='M0 24.728V5.272A5.272 5.272 0 0 1 5.272 0h19.456c2.91 0 5.274 2.36 5.274 5.272v19.456a5.276 5.276 0 0 1-5.274 5.274H5.272A5.274 5.274 0 0 1 0 24.728zM30.002 0'/%3E%3Cpath fill='%2300f' stroke='%2300f' stroke-linecap='round' stroke-width='2.636' d='m3.75 11.25 7.5-7.5M3.75 18.75l15-15M3.75 26.25l22.5-22.5M11.25 26.25l15-15'/%3E%3Cpath fill='%2300f' d='M27.374 24.377a2.999 2.999 0 1 0-5.999 0 2.999 2.999 0 0 0 3.002 2.997 2.997 2.997 0 0 0 2.997-2.997zm-2.997 0'/%3E%3C/svg%3E" />
*/
  #endif
}
```

chromoxdor, 2025-05-05:
favicon is back, issue with safari persists...

TD-er, 2025-05-05:
Can you see where the browser is waiting for?
I can also try to add a setting to force all static files to be loaded from a CDN.

To test whether this could be useful, you can define `WEBSERVER_USE_CDN_JS_CSS` in the build settings.
For example in the platformio_xxx_envs.ini file where your build is defined, you can add
`-D WEBSERVER_USE_CDN_JS_CSS`
to the `build_flags`

TD-er, 2025-05-05:
In Chrome on Windows, the load times (with the inline SVG favicon) are as snappy as they should be:

[image]

ESP32-S3 with 16M flash/8M PSRAM using WiFi.

GravityRZ, 2025-05-05:
can we rule out that the referrer policy has nothing to do with this?
in my case it wastes time by jumping back to the previous page(which is the referrer page)

when you instantly load a page from a favorite it has no delay and that page shows no referrer.
i also heared that these policies changed on both chrome and safari.

incase this has nothing to do with this, please ignore

chromoxdor, 2025-05-05:
> Can you see where the browser is waiting for?

ESP32-C3 4MB Normal Spiffs

loading times when entering the url directly:

[image]

loading times when clicking the link:

[image]

chromoxdor, 2025-05-05:
> favicon is back, issue with safari persists...

I was checking the favicon on chrome where it is indeed working again. 
But in safari the favicon is not been displayed. I think in general safari has some issues with inline SVGs...

TD-er, 2025-05-05:
Have you also tried to make a build with `-DWEBSERVER_USE_CDN_JS_CSS` ?

Most simple way is to add it to the `build_flags` in `[core_esp32_IDF5_3_2__3_1_0_LittleFS]` in `platformio_core_defs.ini`

This way the ico is still served from the ESP, but the rest is all served from a CDN

[image]

TD-er, 2025-05-05:
Sub-100 msec is exceptionally fast and even faster than what 'should be' as the beacon interval time for WiFi typically is 102.4 msec.
So you would expect a page load via WiFi to be X msec + 50 msec on average.
Timing stats on the `handle webpage` are only about the actual render time.
On my S3, this is 6.3 msec minimum, 42.7 average and 250 msec max. (over 90 page loads)

TD-er, 2025-05-05:
> > This way the ico is still served from the ESP, but the rest is all served from a CDN
> 
> I lost track, why are we doing this again. The favicon working or not seem to have nothing to do with the "href-issue" since it looks to me that safari is simply ignoring the favicon if it doesn't like it...

It was more a remark about what still needs to be changed when serving stuff all from CDN.
It was something I just noticed less than a minute before I typed it :)
So intended to make sure you realize where the data was coming from and a completely failed attempt to not confuse you :)

TD-er, 2025-05-05:
Just a question about the inline SVG icon...
Does it cause issues on Safari, like some floating icon somewhere?
Or is it just ignored?

If the latter, then I don't need to add user agent checks.

chromoxdor, 2025-05-05:
> Just a question about the inline SVG icon... Does it cause issues on Safari, like some floating icon somewhere? Or is it just ignored?
> 
> If the latter, then I don't need to add user agent checks.

It seems to be ignored...

TD-er, 2025-05-05:
@chromoxdor I assume you have your CSS on the file system of your ESP, right?
Do you have a `.css.gz` version or `.css` ?

Point is that if I want to serve the file inline, I can't have a .gz version.

I also have to make sure this isn't becoming too complex.

So my idea:
- Split off ESP8266 implementation into a separate .cpp file as that one doesn't seem to have the same issues with parallel requests as much as the ESP32 has.
- Check if file exists on file system. If it does and isn't .gz, then serve inline. If it exists as `.css.gz`, it must be linked to be loaded after the page has loaded.
- Detect whether file has to be served from CDN (either setting or compile-time define)
- If file is served from the ESP, then we must also decide whether there must be a check for whether it has been changed or 304 otherwise.

Or if anyone knows how to really serve a file with a TTL which is honored by the browser.
Have already spent a lot (!!) of time on this and have not been successful.

[17 more replies]