letscontrolit/ESPEasy #5212: new [BUG] with Christmas Edition 2024 ---> RTTTL doesn't full play

dyter69, 2025-01-04:
Command rtttl:

- Work with **ESP_Easy_mega_20240822_collection_B_ESP32_IRExt_4M316k**

- Not work with **ESP_Easy_mega_20241222_collection_B_ESP32_IRExt_4M316k**

tonhuisman, 2025-01-04:
I'll check if this command was excluded from the build or has stopped working for another reason.

tonhuisman, 2025-01-04:
@dyter69 
I've found the culprit, AFAICS, and applied a fix. [This Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/12612976911) is building the PR, can you please test once the build is complete? 

You can download the `Bin-ESP32-collection_B_ESP32_4M316k_ETH-20250104_PR_5213_12612976911` (.zip) file, extract the .bin file (not the .factory.bin) and upload using the `Update Firmware` button on the Tools page.

dyter69, 2025-01-04:
I tested but same problem, I hear the fist note of music.

tonhuisman, 2025-01-04:
Ah, that it actually started playing wasn't clear from the OP, I got the impression the command wasn't included in the build 🤔 

It this case it may be something to do with the framework being updated to IDF 5.1.x/5.3.x. More investigation is needed.

dyter69, 2025-01-04:
the command work ! but something stop the sound after one second I think

tonhuisman, 2025-01-04:
A few options:
- Can you switch to another GPIO pin, if only for testing? (even though GPIO 15 isn't a [special GPIO](https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp32))
- Do you have a rule that turns off the GPIO to stop the sound? (was sometimes needed for the old implementation, but shouldn't be needed any longer), playing the sound is now handled async, no longer blocking, and should fully stop/be quiet once completed.

dyter69, 2025-01-05:
Sorry but I have switched to **20230304** because I have lot of lost wifi connections since the **20240414**. And I found if the GPIO2 trigged to 1 will cause boot problem.

tonhuisman, 2025-01-11:
@dyter69 The RTTTL feature needed a fix because of the IDF 5.x updates. These are now applied, and [this GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/12723778121) is building for you to test.
Can you please report your findings?

TD-er, 2025-01-11:
> And I found if the GPIO2 trigged to 1 will cause boot problem

What do you mean by this remark?
If it is when pulling the GPIO-2 to some state, it will not boot, then that's nothing new.
It is part of how ESP32 and ESP8266 operate. See: https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#esp32-classic
So I guess you mean something different?

dyter69, 2025-01-11:
ok I tested with ESP_Easy_mega_20250111_collection_B_ESP32_IRExt_4M316k, buzzer work longer but music is bad with somes noises

TD-er, 2025-01-11:
Do you have an IR receiver task active?
Can you check if it makes a difference when you disable such a task?

dyter69, 2025-01-11:
Yes same problem when I disable all device, I listen a modem sound !

dyter69, 2025-01-11:
this play a modem sound: http://192.168.0.15/control?cmd=rtttl,15:d=32,o=5,b=63:16e,f,e,8b,16e6,f6,e6,8b,16e,f,e,16b,16e6,4d6,8p,4p,16e,f,e,8b,16e6,f6,e6,8b,16e,f,e,16b,16e6,4f6

dyter69, 2025-01-11:
It's strange, but sometimes it works with some notes are missing

tonhuisman, 2025-01-11:
Can you share the RTTTL command(s) you're using?
The playback is very dependent on the amount of time that is shared by other plugins, as TD-er already suggested by asking to disable all other tasks, so if the load is high, it may miss some notes. But that's not really changed since the last builds, though maybe the base-load is somewhat higher on the IDF 5.x releases.

dyter69, 2025-01-11:
It's been working like this for 3 years, and I've been updating it regularly:

```
http://192.168.0.15/control?cmd=rtttl,15:d=32,o=5,b=63:16e,f,e,8b,16e6,f6,e6,8b,16e,f,e,16b,16e6,4d6,8p,4p,16e,f,e,8b,16e6,f6,e6,8b,16e,f,e,16b,16e6,4f6
http://192.168.0.15/control?cmd=rtttl,15:d=4,o=5,b=250:e,8p,8f,8g,8p,1c6,8p.,d,8p,8e,1f,p.,g,8p,8a,8b,8p,1f6,p,a,8p,8b,2c6,2d6,2e6,e,8p,8f,8g,8p,1c6,p,d6,8p,8e6,1f.6,g,8p,8g,e.6,8p,d6,8p,8g,e.6,8p,d6,8p,8g,f.6,8p,e6,8p,8d6,2c6
...
```

TD-er, 2025-01-11:
Just a thought... if you have a status LED defined (top of Hardware tab), can you disable this, save the settings and power cycle the ESP board.

dyter69, 2025-01-11:
same problem, I go bak to ESP_Easy_mega_20240822_collection_B_ESP32_IRExt_4M316k and work fine !

TD-er, 2025-01-11:
Can you copy what is shown on the sysinfo page about the used SDK version?
And info about the CPU section?

dyter69, 2025-01-11:
This last firmware so work ESP_Easy_mega_20240822_collection_B_ESP32_IRExt_4M316k:

```
<html>
<body>
<!--StartFragment-->
Unit Number: | 0
-- | --
Local Time: | 2025-01-11 17:27:09
Time Source: | NTP
Time Wander: | 0.000 [ppm]
Uptime: | 00h28m
Load: | 48.27 [%] (LC=7617)
Internal Temperature: | 51.94 [°C]
CPU Eco Mode: | false
Boot: | Soft Reboot (0)
Reset Reason: | CPU0: Software reset CPU CPU1: Software reset CPU
Last Action before Reboot: | Background Task
SW WD count: | 0
Memory
Heap Size: | 306368 [byte]
Heap Min Free: | 170348 [byte]
Free RAM: | 192104 [byte] (170348 - )
Heap Max Free Block: | 110580 [byte]
Free Stack: | 7408 [byte] (4140 - )
Network
IP Config: | Static
IP / Subnet: | 192.168.0.15 / 255.255.255.0
Gateway: | 192.168.0.1
Client IP: | 192.168.0.82
DNS: | 192.168.0.2 / 0.0.0.0
Allowed IP Range: | 192.168.0.0 - 192.168.0.255
Connected: | 00:27:49.884
Number Reconnects: | 0
WiFi ?
WiFi Connection: | 802.11n (HT20) (RSSI -64 dBm)
SSID: | La roche network (F0:3E:90:17:9E:38)
WiFi TSF time: | 10:01:25.140679
Channel: | 8
Encryption Type: | WPA2/PSK
Last Disconnect Reason: | (1) Unspecified
Configured SSID1: | La roche network
Configured SSID2: |  
STA MAC: | 30:C6:F7:31:0B:10
AP MAC: | 30:C6:F7:31:0B:11

WiFi Settings
Force WiFi B/G: | false
Restart WiFi Lost Conn: | true
Force WiFi No Sleep: | true
Periodical send Gratuitous ARP: | false
Connection Failure Threshold: | 0
Max WiFi TX Power: | 17.50 [dBm]
Current WiFi TX Power: | 14.00 [dBm]
WiFi Sensitivity Margin: | 3 [dB]
Send With Max TX Power: | false
Extra WiFi scan loops: | 0
Use Last Connected AP from RTC: | false
Passive WiFi Scan: | true
Hidden SSID Slow Connect: | false
Include Hidden SSID: | false
Enable SDK WiFi Auto Reconnect: | false
Firmware
Build:⋄ | 20240822 - Mega32
System Libraries:⋄ | ESP32 SDK 4.4.6
Git Build:⋄ | mega-20240822
Plugin Count:⋄ | 80 ["Normal","Collection_B, IR with AC"]
Build Origin: | GitHub Actions
Build Time:⋄ | Aug 22 2024 20:38:28
Binary Filename:⋄ | ESP_Easy_mega_20240822_collection_B_ESP32_IRExt_4M316k
[... 113 lines cut ...]
ESP Board
ESP Chip ID:	3214096 (0x310B10)
ESP Chip Frequency:	240 [MHz]
ESP Crystal Frequency:	40 [MHz]
ESP APB Frequency:	80 [MHz]
ESP Chip Model:	ESP32-D0WDQ6
ESP Chip Features:	Wi-Fi bgn / BLE
ESP Chip Revision:	1.0
ESP Chip Cores:	2
ESP Board Name:	Espressif Generic ESP32 4M Flash ESPEasy 1810k Code/OTA 316k FS
Storage
Flash Chip ID:	Vendor: 0x68 Device: 0x4016
Flash Chip Real Size:	4096 [kB]
Flash IDE Size:	4096 [kB]
Flash Chip Speed:	80 [MHz]
Flash IDE Speed:	40 [MHz]
Flash IDE Mode:	DIO
Flash Writes:	7 daily / 7 boot
Sketch Size:	1825 [kB] (1856 kB free)
Max. OTA Sketch Size:	1856 [kB] (1900544 bytes)
SPIFFS Size:	283 [kB] (126 kB free)
Page size:	256 [byte]
Block size:	8192 [byte]
Number of blocks:	35
```

TD-er, 2025-01-11:
And which GPIO are you using for the RTTTL?

dyter69, 2025-01-11:
15 (you can see it in the previous conversations)

TD-er, 2025-01-11:
I don't think it makes a difference what pin is used, but just to be sure I can first reproduce what you're seeing

tonhuisman, 2025-01-11:
> It's been working like this for 3 years, and I've been updating it regularly, system load is about 30%:
> 
> ```
> http://192.168.0.15/control?cmd=rtttl,15:d=32,o=5,b=63:16e,f,e,8b,16e6,f6,e6,8b,16e,f,e,16b,16e6,4d6,8p,4p,16e,f,e,8b,16e6,f6,e6,8b,16e,f,e,16b,16e6,4f6
> http://192.168.0.15/control?cmd=rtttl,15:d=4,o=5,b=250:e,8p,8f,8g,8p,1c6,8p.,d,8p,8e,1f,p.,g,8p,8a,8b,8p,1f6,p,a,8p,8b,2c6,2d6,2e6,e,8p,8f,8g,8p,1c6,p,d6,8p,8e6,1f.6,g,8p,8g,e.6,8p,d6,8p,8g,e.6,8p,d6,8p,8g,f.6,8p,e6,8p,8d6,2c6
> ...
> ```

Hmm, you're missing a comma after the GPIO pin 15 there, in both commands... that'll make some difference, I assume 🤔

tonhuisman, 2025-01-11:
@dyter69 I've been made aware of a typo I made in the previous build, that I fixed now. A new build is [running here](https://github.com/letscontrolit/ESPEasy/actions/runs/12727357338), can you please test again, to see if this solves the playback?
I've tried your songs, but they don't play very nice, could you also try to play the Arkanoid song from the examples?
`rtttl,15,Arkanoid:d=4,o=5,b=140:8g6,16p,16g.6,2a#6,32p,8a6,8g6,8f6,8a6,2g6`

TD-er, 2025-01-15:
@dyter69 Were you able to test the test build linked by Ton?

uzi18, 2025-01-15:
Both examples works ok here: https://adamonsoon.github.io/rtttl-play/
Arkanoid also tested ok.

tonhuisman, 2025-01-15:
> Both examples works ok here: https://adamonsoon.github.io/rtttl-play/ Arkanoid also tested ok.

We're trying to solve an issue about not properly playing RTTTL songs _at all_. The poor playback in my test-setup may be caused by the ESP or buzzer I'm using, as the provided songs don't sound very nice too _my_ ears but Arkanoid plays 'normal', so there remain 2 questions:
- Does it play as expected for @dyter69 with the latest code adjustments?
- Does the Arkanoid song play as is to be expected (you can use the adamonsoon site for reference), with rather clearly defined notes? (the other songs sound quite blurry)

NB: The blurry playback may be related to the buzzer I'm using, and the frequencies it works best with.

dyter69, 2025-01-15:
I all ready tested with Bin-ESP32-collection_A_ESP32_IRExt_4M316k-20250104 and sound i poor

TD-er, 2025-01-15:
> I all ready tested with Bin-ESP32-collection_A_ESP32_IRExt_4M316k-20250104 and sound is poor

But did you test with the build linked by Ton here: https://github.com/letscontrolit/ESPEasy/issues/5212#issuecomment-2585400126
As he had a typo in his code where ESP32-code wasn't included in the build....

dyter69, 2025-01-15:
yes same problem

tonhuisman, 2025-01-15:
@dyter69 I've hooked up an ESP32 with 4MB Flash, and installed the same build you used. First tested with the 2023-ish build on that unit, and playback sounded quite normal, so I knew what to expect/achieve 🙄 
Using the previous fixes it sounded quite bad, and the volume was higher then before, so after reviewing the code I tested the small fix I made now, and sound is the same as with the old build.
A [fresh GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/12797786364) is building, so if you can please verify it also works for you, we can resolve this issue.
TIA.

dyter69, 2025-01-16:
Ok I flash ESP_Easy_mega_20250115_collection_B_ESP32_IRExt_4M316k:

it almost works but at times it sounds like a cricket or modem !

tonhuisman, 2025-01-16:
> it almost works but at times it sounds like a cricket or modem !

That is most likely caused by (busy) activity of the ESP, it's using cooperative multi-tasking like scheduling, for that's how the AnyRtttl library works.
I'll see how that can possibly be improved, but will probably have to wait a bit.
I suggest this fix be merged now, as the fix (only) involves the playing of sounds that wasn't working anymore because of changes in the IDF 5.x framework, and not by the implementation in ESPEasy or changes in the AnyRtttl library.
Any further improvements should IMHO better be handled separately.

dyter69, 2025-01-16:
no problem I downgrade to ESP_Easy_mega_20240822_collection_B_ESP32_IRExt_4M316k, thank !

TD-er, 2025-01-16:
Discussed with Ton and it is best to merge the fixes for now as it is at least an improvement.
However, there might still be room for improvement when running the RTTTL playback on an ESP32 in a separate RTOS task.

So that's why I will re-open it for now so it will remain 'in focus'.

PR #5213 (2025-01-16): [Bugfix] RTTTL feature was not working with ESP32 IDF 5.x
Resolves #5212

Bugfix
- Feature RTTTL ~wasn't enabled for ESP32 builds because of~ a missed S&R
- Apply IDF 5.x adjustments to RTTTL feature

TODO:
- [ ] Testing by reporter