letscontrolit/ESPEasy #5219: [BUG]favicon not shown Build: ESP_Easy_mega_20241222_normal_ESP8266_4M1M_VCC Dec 22 2024

Suppenlaender, 2025-01-08:
**Describe the bug**
The intern web-server does not correctly generate the href for the favicon.ico and/or does not load correctly and/or favicon does not exist

There always a link like:
<link rel='icon'  type='image/;x-icon'  href='[static_3601315144_favicon.ico](view source:http://localhost/static_3601315144_favicon.ico)' /> (localhost is an placeholder for my IP-Address)

I think there are something wrong with the generation of that href link (looks like inserting of time code and _static_)

**To Reproduce**
Steps to reproduce the behavior:

Starting the web server address at any level --> no favicon shown

**Expected behavior**

For me this is only a cosmetic lack

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]
This is a only a detail of the src code

**Used platform (please complete the following information):**
 - ESP type:  ESP8266 
 - Build version: mega-20241222
 - Build set normal

**Platform Specifics (when applicable, please complete the following information):**
 
ESP Board
ESP Chip ID: | 15794578 (0xF10192)
ESP Chip Frequency: | 80 [MHz]
ESP Chip Model: | ESP8266EX
ESP Chip Cores: | 1
ESP Board Name: | Espressif Generic ESP8266 ESPEasy 4M Flash 1M FS

tonhuisman, 2025-02-16:
> web-server does not correctly generate the href for the favicon.ico and/or does not load correctly and/or favicon does not exist

This is generated in this way so the fav-icon is retrieved from browser-cache after an initial 'download' (hosted by the ESP), so the page can load quicker, without doing another request on the webserver.

TD-er, 2025-02-24:
Looking at the code now and there seems to be a mixup in code paths, as even MAX builds don't seem to serve favicon.ico