letscontrolit/ESPEasy #5220: [BUG] GPIO 10 and 11 not available

XerxesKB, 2025-01-08:
**Describe the bug**
GPIOs 10 and 11 on the Olimex ESP32-C6-EVB board are unusable with the current build. Both pins are connected to relays, but are not available.
If triggered via a command, it fails, and the log displays the error message "GPIO: port#11 is out of range."

**Screenshots**
[image]
[image]

**Used platform (please complete the following information):**
 - ESP type: ESP32-C6
 - Build version: mega_20241221
 - Build set normal

**Platform Specifics (when applicable, please complete the following information):**
 - Flash size: 4M
 - Brand/Model: Olimex ESP32-C6-EVB

TD-er, 2025-01-08:
Can you show the flash and CPU section of the sysinfo page?

I guess this board is having an external flash chip with less GPIO's connected to the flash chip so only DIO possible.

XerxesKB, 2025-01-08:
[image]

TD-er, 2025-01-08:
I found the bug in my code, but I am not 100% sure yet which GPIO pins are connected to the flash on the C6.
In the ESPEasy code, GPIO 10 and 11 are blocked as being connected to (embedded) flash as well as GPIO-24 .. GPIO-30 (excl. GPIO-17)

However, based on the Olimex schematic, ESP32-C6 TRM and ESP32-C6-DevKitM-1, I have to revise those numbers...

From the ESP32C6 TRM:
[image]

From the [ESP32-C6-DevKitM-1](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitm-1/user_guide.html):
[image]

These would suggest:
- FSPID = GPIO-7
- FSPIQ = GPIO-2
- FSPIWP = GPIO-5
- FSPIHD = GPIO-4
- FSPICLK = GPIO-6
- FSPICS0 = GPIO-16

TD-er, 2025-01-08:
OK, those pins I mentioned are not connected to the internal flash.

See page 19 in the [ESP32-C6 datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-c6_datasheet_en.pdf0
The red pads are the ones connected to flash.
No idea where the GPIO-10/11 limitation comes from, but I will remove it and make a test build for you.

XerxesKB, 2025-01-08:
Tested. Now i can toggle the relais just fine.

TD-er, 2025-01-08:
Great!
Thanks for testing.

PR #5221 (2025-01-08): [BUG] GPIO 10 and 11 not available (#5220)
Fixes: #5220

Remove GPIO-10 and -11 from the 'assigned to flash' checks.
commit 5db4d00b (2025-01-08): [BUG] GPIO 10 and 11 not available (#5220)

Fixes: #5220
closed 2025-01-08 by TD-er