letscontrolit/ESPEasy #5236: [FR]Is it possible to add the TFT ILI934x/ILI948x to the ESP32 Climate version?

hrikken, 2025-01-30:
I've tried to get up and running with VSCode / Platformio to build my special request but that ends up in errors. Since I'm not a hard core programmer I got lost in the setup and customizations.

I would be grateful if the TFT ILI934x/ILI948x plugin cloud be added (by someone) to the Climate ESP32 4M315 version.

Is that possible?

Thanks in advance,

Harry

TD-er, 2025-01-30:
Maybe the other way around is more practical?
What plugins from the climate build do you need?

hrikken, 2025-01-30:
Thanks for the reply. What I need from climate is:
- The ILI displays
- SEN55 
- BMx2/6/80
- SHT3/4x
- SCD4x

hrikken, 2025-02-03:
Maybe I misunderstood the replies: is someone going to create the binary for me?  

Regards,

Harry

tonhuisman, 2025-02-03:
I've added P095 (ILI9xxx), P099 (XPT2046), P116 (ST77xx) and P123 (I2C touchdisplays) to the regular Custom build:

[ESP_Easy_mega_20250203_climate_ESP32_4M316k.zip](https://github.com/user-attachments/files/18648017/ESP_Easy_mega_20250203_climate_ESP32_4M316k.zip)

Includes both the regular .bin file for uploading via OTA, and the .factory.bin for uploading via a Flash upload tool.

hrikken, 2025-02-03:
Flashed it. Working right away ! Thanks a lot...

closed 2025-02-07 by hrikken

uzi18, 2025-02-07:
@tonhuisman do you have any reference for such st/ili problematic LCDs and solutions? 
Have one marked as ST7789 but this driver does not work for me. Some users reports as it works with ili9341 driver but with different init configuration.

tonhuisman, 2025-02-07:
> [@tonhuisman](https://github.com/tonhuisman) do you have any reference for such st/ili problematic LCDs and solutions? Have one marked as ST7789 but this driver does not work for me. Some users reports as it works with ili9341 driver but with different init configuration.

You can just try all available options in both P095 and P116 until you find one that works. Shouldn't break anything if it doesn't match, might just display something weird.
If you want to experiment with different initialization sequences (quite time-consuming, tbh), I've used examples from [Bodmer's TFT_eSPI](https://github.com/Bodmer/TFT_eSPI) library, and integrated that in ESPEasy. Please keep the available configurations in the current enumeration order/values, as these are stored in config, so should stay compatible with users setups, if you want to add a new configuration to either of the plugins.

NB: I'm planning on replacing the libraries currently in use by Arduino_GFX and LVGL graphics, as I want to add support for the 8-bit parallel displays seen on some ESP32-S3 based units. (Work on that will probably start in the coming weeks.)
This should allow for more flexible configuration adjustments, for stubborn/specific displays like yours.

uzi18, 2025-02-09:
Ok have followed your advice. So it looks like ILI9341 works even marked as ST7789. Now need to dig into init functions path - working set of registers ;)