letscontrolit/ESPEasy #5237: [FR] Make it optional to use SMTPS for email notifications on ESP32 devices.

jbro90, 2025-01-31:
I am using a simple postfix MTA as destination for the email notifications of my ESPEasy devices. It is a simple internal MTA and there is no specific need for secure data transmission. 

However in recent releases the email notifications for ESP32 devices seems to default to SMTPS, without any facilities to use the unencrypted protocol.

I have extensively tried to reconfigure my postfix MTA to accept SMTPS traffic on port 465 however am still not able to deliver emails from ESP32 devices.

What I would like is an additional option on the notifications-email page in which I can specify to use SMTP or SMTPS . A big thank you to the dev team for prioritizing security! However, having a simpler protocol available would be beneficial and could also aid during troubleshooting exercises.

TD-er, 2025-01-31:
As far as I know/have read, the difference between TLS and SSL is that the one using port 465 (the not yet working one) does make an unencrypted connection and then tries to negotiate the cypher and start the SSL connection.

So far I have no clue why it doesn't work, which makes it a bit harder to fix it.

jbro90, 2025-01-31:
@tonhuisman: I have tried port 587 and unfortunately it doesn't work either. In the MTA logs it behaves similar as port 25, indicating that probably encrypted data is in the input while unencrypted data is expected. According the TLS protocol this is expected behavior as the encrypted part on port 587 only starts after the STARTTLS command. 

@TD-er: Everything is working fine when I'm using a linux mail client to send ssl encrypted mail to port 465 of the MTA. Debug info of the MTA during mail transmission of the ESP shows that initial steps of the security handshake do occur, but then the MTA logs "no shared cipher". 

ESPEasy  syslog at debug-dev level shows:
`NeoPixel-149_EspEasy: connectClient: connect failed to '192.168.10.38' port: 465`
`NeoPixel-149_EspEasy: Email: Error connecting to r38.ho.me:465 Error code: 296`

However we are now troubleshooting smtps facilities of ESPEasy, and it was my intention to have an option to use plain smtp instead of smtps or TLS in the ESP32 arena. Does this option have any chance or should I rather use the good old 8266 which is not capable to do any encryption processing :)

TD-er, 2025-01-31:
Ah check, I thought it was still possible to _not_ use encryption when using port 25.
Apparently not, or else you wouldn't ask for it :)

tonhuisman, 2025-01-31:
@jbro90 I've tried to fix this by explicitly testing for port 25 in the configuration to use the default (insecure) SMTP communication, also on ESP32.
Can you please test a, ESP32 build from [this GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/13080837186), to validate this? (for ESP8266 there should be no functional change, as TLS is not supported on that hardware platform).

jbro90, 2025-02-01:
@tonhuisman I have installed one of the suggested [binaries](https://github.com/letscontrolit/ESPEasy/actions/runs/13081672184/artifacts/2519825705) and mail is now nicely delivered to its destination. So postive results for validation of the fix. I'll do some more analysis on the smtps issues i ran into, but that's another story. Anyway thanks for your attention and swift action to have smtp back on track.

tonhuisman, 2025-02-01:
I've added the specific port information to the N001 documentation.

jbro90, 2025-02-01:
Thanks again!!!

closed 2025-02-01 by jbro90

tonhuisman, 2025-02-01:
No need to manually close this, as it will be closed by GH when the PR is merged (but also no need to re-open, it's fine as it is 😃). We usually keep issues open until merged, to help avoid people to report duplicates.

PR #5238 (2025-02-01): [N001] Use a non-TLS client for SMTP on port 25
Resolves #5237 

Bugfix:
- Even when using port 25 a secure client was started for the mail-server connection. Now when using port 25, or port 2525, a default WiFiClient is used, instead of a secure client.
- Close & destroy client also when server-connection (login?/network?) fails.
- Documentation updated about the use of ports 25 and 2525 for ESP32 builds with TLS included.

TODO:
- [x] testing by requester ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/5237#issuecomment-2628921164))