letscontrolit/ESPEasy #5239: [BUG] When not subscribing to an MQTT controller, connection can't be established due to malformed packet

Boerny41, 2025-01-31:
**Describe the bug**
When leaving the `Controller Subscribe:` field empty, the device can no longer connect to MQTT:
`Client espCo2 disconnected due to malformed packet` 

Sending the data back to the device seems pretty useless and produces a lot of spam.

**To Reproduce**
Steps to reproduce the behavior:
1. Add MQTT Server, don't change default settings
2. Check logs of MQTT server to see the client connecting
3. go to the EasyESP device, `Controllers` → `YourMqttServer` → leave `Controller Subscribe:` blank
4. See error

**Expected behavior**
When leaving the field blank, I expect the ESP subscribing to nothing, but still sending its data. 

**Used platform (please complete the following information):**
 - ESP type: [e.g. ESP8266 & ESP32-C3]
 - Build version: [8266 ran on old version 2021XXX updated to mega-20241222. 20241222 for ESP32]
 - Build set [e.g. normal & climate]

tonhuisman, 2025-01-31:
Hm, I guess this is a generic MQTT Controller issue.
What MQTT Controller do you have configured? (so I can make a change there for you to test, and apply the same fix to other conrtollers)

TD-er, 2025-01-31:
I think this is indeed a bug which probably has been there for years.
The MQTT broker will let you know there was an error by disconnecting the client.
And an empty subscribe topic seems to be worthy of an error.

tonhuisman, 2025-01-31:
@Boerny41 I've made a fix for this issue. (Also for P037 MQTT Import, that could use a small improvement in this area.)
Can you please test using a build from [this GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/13081271395), and report your findings here?

Boerny41, 2025-01-31:
I'm afraid not, when trying to flash the normal bin I get: `Update error: Could Not Activate The Firmware`
When trying to flash the .factory.bin I get `Update error: Decryption error`

Tried to flash it on a freshly installed ESP32-C3 - mega-20241222

TD-er, 2025-01-31:
Which exact file did you flash?
And what exact firmware file is now installed (shown in the bottom right of each page and on the sysinfo page)
The 'factory' file can only be flashed via serial/USB.
The one **_without_** "factory" can be flashed via the update link.

Boerny41, 2025-02-01:
Currently installed is:
Build: [ESP_Easy_mega_20241222_climate_ESP32c3_4M316k_CDC Dec 22 2024]

Tried to flash:
ESP_Easy_mega_20250131_climate_ESP32_4M316k.bin via `Tools` -> `Update Firmware`

tonhuisman, 2025-02-01:
> Currently installed is: Build: [ESP_Easy_mega_20241222_climate_ESP32c3_4M316k_CDC Dec 22 2024]
> 
> Tried to flash: ESP_Easy_mega_20250131_climate_ESP32_4M316k.bin via `Tools` -> `Update Firmware`

That's not a match, you need an `ESP32c3` build, but you selected a classic ESP32. Completely different CPU architecture...

Boerny41, 2025-02-01:
I think it was the only one available yesterday. Now there are a lot more option, but something seems to have broken.

[image]

tonhuisman, 2025-02-01:
> Now there are a lot more option, but something seems to have broken.

GH made a booboo, yesterday, but AFAICS your required build should be [this one](https://github.com/letscontrolit/ESPEasy/actions/runs/13081271395/artifacts/2519671273)

Boerny41, 2025-02-01:
Yes, this seems to be working! Thank you all!

While talking about MQTT I think there is another bug, but I couldn't verify it with MQTT logs so I'm not entirely certain.
The `Single event with all values:` seems to do nothing.

I've changed the topic to `%sysname%/%tskname%` so all the values from the BME280 get sent via the same topic.
The checkbox is marked, so I should get a single MQTT message with all three values, however I still get three individual ones.

Down below are some screenshots of the relevant settings:

**MQTT settings**
[image]

**Sensor settings***
[image]

**Received MQTT messages**
[image]

Firmware used on this one is the latest public from December: 
ESP_Easy_mega_20241222_climate_ESP8266_4M1M Dec 22 2024

TD-er, 2025-02-01:
Nope, single event is about events as handled in rules.
It is NOT about joining taskvalues into a single MQTT message.

tonhuisman, 2025-02-02:
> While talking about MQTT I think there is another bug, but I couldn't verify it with MQTT logs so I'm not entirely certain. The `Single event with all values:` seems to do nothing.
>
> I've changed the [...]

That's not how MQTT works. MQTT is about small messages to specific topics, and takes max. 1 payload per topic-message, not multiple.

Sending all values to a single topic will just mess up your MQTT data, as these values are of different unit and scale, so the most used feature, creating graphs of measured values, will be a chaos and obviously useless.

PR #5240 (2025-02-03): [MQTT] Don't subscribe to empty topics
Resolves #5239 

Bugfix:
- MQTT: Don't subscribe to empty topics

TODO:
- [x] Testing by reporter ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/5239#issuecomment-2629143381))
closed 2025-02-03 by TD-er