letscontrolit/ESPEasy #5246: [FR] Add HTTPS support to HTTP rule commands

BasvanH, 2025-02-07:
**Describe the solution you'd like**
I want ESP Easy to communicate with public API interfaces where it's mandatory to use HTTPS. Even within your own network it would be feasible to use encrypted requests because of security concerns.

**Describe alternatives you've considered**
Use a proxy which tunnels the HTTP request to HTTPS, but that requires to run this somewhere. It's prefeered that EasyEasy has no dependencies.

**Additional context**
So for these commands I would like HTTPS support:
SendToHTTP
PutToHTTP
PosttoHTTP

Thanks in advance.

tonhuisman, 2025-02-07:
It's already on the TODO list, registered in #3838

BasvanH, 2025-02-07:
I knew it was somewhere already mentioned, but couldn't find it.

Do you have any estimation in when this will be implemented? Not to push what so ever, but this decides what road to take for a current project. Thanks!

TD-er, 2025-02-07:
It does depend on a number of other pending changes:
- WiFi/Network rewrite must be done first
- Transition to PioArduino, which will also determine whether I will continue using BearSSL or mBedTLS (which allows me to validate certificates)

The first one has already been taken way more time than I expected and is held back by other personal time consuming issues.
The transition to PioArduino is hard to estimate how much time it will take as I need to use the recently implemented feature to essentially build your own platform build.
This is needed as currently when I build ESPEasy using 'vanilla' PioArduino, the builds are 700 - 900 kB larger than we have now. So I need to make our own dependency list of what to include and what to exclude.
I can of course start with what Jason2866 has been doing for our ESPEasy specific build of the platform images, but changing modules may open a can of worms as it is hard to predict what will be linked and thus increasing the build size.
Right now, the current Arduino3.x/ESP-IDF5.x code does not allow to use LTO (which lets the linker decide which function is actually used and which isn't), which does make it rather tricky to enable/disable features, encryption ciphers, etc.

BasvanH, 2025-02-12:
Thanks for the indepth insights, that looks like a lot of work needs to be done prior to the SSL support. For now I will build my own code to support my needs, I will keep a lookout when ESPEasy gets SSL support.

Thanks for the great work!

uzi18, 2025-02-12:
Please try to catch headers sent from https server