letscontrolit/ESPEasy #5247: Cannot draw BMP files to ILI9341

hrikken, 2025-02-07:
Hi folks,

Maybe I'm not reading the docs good enough but recently I've tried to write a BMP image to an ILI9341 display.

Step 1: I saved a bmp file named "cloudy.bmp" file to a microsd card (FAT format) which I inserted in the ILI display.

Step 2: **I entered the following command: _tft,bmp,0,0,cloudy.bmp_**

After entering the command above via the tool menu, espeasy replies with:

**_Command unknown: tft,bmp,0,0,cloudy.bmp_**

I've tried quite a few other command options but the reply is always the same.

Any thoughts or recommendations?

Thanks in advance,

Harry

TD-er, 2025-02-07:
Just for reference the command syntax: https://espeasy.readthedocs.io/en/latest/Plugin/P095.html#commands-available

Maybe you can use `ili9341` instead of `tft` ?

Also make sure the bmp file is either 24-bit or 1-bit format.
Can you also try uploading a small bmp file to the internal file system instead of the SD card, just to test if the command syntax is correct?

I don't expect the filename to be case-sensitive, but just to be sure, try to use lower case letters in the filename.

tonhuisman, 2025-02-07:
What's the size in pixels and orientation of your cloudy.bmp? There's a check on the size of the bmp vs the available resolution of your display, and there's no clipping implemented, so the bmp pixel-sizes have to be smaller or equal to the display resolution, in the current Rotation setting.

uzi18, 2025-02-07:
You can also attach your cloudy.bmp here in post

hrikken, 2025-02-07:
[cloudy.bmp.zip](https://github.com/user-attachments/files/18707739/cloudy.bmp.zip)

hrikken, 2025-02-07:
None of these work.

uzi18, 2025-02-07:
First is 24bit 512x512px

hrikken, 2025-02-07:
I've uploaded a 50 x 50 file to the internal file system and that didn't work either: 

[cloudy5050bmp.bmp.zip](https://github.com/user-attachments/files/18707899/cloudy5050bmp.bmp.zip)

tonhuisman, 2025-02-07:
```
Info   : showBmp: bitmap w:50, h:50, dpt:32, colors:1, cmp:3, pl:1, x:0, y:0
Info   : showBmp: x:0, y:0, dw:160, dh:128
Error  : showBmp: Only uncompressed and 24 or 1 bit color-depth supported.
```
Tried your 50x50 cloudy.bmp file on a 80x160 px display, but there's something weird with the file, so that's why it's not working.
Looks like it's a 32 bit (dpt;32) color or compressed bmp. Windows shows it correctly, but bmp support in ESPEasy is quite limited to save space/code size.

TD-er, 2025-02-07:
[cloudy5050_24b.zip](https://github.com/user-attachments/files/18713880/cloudy5050_24b.zip)

I converted the file into 24-bit BMP using IrfanView.

hrikken, 2025-02-08:
> [cloudy5050_24b.zip](https://github.com/user-attachments/files/18713880/cloudy5050_24b.zip)
> 
> I converted the file into 24-bit BMP using IrfanView.

Thanks guys. 
That Irfanview convert did the trick. 
Time for me to figure out why SnagIt corrupts the file. IrfanView is an golden-oldie.
Sorry for the hassle.

Regards,

Harry

closed 2025-02-08 by hrikken

uzi18, 2025-02-08:
Original picture is png? They have alpha channel

TD-er, 2025-02-08:
Well it was a bit ( ;) ) strange indeed.
It was marked as 24-bit, but still using 32 bit per pixel.
I can imagine this is faster for viewing as you can read 32 bit per pixel and don't need to perform bitshifts.
Anyway, using older tools is often the best when you need to have a specific format as they don't try to "think and decide what is best".