letscontrolit/ESPEasy #5249: [BUG]Strange behavior of calculated measurement in P2P networking

hrikken, 2025-02-08:
**Describe the bug**
I added a sensor (in this case a DS18B20 1-wire).
When I send the measurement to a second controller via P2P networking it works fine.
However as soon as I add a formula (f.i. a calibration of the sensor with -3 C e.g. (%value%)-3) the value is only sent once and after that never again. 

**To Reproduce**
Steps to reproduce the behavior:
1. Enable P2P networking on both sides
2. Add a sensor (DS18B20) on the sending unit
3. Enable 'Send the value to controller'
4. Subtract -3 via a formula (%value%)-3.
5. Submit
6. Add the sensor via P2P on the receiving node (e.g. Remote unit = 8)

**Expected behavior**
I would expect to receive the calculated value of the sensor on the receiving A clear and concise description of what you expected to happen.

**Used platform (please complete the following information):**
 - ESP type: ESP32
 - Build version: Climate with ILI93xx added [ESP_Easy_mega_20250203_climate_ESP32_4M316k Feb 3 2025]
 - Build set: Climate

hrikken, 2025-02-08:
Sorry to be such a nuisance

TD-er, 2025-02-08:
And what if you use the formula also on the receiving node?
I think I need to check the p2p code to be sure.
What I think is happening is this:

ESPEasy does use a large struct to store the taskvalues. This is historically a 'float', except for some specific plugins like RFID readers.
However to make sure not a lot of resolution is lost when applying the formula, the raw value is stored in this struct and every time a value is requested, the formula is applied and this calculation is done as `double` (more decimals precision)

This also fixes strange issues we had before, where a value was restored at boot from RTC memory and the formula was applied twice.

So what I think is happening here is that the p2p controler is using the raw values to send to the other node.
Not sure if the formula should also be transferred to the remote node, or just the calculated result.
Problem is that in a formula you can also use `%pvalue%` which is the previous value.
But when the receiving node misses a single update, this yields different values compared to the sending node.

So it isn't that simple to 'fix'.
Maybe just setting the same formula on the receiving node will be a usable work-around for now till we can make a more sensible fix.

tonhuisman, 2025-02-08:
> Sorry to be such a nuisance

This is not a nuisance, but helpful in finding issues with ESPEasy. 👍

hrikken, 2025-02-09:
Hi TD-er, 
Your suggestion to copy the formula on the receiving end doesn't fix the issue. 
I've checked the log on the receiving end but values are still coming in only they keep the same all the time. 
As soon as I remove the formula at the sender the new measurements are coming in regularly.

hrikken, 2025-02-09:
Workaround via MQTT. For the time being ok now.