letscontrolit/ESPEasy #5251: [FR] esp32 lolin32 lite - i2c pins for SDA/SCL 2/15 not 21/22

gmx168, 2025-02-10:
I am using Build: ESP_Easy_mega_20241222_neopixel_ESP32_4M316k Dec 22 2024 for ESP Lolin32 lite. 

**Describe the solution you'd like**
The proposed and hard coded I2C pins SDA/SCL for generic ESP32 are 21/22. The problem is that at ESP Lolin32 lite are SCL/SCA 2/15 and pin 21 is not provided. Pin 22 is strapped to internal LED Blue sinking to 3V.

**Describe alternatives you've considered**
Possible to change set of i2c to different set according to the ESP32 used. I need only one bus at the time.

**Additional context**
I2c used to control LCD2004.

tonhuisman, 2025-02-10:
> The proposed and hard coded I2C pins SDA/SCL for generic ESP32 are 21/22. The problem is that at ESP Lolin32 lite are SCL/SCA 2/15 and pin 21 is not provided. Pin 22 is strapped to internal LED Blue sinking to 3V.

For an ESP32 clean install we explicitly _do not_ set the I2C GPIO pins to a default value. You can create a Custom build where you can set these GPIO pins as desired.
We also don't use specific ESP32 hardware definitions (that is currently already a list of several hundreds, slightly different, boards, not easily maintainable), instead we have 'safe' defaults as much as possible, and you'll probably need the hardware documentation to configure these interfaces, especially for boards with sensors, displays, etc. included.

gmx168, 2025-02-11:
@TD-er it is blocked, change of i2c pins are forbidden in the Hardware definition.
@tonhuisman I will try it. Otherwise I put another esp8266 as LCD host display and interconnect to the esp lolin32 lite via p2p.

tonhuisman, 2025-02-11:
> it is blocked, change of i2c pins are forbidden in the Hardware definition.

Can you show a complete screenshot of your Hardware page?
Most likely you have conflicting GPIO definitions (SPI?), or a Priority task defined (AXP192/AXP2101?), causing the I2C pins to be blocked.

gmx168, 2025-02-11:
Sure.

[image]

TD-er, 2025-02-11:
No the config tab in ESPEasy.
And then of the complete page, including the section at the bottom where you can define the boot states of each pin.
There it shows also the reason why a pin might be blocked.

tonhuisman, 2025-02-11:
> Can you show a complete screenshot of your Hardware page?

I was trying to ask for a screenshot from the ESPEasy Hardware configuration page..., maybe I wasn't expressing myself clear enough 🤔

gmx168, 2025-02-11:
no problem. you meant here?:

[image]
when I change to needed ones without submitting to the interface, it looks like below:

[image]

still before submittal:

[image]
observed in the log external com terminal:

[image]
even no change.

Afetr submitting - the values are changed back to the 21/22

TD-er, 2025-02-11:
And what is assigned to GPIO-15 and -2?
So the rest of the Pin mode GPIO-xx part of the page

TD-er, 2025-02-11:
You can also have a look here:
https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp32

Those pins you try to use aren't the best choice for I2C, but I understood you don't really have a choice as the manufacturer already picked those.

gmx168, 2025-02-11:
"Sensors": [
    {
      "DataAcquisition": [
        {
          "Controller": 1,
          "IDX": 0,
          "Enabled": "false"
        },
        {
          "Controller": 2,
          "IDX": 0,
          "Enabled": "false"
        },
        {
          "Controller": 3,
          "IDX": 0,
          "Enabled": "false"
        }
      ],
      "TaskInterval": 60,
      "Type": "Output - NeoPixel (Basic)",
      "TaskName": "GATE_PREP_LED1",
      "TaskDeviceNumber": 38,
      "TaskDeviceGPIO1": 4,
      "TaskEnabled": "true",
      "TaskNumber": 2
    },
    {
      "DataAcquisition": [
        {
          "Controller": 1,
          "IDX": 0,
          "Enabled": "false"
        },
        {
          "Controller": 2,
          "IDX": 0,
          "Enabled": "false"
        },
        {
          "Controller": 3,
          "IDX": 0,
          "Enabled": "false"
        }
      ],
      "TaskInterval": 60,
      "Type": "Output - NeoPixel (Basic)",
      "TaskName": "GATE_PREP_LED2",
      "TaskDeviceNumber": 38,
      "TaskDeviceGPIO1": 5,
      "TaskEnabled": "true",
      "TaskNumber": 3
    },
    {
      "TaskValues": [
        {
          "ValueNumber": 1,
          "Name": "stanbram2",
          "NrDecimals": 0,
          "Value": 1
        }
      ],
      "DataAcquisition": [
        {
          "Controller": 1,
          "IDX": 0,
          "Enabled": "false"
        },
        {
          "Controller": 2,
[... 232 lines cut ...]
          "Controller": 1,
          "IDX": 0,
          "Enabled": "false"
        },
        {
          "Controller": 2,
          "IDX": 0,
          "Enabled": "false"
        },
        {
          "Controller": 3,
          "IDX": 0,
          "Enabled": "false"
        }
      ],
      "TaskInterval": 0,
      "Type": "Switch input - Switch",
      "TaskName": "EXTERNAL_LAMP_R",
      "TaskDeviceNumber": 1,
      "TaskDeviceGPIO1": 25,
      "TaskEnabled": "true",
      "TaskNumber": 11
    }
  ],
  "TTL": 1000

TD-er, 2025-02-11:
What I meant is this part of your setup

[image]

On the Hardware tab

gmx168, 2025-02-11:
The issue goes for the problem definition:

[image]
How to add LCD2004 with non standard PINS like 21/22 as SDA/SCL?

TD-er, 2025-02-11:
On a test build on my ESP32-node here on my desk, I am able to set the pins you mentioned:

[image]

Are you sure you can change any setting at all?

Can you also have a look at the flash info on the system info page:
It should show something like this:

[image]

TD-er, 2025-02-11:
OK, so there should be no reason why the pins should be rejected.
This makes me believe we're looking at another problem.
I've seen reported issues by some users where they were unable to save any settings, so can you please take a look at the size of the file system as reported on the system info page?

gmx168, 2025-02-11:
sure: Load is not so high, no rules at this moment.

[image]

gmx168, 2025-02-11:
ok, solved issue so far.
Requested information for double verification:

[image]

the problem was related to the predefinition of onbard LED, I put Onbard wifi led as GPIO22, if I put empty 

[image] I can change SCL/SDA pins.

[image]

Thank you for helping non-existing problem so far, the system behaves as designed.

gmx168, 2025-02-12:
I am programming directly in C++ for the modules like Arduino (for fun and brain stretching), I am not using Tasmota/Esphome.

For some functions better to switch to Espeasy than just digging in the code directly.

Good to know that there is a great support here. 

Thank you.

closed 2025-02-12 by gmx168

TD-er, 2025-02-12:
You're welcome. :)

About the 'filesystem size' as possible cause I was thinking of... I don't yet know what could cause such wrong partitioning. I have now seen about 2 reports of it, both quite recent.
At least one of them was using a flashing tool from one of the other firmwares (ESPHome) to flash an ESPEasy binary.
And I know that if there is a file system partition present, ESPEasy will not adjust it but just uses it. 

If a wrong partitioning is being used, you might see issues when trying to save settings as this requires some free space on the file system to actually save it and I don't get an error when saving as this error is detected when the cache finally flushes its data to the file system.