letscontrolit/ESPEasy #5252: ETH can't get IP from DHCP

mehdi-song, 2025-02-11:
**Describe the bug**
Hi
Sometimes ESP32-DevKit can't get IP from dhcp and ETH not start.

**To Reproduce**
Steps to reproduce the behavior:
in serial mode if ETH cable disconnected from w5500 switvh to wifi and after connect the cable shown as : 
20:56:30.002 : (217512) Info   : processEthernetConnected()
20:56:30.003 : (217512) Info   : WiFi : WifiDisconnect()
20:56:30.223 : (217944) Info   : WIFI : Disconnected! Reason: '(1) Unspecified'
20:56:30.224 : (217848) Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
20:56:30.325 : (217960) Info   : Set Network mode: Ethernet 

**Screenshots**

[image]

Port config : 

[image]

**Used platform (please complete the following information):**
 - ESP32
- Build: [ESP_Easy_mega_20241222_normal_ESP32_4M316k_ETH Dec 22 2024](https://github.com/letscontrolit/ESPEasy/releases)

**Platform Specifics (when applicable, please complete the following information):**
 - Flash size: [e.g. 4M]
 - Browser + OS [ Windows 10/FirFox]
- w5500 SPI
**Additional context**
Add any other context about the problem here.

TD-er, 2025-02-11:
What kind of board is this?
I find the choice of pins a bit strange.

See: https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp32
GPIO-2 does have a pull-down resistor and GPIO-15 has a pull-up resistor.

mehdi-song, 2025-02-12:
ESP32 DEVKITV1 
W5500 SPI -> https://www.aliexpress.com/item/1005007124700235.html

mehdi-song, 2025-02-17:
Thanks,  I will test "ESP32-classic" (or "M5Core") , but is that OK in the console show `INIT  : SPI Init (without CS)` ?

tonhuisman, 2025-02-17:
> but is that OK in the console show `INIT : SPI Init (without CS)` ?

It's just an initialization message, reminding the user that CS ('Chip Select') should be set in the Task config page (except for SD-card or SPI-Ethernet, that have their specific CS-pin settings on the Hardware page).

mehdi-song, 2025-02-21:
> OK, so you could change pins and check wiring. That's positive as it is quite possible this is a wiring issue (or at least signal integrity related). I have several W5500 devices here and those do work just fine. Not saying it can't be software related, but it is a bit more likely this is a matter of signal quality.
> 
> Could you test using the wiring as described here for an "ESP32-classic" (or "M5Core") https://espeasy.readthedocs.io/en/latest/Hardware/Hardware.html#spi-ethernet-esp32-boards
> 
> Best to pick GPIO pins which do not have a warning symbol next to its label.
> 
> Also make sure not to use long wires and also make sure the GND and 3V3 are not too thin as Ethernet may take quite some energy. (quite high frequency spikes of higher current) You could also add some capacitor close to the Ethernet board between 3V3 and GND. Something between 22uF and 100 uF. (make sure polarity is correct)
> 
> The SPI frequency is in the order of 10 MHz (not sure if it was 8 MHz or 20 MHz, have to check the code), so longer cables will cause issues.

I tested everything and not any chance to use w5500 with esp easy mega .
The wiring checked, and all settings were correct. 
Can you help me with what can I do?