letscontrolit/ESPEasy #5253: [BUG] Issue with NeoPixel in Christmass max release

niko-lay, 2025-02-13:
**Describe the bug**
I'm trying to connect 8x8 ws2812 matrix to esp32-s3 module as display. With Christmas build **20241222** and **max** firmware this  doesn't work, but works fine with previous  **20240822** **max** build. After enabling plugin P131, web interface became unresponsive (response time increases from ~500ms to 10-15s, sometimes no answer form board) and matrix doesn't show any text. I also tried to use different pin, this didn't helped

**To Reproduce**
Steps to reproduce the behavior:
1. connect board via usb to laptop
2. erase chip and flash fresh firmware 
```
esptool.py --port /dev/ttyACM0 erase_flash
esptool.py --port /dev/ttyACM0 write_flash 0x0 ESP_Easy_mega_20241222_max_ESP32s3_16M8M_LittleFS_OPI_PSRAM_CDC_ETH.factory.bin
```
3. make basic board config: wifi, clock and debug level. I did this via serial console, but same result if made configuration via web
```
WifiSSID,<SSID>
WifiKey,<PASS>
UseNTP,1
TimeZone,60
Debug,4
WifiConnect
```
4. Add 1 device with P131 Display - NeoPixel and make simple config as described on  screenshots below
5. check matrix state and  board logs 
```
00:02:36.697 : (253876) Info   : WD   : Uptime 3  ConnectFailures 0 FreeMem 254020 WiFiStatus: WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
00:02:36.698 : (253972) Debug More : UDP  : Send Sysinfo message
00:02:36.710 : (253920) Debug More : UDP  : Send Sysinfo message
00:02:38.045 : (251920) Info   : Skip saving task settings, not changed
00:02:38.058 : (251920) Info   : ESPEasy console using ESPEasySerial
00:02:38.079 : (251872) Info   : ESPEasy console fallback enabled
00:02:38.080 : (251888) Info   : SaveToFile: free stack: 6004
00:02:39.362 : (252424) Info   : FILE : Saved config.dat offset: 0 size: 3028
00:02:39.363 : (252424) Info   : SaveToFile: free stack after: 6004
00:02:39.373 : (252424) Info   : Skip saving SecuritySettings, not changed
00:02:39.383 : (252004) Info   : CRC  : ProvisioningSettings CRC   ...OK 
00:02:39.702 : (252424) Info   : static_file: /favicon.ico to: 192.168.0.164
00:02:39.704 : (252364) Debug  : HTML : Request file /favicon.ico
00:02:39.708 : (254028) Debug  : LoopStats: shortestLoop: 41 longestLoop: 5685431 avgLoopDuration: 86.00 loopCounterMax: 731707 loopCounterLast: 367864
00:02:39.708 : (254064) Debug  : Scheduler stats: (called/tasks/max_length/idle%) 367862/1808/6/75.04
00:03:03.334 : (249844) Info   : SaveToFile: CustomTaskSettings file: config.dat task: 1
00:03:03.335 : (249984) Info   : SaveToFile: free stack: 4836
00:03:04.347 : (250352) Info   : FILE : Saved config.dat offset: 33792 size: 28
00:03:04.348 : (250352) Info   : SaveToFile: free stack after: 4836
00:03:04.363 : (251280) Info   : SaveToFile: TaskSettings file: config.dat task: 1
00:03:04.364 : (251408) Info   : SaveToFile: free stack: 5796
00:03:05.421 : (251440) Info   : FILE : Saved config.dat offset: 32768 size: 536
00:03:05.422 : (251440) Info   : SaveToFile: free stack after: 5796
00:03:05.435 : (251488) Info   : ESPEasy console using ESPEasySerial
00:03:05.456 : (251488) Info   : ESPEasy console fallback enabled
00:03:05.459 : (251460) Info   : SaveToFile: free stack: 6004
00:03:06.888 : (251488) Info   : FILE : Saved config.dat offset: 0 size: 3028
00:03:06.889 : (250960) Info   : SaveToFile: free stack after: 6004
00:03:06.899 : (251488) Info   : Skip saving SecuritySettings, not changed
00:03:06.913 : (251088) Info   : CRC  : ProvisioningSettings CRC   ...OK 
00:03:06.926 : (251340) Info   : neomatrix: INIT, matrixType: 0b0
00:03:06.927 : (250884) Debug  : updateFontMetrics: size: 1, fg: ffff, bg: 0
00:03:06.938 : (250912) Info   : NEOMATRIX: Init start.
00:03:06.938 : (250256) Info   : NEOMATRIX: Init done, address: 0x3fcb5a24 valid, commands: neomatrix/neomatrixcmd, size: w:8, h:8
00:03:06.939 : (250212) Info   : AdaGFX_helper: GFX Init.
00:03:06.950 : (250036) Debug  : AdaGFX: Init, x: 8, y: 8, colors: 65535, trigger: neomatrix, Features: asciitable, lm/lmr, bmp, btn, win, getconf
00:03:06.971 : (250132) Debug  : AdaGFX: tr: neomatrix x: 8, y: 8, text columns: 1 rows: 0
00:03:06.972 : (250132) Debug  : AdaGFX: tr: neomatrix x: 8, y: 8, text columns: 1 rows: 0
00:03:06.983 : (243396) Debug  : updateFontMetrics: size: 1, fg: ffff, bg: 0, xp: 0, yp: 0
00:03:07.001 : (243396) Debug  : AdaGFX: parse result: 'hello'
00:03:07.003 : (243424) Debug  : AdaGFX: parse result: 'hello'
00:03:07.004 : (243396) Debug  : updateFontMetrics: size: 1, fg: ffff, bg: 0, xp: 0, yp: 0
00:03:07.015 : (243348) Debug  : printText: clear: _x:8, oTop:0, _y:0, xTx:8, yTx:0, wTx:6, hTx:8, oBot:0, _res_x/max:8/8, str:h
00:03:07.353 : (244424) Info   : static_file: /esp.css to: 192.168.0.164
00:03:07.355 : (244376) Debug  : HTML : Request file /esp.css
00:03:07.380 : (246104) Info   : WD   : Uptime 3  ConnectFailures 0 FreeMem 246248 WiFiStatus: WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
00:03:07.381 : (246024) Debug More : UDP  : Send Sysinfo message
[... 66 lines cut ...]
`

**Expected behavior**
Web interface remains responsible and ws2812 matrix shows `hello` text

**Screenshots**

> P131 basic  configuration which don't work with  ESP_Easy_mega_**20241222**_max_
> [image]

>P131 basic  configuration which works fine on prevision firmware verion
> [image]

**Used platform (please complete the following information):**
 - ESP type: ESP32-S3
 - Build version: ESP_Easy_mega_20241222_max_ESP32s3_16M8M_LittleFS_OPI_PSRAM_CDC_ETH 
 - Build set: max

**Platform Specifics (when applicable, please complete the following information):**
 - Flash size: 16M (N16R8 module)
 - Brand/Model: esp32-s3-wroom-1
 - Browser + OS: Firefox+Linux

**Additional context**
Add any other context about the problem here.

TD-er, 2025-02-13:
Hmm that's odd.
I do have an ESP32-S3 board here with 5x29 pixel running exactly this plugin.
Only difference is that my board is not having PSRAM and I'm using GPIO-42, not GPIO-3.
But I don't see how a different GPIO would make a difference here as on the S3 this pin isn't used by something like UART which might block it.

Do you need to have the ESPEasy console also present on the USB HWCDC port?
If so, can you maybe disable the serial port console or at least set the serial port to HW Serial0 and not "USB HWCDC" which is the default on a new installation?

Can you also enable "Collect Timing Statistics" (also on Tools->Advanced)?
After enabling it, you may let it run for a while and then open the page Tools->Timing Stats.

N.B. on every load of that page, the timing stats are reset.

I'm interested to see what takes the most time to execute on that ESP.

niko-lay, 2025-02-13:
I fully disabled Serial, but this didn't helped
[image]
Also changed pins - the same issue

and here is timings stats from my node

[image]

@TD-er ,  could you please guide me,  what else can I  try to debug?

niko-lay, 2025-02-13:
also, I've configured "P038 - Output - NeoPixel (Basic)" plugin and faced similar results

Here is plugin config 
[image]

and timing stats. Significant time in write function appears only after sending some command, i.e. "NeoPixelAll,25,10,20"

[image]

TD-er, 2025-02-13:
Well at least it is clear this has to do with the interaction with the addressable LEDs.
What else is configured on that board?

Can you test by lowering the amount of LEDs in the settings? Just to see if that makes a difference.

TD-er, 2025-02-13:
Hmm as comparison as how it should be....
My ESP32-S3 with the 29x5 pixel 'display'

[image]

TD-er, 2025-02-13:
Could you try running one from this (experimental) build?
https://github.com/letscontrolit/ESPEasy/actions/runs/13311055907

N.B. This is incomplete WiFi/network test code, so make sure you can flash back using USB when needed.
But it is the only build I have now with the latest Arduino 3.1.2 code, released today.

niko-lay, 2025-02-13:
> What else is configured on that board?

nothing, it's new s3 board for experiments

> Can you test by lowering the amount of LEDs in the settings? Just to see if that makes a difference.

You are right, reducing the number of LEDs made matrix working  and the processing time of the function "P131 Display - NeoPixel Matrix	TEN_PER_SECOND" became about 1ms. Plugin works in 7x8 and 8x7 configuration, but any variants larger than 8x8 are not workable,  I think 64 LED is some internal limit

TD-er, 2025-02-13:
I did find the 10 sec timeout code.
In `NeoEsp32RmtMethod_idf5.h`:
```c++
void Update(bool maintainBufferConsistency)
    {
        // AddLog(2,"..");
        // wait for not actively sending data
        // this will time out at 10 seconds, an arbitrarily long period of time
        // and do nothing if this happens

        if (ESP_OK == ESP_ERROR_CHECK_WITHOUT_ABORT(rmt_tx_wait_all_done(_channel.RmtChannelNumber, 10000 / portTICK_PERIOD_MS)))
        {
         ```

Now we have to check _why_ this code is triggered here.

TD-er, 2025-02-13:
Can you also try reducing the refreshrate while increasing the pixel count?

niko-lay, 2025-02-13:
> Could you try running one from this (experimental) build?
> https://github.com/letscontrolit/ESPEasy/actions/runs/13311055907

checked with the new firmware ESP_Easy_mega_20250213_max_ESP32s3_16M8M_LittleFS_OPI_PSRAM_CDC_ETH Feb 13 2025  the result is exactly the same, size 7x8 works, 8x8 and larger - not

> Can you also try reducing the refreshrate while increasing the pixel count?

do you mean "speed" field in line ? If so, already changed value from 1 to 20 and this didn't helped

I finally found the exact limit, I can light 63 LEDs in "P038 NeoPixel (Basic)" without any problems, but if I specify 64 or more LEDs number  in plugin settings - the matrix stops working and I see   10s WRITE in Timing Statistics.

I guess maybe there is some kind of overflow or buffer overrun going on?

TD-er, 2025-02-13:
I think I may have found the cause here...
Will make a PR in a few minutes and I can make a 'max' build for you now to test.

TD-er, 2025-02-13:
Can you test [this build](https://www.dropbox.com/scl/fi/xx9lesx24nf6265hubsmu/ESP_Easy_mega_20250213_max_ESP32s3_16M8M_LittleFS_OPI_PSRAM_CDC_ETH.zip?rlkey=p6jo8z1aqv2zfwhsn3dwbf2u6&dl=0)?
The zip file contains both the 'factory' as well as the normal bin file

TD-er, 2025-02-13:
The idea behind this change is as follows...

For builds with "OPI PSRAM", the entire sketch is uploaded from flash to PSRAM at boot as PSRAM is way faster than flash. (about 3x faster)
This is also why the reported PSRAM size is about 5 MB while you have 8 MB PSRAM.

Another side-effect is that there is another threshold for allocating memory to internal or PSRAM based on size.
So starting from 64 LEDs, the `malloc` switches to use PSRAM instead of internal RAM.

Problem is that we can't use DMA from PSRAM while we have read- and write-access to it.
So the change in this test build is to force allocating memory on internal RAM and use a flag to make sure it is DMA accessible.

I also reduced the timeout to 30 msec, which should be enough to address roughly 700-ish LEDs in a single call.
Maybe I should set it to 100 msec, which is also enough for not really blocking the ESP, but should still not happen anyway.

TD-er, 2025-02-13:
Pull Request for this: https://github.com/letscontrolit/ESPEasy/pull/5254

GH Actions test build: https://github.com/letscontrolit/ESPEasy/actions/runs/13318525379

niko-lay, 2025-02-13:
> Can you test [this build](https://www.dropbox.com/scl/fi/xx9lesx24nf6265hubsmu/ESP_Easy_mega_20250213_max_ESP32s3_16M8M_LittleFS_OPI_PSRAM_CDC_ETH.zip?rlkey=p6jo8z1aqv2zfwhsn3dwbf2u6&dl=0)? The zip file contains both the 'factory' as well as the normal bin file

I've flashed factory.bin from your archive and  don't see long timings in  Timing stats, but looks like LED control stop working. Matrix is fully black, for both "Display - NeoPixel Matrix" and  "Output - NeoPixel (Basic)" plugins. 
Unfortunately I don't have an oscilloscope to check actual signals on the  outputs

TD-er, 2025-02-13:
I have it running on my setup here using a 4M ESP32S3 with 2 MB PSRAM.

Can you try [this build](https://www.dropbox.com/scl/fi/j0x93ro73hkua55o1nn9e/ESP_Easy_mega_20250214_max_ESP32s3_16M8M_LittleFS_OPI_PSRAM_CDC_ETH.zip?rlkey=59muu9ip4f3fms9rm1jgb4zxt&dl=0) ?

TD-er, 2025-02-14:
I tried with my scope connected and I can reproduce it on my ESP32-S3 with 16M flash and 8M PSRAM.

So I'll try tomorrow to see what I can do here.

TD-er, 2025-02-14:
Here with 63 pixels, as you can see, the number of pulses is 1000 and more importantly it is _finite_....

[image]

Here with 64 pixels...
It will not end

[image]

TD-er, 2025-02-14:
Can you try [this build](https://www.dropbox.com/scl/fi/jirnw1fai4bd3a9gjyl7i/ESP_Easy_mega_20250214_max_ESP32s3_16M8M_LittleFS_OPI_PSRAM_CDC_ETH_fix.zip?rlkey=ke1pqd4xcze9ep3e33mzppx7d&dl=0) ?

My scope suggests it has been fixed... but it was extremely complex as I had to change my complete build env and change some SDK compile time flags to move functions to IRAM, driver objects to RAM and enable serving interrupts when cache has been disabled...

niko-lay, 2025-02-14:
> Can you try [this build](https://www.dropbox.com/scl/fi/jirnw1fai4bd3a9gjyl7i/ESP_Easy_mega_20250214_max_ESP32s3_16M8M_LittleFS_OPI_PSRAM_CDC_ETH_fix.zip?rlkey=ke1pqd4xcze9ep3e33mzppx7d&dl=0) ?

thanks, I'll  check this build little bit later today

niko-lay, 2025-02-14:
This version works fine with the 8x8 matrix

TD-er, 2025-02-14:
OK thanks for testing.
Now I 'only' have to fix the build scripts I had to break to make this work for the ESP32-S3 with OPI flash.

niko-lay, 2025-02-15:
I would like to clarify one point, is this issue related only to OPI variants of RAM or any board with external RAM could  also be affected?  If so, probably it significant change for a ESPEasy  and all  plugins should be re-tested

TD-er, 2025-02-15:
The issue is with ESP boards with enough PSRAM to hold the sketch.
Right now this is limited to the ESP32-S3 with OPI PSRAM.
However I will apply the same SDK patches to other builds as well as it allows for a more stable handling of RMT for controlling addressable LEDs.
The change in SDK parameters does allow for handling handling serving interrupts when cache lines are blocked, which can also occur in other situations.
I do need to investigate if this may also break some multi-thread code, so I will have to write some tests to make sure I understand its impact.
As soon as I fully understand the consequences of this, I will 'port' it to all other ESP32-variants.

PR #5254 (2025-02-16): [NeoPixel] Fix issue running addressable LEDs on ESP32S3 with OPI PSRAM
Fixes: #5253

For builds with "OPI PSRAM", the entire sketch is uploaded from flash to PSRAM at boot as PSRAM is way faster than flash. (about 3x faster) This is also why the reported PSRAM size is about 5 MB while you have 8 MB PSRAM.

Another side-effect is that there is another threshold for allocating memory to internal or PSRAM based on size. So starting from 64 LEDs, the malloc switches to use PSRAM instead of internal RAM.

Problem is that we can't use DMA from PSRAM while we have read- and write-access to it. So th [... cut]
commit 6e386150 (2025-02-16): [NeoPixel] Fix issue running addressable LEDs on ESP32S3 with OPI PSRAM

Fixes: #5253

For builds with "OPI PSRAM", the entire sketch is uploaded from flash to PSRAM at boot as PSRAM is way faster than flash. (about 3x faster)
This is also why the reported PSRAM size is about 5 MB while you have 8 MB PSRAM.

Another side-effect is that there is another threshold for allocating memory to internal or PSRAM based on size.
So starting from 64 LEDs, the malloc switches to use PSRAM instead of internal RAM.

Problem is that we can't use DMA from PSRAM while we have read- and write-access to it.
So t [... cut]
closed 2025-02-16 by TD-er