letscontrolit/ESPEasy #5255: [BUG] Error accsing files on SD card ( Easy_mega_20250209)

dosipod, 2025-02-14:
Hello, 

In the example st77xx,bmp,xpic01.bmp works fine from fs (  showBmp: Done ) but the same pic copied to sd
and using st77xx,bmp,2,2,xpic01-sd.bmp does not work ( showBmp: File signature error  ).If the file name 
is wrong then the error is diffrent (showBmp: file not found) .
The same issue for other files as you can not download or view any of them on SD , the log.txt when downloaded
from sd shows zero byte size . 

This is very close to the case opened here https://github.com/letscontrolit/ESPEasy/issues/4769  and all I can think of is the size of the sd as it is (KIOXIA Exceria MicroSD 32GB ) but  I do not have older sd with 4GB to test but I have a lot of them used with HA , cams ..etc  and clearly 
ESPEasy  is writing to the sd but not reading it correctly . 

This was tested on latest release and also on gihthub run (Build: ESP_Easy_mega_20250209_display_ESP32_4M316k Feb 9 2025) with multiple collections .We are testing multiple other OSS with students  so have no real interest in the outcome for practical uses and we will be happy to provide more info or do extreme testing  but please limit the info requests to single post as I am not a fan of the chit shat hence the attached zip 
file is having all the info we  could think of including an example pic  for others to reproduce the issues 

[read_pic_from_sd_issue.zip](https://github.com/user-attachments/files/18795187/read_pic_from_sd_issue.zip)

Thank you  , 
ALDIY

TD-er, 2025-02-14:
What resolution does your display have?
The BMP you want to show should not exceed the boundaries of your display.

Have you tried to save the bmp file to the local file system on the flash (thus not SD card) to test if the image is in the right format?

I did save the file again using Irfanview, just to make sure it isn't an encoding issue.
Can you try this file?

[test01.zip](https://github.com/user-attachments/files/18795370/test01.zip)

dosipod, 2025-02-14:
I thought it was clear ( There is more details in the zip file as we did really take our time to test  ) , the same BMP   or files works when accessed from FS , so the only issue is accessing ANY files on SD which is the only  bit if you can test then that would help .

And sure the same result with the file you give as it still works on FS but not on SD , you do not have an SD to test with?
 showBmp: File signature error is telling there is an issue when trying to read from SD so Irfanview has no impact in this case but will
keep that in mind for other testing .

[image]

TD-er, 2025-02-14:
I don't have a setup right now to test SD cards.
Could you try to format the card using the official SDcard.org formatter?
https://www.sdcard.org/downloads/formatter/
And how are the files being written to the card?
Do you upload via ESPEasy or are the files uploaded using a computer and then insert the card in your ESP board?

dosipod, 2025-02-14:
I have tested both upload via ESPEasy  and files uploaded using a computer  but still the same error on SD card .The same file upload via ESPEasy to the FS works fine .

Now also tested https://www.sdcard.org/downloads/formatter/ and the same result with upload via ESPEasy  and files uploaded using a computer after formatting the card . 

I like espeasy way of dealing with  flies as it will tell you if the file is not there but it is still access error on sd (showBmp: File signature error )

[image]

TD-er, 2025-02-14:
I think the SD card can only be accessed to the first N sectors. (upto 2GB in whatever sector size the old standard had, I think 512 bytes)
That would explain why the file system table can be read, but not the files on it.

Could you try to partition the card using fdisk? Preferrably using Linux.
And then maybe try to format the card in a camera after partitioning it.
Maybe we should make some image of an existing 2 GB old card so it can be written using tools like `dd` under Linux.

dosipod, 2025-02-14:
I may have access to mac shortly to test multiple scenarios with fdisk if we are aware what standard espeasy is after .Or at least 
where in code is that so we could test .

I am still digging for old < 4GB cards to test with but may have to buy some or seek phone repair shops which will delay testing 

Meanwhile if you or others could  kindly try to  confirm SD card access is working on any setup ( just to access text files ) then that
might be good news

tonhuisman, 2025-02-14:
I will be able to test later today (CET timezone, in the evening).
I have some suspicions about what's going on here, but most likely the 2GB max size (with 512 byte sector size) is the major culprit here.

dosipod, 2025-02-14:
Thank you , I have also tested on a smaller 8GB card but still faced the issue , not sure I will find <2GB cards but will try few tricks with  fdisk to force it to be <2GB.

I was also looking at AdafruitGFX_helper.cpp as that is the source of the signature error but try to not focus on BMP but rather on SD file access in general as it seems not limited to BMP but I do not get clear error when trying to access text files as with BMP but the text file download with zero byte so might be more of a logic rather then the error with BMP.

Here the upload to SD is fine but will see if the logging or debug flags could provide more info on access but we are not familiar with the code base but should be good exercise 
[image]

TD-er, 2025-02-14:
If your Mac allows for using the old fashioned fdisk to make a FAT-16 partition on the first N sectors covering only like 1 GB of area (just being safe not to hit any corner case sector count) then that would also be a good idea.

Or maybe you have some camera from before SD-HC support became popular.
If I'm not mistaken, my old Canon compact camera (A570 ??) from 2007 was one of the first to have SD-HC support.

TD-er, 2025-02-14:
Hmm found this site regarding the good old MSX. (I had one in 1983...)

Anyway, they have a zip file with bootsector + file system area of various formatted SD cards. Just like we need here.
https://www.msx.org/downloads/fat16-disk-images-for-easy-unzip-blow-to-sd-card

After writing such an image to the start of the SD card, you probably need to re-insert the card.
No idea if you also need to format it.

dosipod, 2025-02-14:
As I still not having access to mac but will try that tonight among other things  .

For now I tried with Diskpart  and format it as FAT ( that is FAT-16 ) but unfortunately it is the same case but also BMP files uploaded to SD card are not with the correct size so that might be worst ( also Irfanview  file you sent is truncated on FAT-16) 

[image]

Could be a tall order but is there a way even a hack to initial the sd from the esp to make sure it is formatted correctly

TD-er, 2025-02-14:
Maybe you could make a really small BMP file and see if that will work.
So keep it below 3k

dosipod, 2025-02-14:
Yeah I tried that before and the same small bmp files works from FS but not from SD .
We will try multiple other tests and we will  also try with multiple text files as smaller files of few bytes do not get downloaded correctly

TD-er, 2025-02-14:
Can you also measure the supplied voltage to the SD card?
Maybe it isn't stable enough?

tonhuisman, 2025-02-15:
@dosipod I've investigated the issue, and it seems something has changed in the latest Arduino libraries, causing some file-system based operations to a) not work as expected, and b) work differently for files on internal flash vs. on SD-Card 😱 
I've tried to implement some workarounds and fixes. Can you please test a build from [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/13348510902) (all my tests where positive, using both your .bmp files and other files, uploaded to the SD-Card.

NB: I've used a 1GB image as posted above, so it actually behaves like a < 2GB SD card! (Actual SD card is 32 GB). There's currently no support for `FAT32` or `exFat` file-systems in ESPEasy.

dosipod, 2025-02-16:
@tonhuisman You are a prince , both text files access and bmp seems to be fine on my unit now after your PR just in time really as we were  really testing a lot and about to report the testing result from another esp32 ( using  cheap yellow display )  to rule out any  HW concerns @TD-er you may be having  .

You may close this at your discretion but will soon try to post the detailed result ( specifically testing with different formatted SDs)   either in here or in the fix PR  so your call 
Thank you ,

tonhuisman, 2025-02-16:
> You may close this at your discretion

This issue will be auto-closed when the PR is merged, so no action needed here 👍 

Thanks for your confirmation, looking forward to your findings with the different SD-card formats.

closed 2025-02-16 by TD-er

TD-er, 2025-02-16:
As far as I know, the current used libraries do not have support for FAT32 or exFAT.
So that means you're limited to FAT16, which is limited to a max. of 2 GByte in size.
It may be the vFAT extension (introduced by Windows '95) is working, so you could use filenames with names longer than 8.3 may work.

So unless you can actually use partitions > 2 GByte, I don't think there currently is FAT32 support.

It is however good to know that it is possible to use SD-HC cards as the availability of classic SD cards is now next to non-existent.
I don't know whether SD-HC cards also may have a compatibility-mode. If they do and it is a requirement using the current library implementations, then that might also show up when you try to use partitions over 2 GByte, or partitions starting after the 2 GByte mark.

dosipod, 2025-02-17:
@TD-er I do not see such limtions yet and trust me ESPEasy way of dealing with SD is way better then you may think at least when compared to other OSS. In the below test after updating to latest commit , the unit is now with 
Build: ESP_Easy_mega_20250216_display_ESP32_4M316k Feb 16 2025

I replaced the SD card with 32GB one and dumped around 4 GB of files and I can still access the BMP files , upload , download
so could be the  **used libraries**  got updated if you point to source we could dig for that and test outside ESPEasy   but I will take your word for it as 2GB is plenty of space to work with but my aim is to avoid imposing a restriction so SD cards feature could be used in a simple way without the need for too many steps.
FAT16 did not unfortunately work in the early test even when the formatted SD is less then 2GB but if you tested that then could be different standard used and we are not on the same page with that but it is all good news .

[image]

@tonhuisman This is very stable ,very minor thing is when using folders ( access still works with / added ) .
Below in  two examples "/" is missing from ui  if you click the file . That is not a showstopper for automation and scripting 

[image]

Since this is merged and issue closed we will stop posting  result of small tests here but will try to gather a lot of testing results and open other cases  if we find anything worth reporting . 

Have a nice day

TD-er, 2025-02-17:
If you have any informative updates like "Yay, FAT32 seems to work just fine" or something else, you're always welcome to post them here.

Looking into the folders issue is a good idea anyway.
This is now also possible to use on LittleFS, but so far we have not yet spent time on it as it wasn't compatible with SPIFFS and thus would complicate things when having different features for different builds.
But now that SD-support has matured, it makes sense to take some time to look into this a bit more.

TD-er, 2025-02-17:
By the way, I will be really surprised if SDXC cards will work or will work reliable when using a FAT32 file system of over 32 GB.
I know back in the days when the first 40 GB hard drives came out, that 32 GB limit was causing lots of issues as a lot of tools (and even Windows) were not able to manage it.

PR #5257 (2025-02-16): [SD-card] Bugfix: Reading a bmp file from SD-Card didn't work as expected
Resolves #5255 

Features:
- Fix displaying .bmp files on a TFT display when read from SD-Card
- Implement workaround for `file.position()` not returning for an SD-located file 😮
- Tested:
  - Displaying .bmp files from local storage on TFT display
  - Loading .bmp files from SD-Card and display them on TFT display
  - Uploading files to local Flash storage
  - Uploading files to SD-Card storage

TODO:
- [x] More testing ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/5255#issuecomment-2661330520))
- [x] Fin [... cut]