letscontrolit/ESPEasy #5256: [BUG]ILI9341 display error

hrikken, 2025-02-14:
**Describe the bug**
When text is written on the first line (via the plugin)the topmost pixelrow is always set to the background color (e.g. blacked-out). This renders the characters in the first line distorted. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'ILI93xx' display plugin
2. Settings:
- the display text print mode setting is "Clear then truncate exceeding message"
- rotation 270 
- display model ILI9341 240x320 px
- font: default scaling 2
- Forgr/Back: white / black
- Refresh: 1 sec
3. Line 1 content: %systime% 
4. See error

**Expected behavior**
Display the time on the first line and refresh every second

**Error**
The topmost pixelrow is set in background color (or cleared?).

**Used platform :**
 - ESP type: ESP32
 - Build version: [ESP_Easy_mega_20250203_climate_ESP32_4M316k Feb 3 2025]
 - Build set: climate

**Platform Specifics (when applicable, please complete the following information):**
 - Flash size: 4M]

tonhuisman, 2025-02-14:
Does this also happen if font scaling is set to 1? (Some fonts have their height setting either 0 or have an off-by-1 error)

tonhuisman, 2025-02-16:
@hrikken 
To quote myself:

> NB: This behavior _is_ dependent on the 'Background-fill for text' setting.

Have you tested with that setting enabled and disabled?

I've just tried to reproduce your report (used a very similiar ILI9342 in an M5Stack unit, with command `tft,txl,1,%systime%`), but can't duplicate that behavior on the top-line. I can however see the background color untouched when clearing the screen to blue, and using the above command and then command `tft,txl,2,%systime%`, where it leaves some blue background between the first and second line, when setting "Background-fill for text" is turned _off_. This is the **intended** behavior, and the reason this setting exists! In most situations, you would want to have that setting checked.

The fact it is or isn't changing the top-most line of the display often depends on the manufacturer of the display, it seems, as some cover (or disconnect...) the top-most line of the display, and others don't. The effect can be simulated by using line 1 and 2 with the `txl` sub-command, as demonstrated.

If you want to print text 'on top' of an image, with the image shining through, you should use the `txtfull` sub-command, and specify the background-color the same as the foreground color, resulting in a transparent background (this is by design behavior in the underlying Adafruit GFX library).