letscontrolit/ESPEasy #5265: [BUG]Connection problems with MQQT HiveMQ / TLS problems

hrikken, 2025-02-23:
**Describe the bug**
When I try to connect to HiveMQ MQTT server the ESPEASY cannot connect

**To Reproduce**
Steps to reproduce the behavior:
1. Create an instance (free cluster) on HiveMQ
2. Create a MQTT user with valid credentials 
3. Look up the hostname of the HiveMQ cluster
4. Use the credentials and hostname to create a MQTT controller (OpenHAB) in ESPEasy (with TLS on port 8883)

**Expected behavior**
ESPEasy should create a connection the the HoveMQ MQTT broker but instead it throws an error:
1506427: MQTT : Intentional reconnect
1516438: MQTT : Broker C005 connection failed (49/0)
1516439: MQTT : TLS error code: -1002

**Screenshots**
See above 

**Used platform (please complete the following information):**
 - ESP type: ESP32 M5Stack Atom with BMP280 and SHT40 via I2C
 - Build version: [ESP_Easy_mega_20241222_climate_ESP32_4M316k Dec 22 2024]

**Additional context**
When I use the M5STACK UIflow2 dev. environment (MicroPython) I can connect without any problems.
I've tried a server instance on EMQX Cloud service and the problem there appears to be exactly the same.

TD-er, 2025-02-23:
Is your MQTT broker on the same local subnet?
If not, can you set the timeout to 1000 msec for this controller and try again?

I can't yet find the exact error code, so I don't yet know where to look.

There are so many things go wrong with making a secure connection.

TD-er, 2025-02-23:
Ah found the negative error codes:
```c++
#define ERR_OOM             -1000
#define ERR_CANT_RESOLVE_IP -1001
#define ERR_TCP_CONNECT     -1002
// #define ERR_MISSING_EC_KEY  -1003   // deprecated, AWS IoT is not called if the private key is not present
#define ERR_MISSING_CA      -1004
```
So it seems there are connection issues, so a too low timeout is a good option to start testing.

hrikken, 2025-02-23:
Ok thanks. I'll test it asap.

hrikken, 2025-02-23:
> Is your MQTT broker on the same local subnet? If not, can you set the timeout to 1000 msec for this controller and try again?
> 
> I can't yet find the exact error code, so I don't yet know where to look.
> 
> There are so many things go wrong with making a secure connection.

Both brokers are on the public internet.

hrikken, 2025-02-24:
Just tested it and the problem stays the same. 
If it would help you I could you provide you with logins on both brokers. 
Let me know if you want that and how I can send you the data privately.

TD-er, 2025-02-24:
Yes, I think that's the more sensible next step.
You know my email address.

hrikken, 2025-02-24:
I've sent you an email with the details, Gijs.
No hurry by the way.

Good luck.

TD-er, 2025-02-24:
Just as an update for whoever may end up here via search...
When TLS error is "-1002", the ESP may not be able to reach the host at all.

Steps to check:
- Increase timeout to 1000 msec.
- Check gateway/DNS settings
- Disable ESPEasy p2p UDP port (set to 0), enable NTP and power cycle the ESP node. If ESPEasy can get a system time this way, you know it can reach hosts on the internet. If not, look into what is wrong.

closed 2025-02-24 by TD-er