letscontrolit/ESPEasy #5273: [FR] HSPI and VSPI at the same time

dosipod, 2025-03-04:
Hi , I am testing with CYD ( esp32-2432s028r ) and the way it works is the tft on HSPI and SD on VSPI .I was testing that in tasmota and I recall both display and SD  worked  ( in tasmota SSPI & SPI can both be used at the same time  )  .

In ESPEasy we had the display working fine along with the SD  by first enabling HSPI and then rebooting ( without removing power ) and then enabling VSPI and rebooting again ( also without removing power ) .
This is working fine but as you can guess the issue arise when there power  lose as we have to add HSPI or VSPI and reboot.

Is there a way to enable both HSPI and VSPI at the same time via ui or compile flag ? Any workaround even if dirty is also okay 

[image]

tonhuisman, 2025-03-04:
This is a feature that's on the (rather) short-term TODO list, but haven't started it yet. First some other features to finish 😃

TD-er, 2025-03-04:
To explain a bit more...
Ton is working on adding multiple I2C busses and when that's done, there will be a similar option to define multiple SPI busses.

dosipod, 2025-08-19:
@tonhuisman 
Hello , I am taking a break from OSS and our team are no longer there but as a courtesy  I thought I will test one last time .

No backups are needed so I just serial  flashed the unit  as you see in the pic in step A .
Once the unit is up I tried to update it  with zip file you attached using  factory.bin  in step B which did not work as expected 

I then tried the ESP_Easy_mega_20250817_display_ESP32_4M316k_LittleFS_ETH.bin and it updated but there seems to be RTCWDT_RTC_RESET and  a boot loop as you can see in step C .

As mentioned I am no longer involved on GitHub and may not be of further help in testing as I am out of touch with all of this  . Thank you for reaching out though and wish you all the best 

[image]

tonhuisman, 2025-08-19:
> I will try my best to finish this case with you

That's great, thank you very much!

In the meantime I've found the culprit, I'd used `SPI1` for the 'extra' SPI object, but while that works fine for ESP32-S3 and ESP32-C3, it doesn't for ESP32-Classic 😲, so I renamed it to `SPIe` (with an e for ESPEasy 😛)
Here's a corrected (and tested) ESP32 `Display` build:

[ESP_Easy_mega_20250819_display_ESP32_4M316k_LittleFS_ETH.zip](https://github.com/user-attachments/files/21867647/ESP_Easy_mega_20250819_display_ESP32_4M316k_LittleFS_ETH.zip)

dosipod, 2025-08-19:
@tonhuisman 

I was able to update the esp with the new bin and even found an old  backup  and display worked as expected . The SD however did not work as you can not use SPI Bus 0 and SPI Bus 1 at the same time . 

On the hardware page once you enabled  SPI Bus 0 and SPI Bus 1 ( I tried all combos ) and press submit it will
save but if you press reboot on the tool page ( or via hardware-reset button ) then SPI Bus 1 will go back to disabled .

The below log might be useful as after hardware-reset  I can see an error related to SD ( it is saying  SPI Bus 0 Init without CS which might not be accurate ) 

167: ^^^INIT : Booting version: ESP_Easy_mega_20250819_display_ESP32_4M316k_LittleFS_ETH, (Self built) feature/SPI-add-support-for-m
178: INIT : Free RAM:275164
189: INIT : Cold Boot - Restart Reason: CPU0: Vbat power on reset CPU1: for APP CPU, reset by PRO CPU
200: FS : Mounting...
272: FS : LittleFS mount successful, used 192512 bytes of 323584
344: ESPEasy console using ESPEasySerial
394: CRC : SecuritySettings CRC...OK
434: Current Time Zone: STD time start: 1970-10-25 03:00:00 offset: 180 min
447: CPU Frequency: 240
449: INIT : Check for Priority tasks
452: INIT : SPI Bus 0 Init (without CS)
1453: SD : Init failed 

Another thing I noticed is if you select  SPI Bus 1 and set that to disabled ( there is an error on the hardware page saying SPI bus 1 pins not configured correctly! ) 

[image]

Again please ignore all of that if you could select both SPI Bus 0 and 1 at the same time and the setting is retained after sw or hw reboot from your side  as this CYD is about $2 and a bit funky ( both display and SD  worked  though in tasmota and openHASP ) 

I could also try on a bare esp32 to select both SPI Bus 0 and 1 and see if that would save but as that is not wired to a display or SD , i am not sure if that would be useful thing to try just to confirm that the save is okay , please let me know if that would be of help .

closed 2026-02-10 by TD-er