letscontrolit/ESPEasy #5275: [question] Is there information for compiling one target from command-line with pio/platformio tools?

gitthnx, 2025-03-06:
Hello and thanks for all Your supportive work and time, 

with starting at
`git clone --depth 1  https://github.com/letscontrolit/ESPEasy`  
  
and having a functional platformio environment for compiling (tested with `tools/build_ESPeasy.sh`), i could not find a description for the workflow for compiling only one target board from command-line.
  
Please provide a short example for compiling a esp32-s2 or esp32-s3 board with `pio`/`platformio` commands from command-line (within a git cloned repository) and maybe details of editing of `platformio.ini` (or other .ini files?), if necessary for the defined target, e.g. `default_envs = custom_ESP32_4M316k`. 
There's information for [`VSCode`-gui](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html#starter-guide-for-local-development-on-espeasy), but not for command-line only.

Maybe You can add a short information about how to include a single plugin to a custom build for a target board, from command-line. (thx)

tonhuisman, 2025-03-06:
We have a quite extensive [Developers guide](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html) in our ReadTheDocs.
Documentation for building from the command-line isn't actively maintained, or endorsed, ATM.

TD-er, 2025-03-06:
When you look through the code of the bash shell command, you can see there is a call to `platformio run -e <env>`
Please note this is done with the Python virtual env activated.

You can create a `Custom.h` (mind the capital 'C') based on the given sample in which you explicitly state the needed plugins/controllers and even default WiFi settings.

When building a 'custom' env, this `Custom.h` file is used if it is found.

gitthnx, 2025-03-07:
Thanks for the responses,

[VS Code](https://code.visualstudio.com/docs/setup/setup-overview), if available for a compile/build platform (x86_64, aarch64) is an option with installing the plugins for 'esp32|esp8266' and platformio ('the alien looking icon') and there are step-by-step introductions there, meanwhile. 

From command line some steps are more clear and without additional adjusting to a (maybe unknown) ide/gui. 
Therefore tested with ('platformio.ini' not modified):
  
  `taskset -c 0-1 platformio run -e custom_ESP32_4M316k` (compiles, x86_64, from current 03/2025 git sources)
  `taskset -c 0-1 platformio debug -e custom_ESP32_4M316k`
(while `platformio run -e custom_ESP32_4M316k` stops here (x86_64, Ubuntu 22.04.5) 
with 'CalledProcessError: Command '['/usr/bin/python3', '-m', 'pip', 'list', '--format=json', '--disable-pip-version-check']' returned non-zero exit status 2.:'
but `taskset -c 0-1 platformio run -e custom_ESP32_4M316k` 
and `pio run -e custom_ESP32_4M316k` compile sources to binaries.)

maybe helpful other commands:
  `pio device list`
  `pio upgrade` (upgrades platformio tool)
  `pio pkg update`
  `pio pkg search -p 1 espressif32` 

( Custom.h not yet implemented. (wip) ) (thx)

TD-er, 2025-03-07:
No you should not add those to your global pip install.
That's why I created a Python virtual env in the setup script.
You really should use that.

And you don't need to install espressif32 packages, as those are included in the `platform` and `platform_packages`.
By installing separate packages like this, you will likely create incorrect dependencies which may even result in seemingly successful builds, but will crash when running on the ESP... as in been there done that.

You can also look into the Vagrant build option which is also present in the repo.
This way you won't mess up anything on your system.

Also there are problems with PlatformIO 6.1.17, so that's why we have set it fixed to 6.1.16 in the `requirements.txt`

So please follow the steps in the build script, which essentially does things like this:
- [optional] Fetch the specified code from git (e.g. a specific pull request)
- create virtual env if not available
- activate virtual env
- install required pip packages using `pip install -r requirements.txt`
- Search the platformio*.ini files for available envs to build and build all of them using `platformio run -e <env>`

The only thing you need to do is adapt the lines where this run command is being processed.

Please do not mess up your system installing all kinds of Python packages which might break other things on your system due to incompatible package versions.
Rather use Python virtual env as that's what it is for.

gitthnx, 2025-03-07:
Thanks for the update,

it seems the path for python3 dependencies is still related to 'virtualenv' environment
for testing with python3:
```
python3
```
```
>>> import sys
>>> print(sys.executable)
/root/.espressif/python_env/idf5.5_py3.10_env/bin/python3
>>> quit()
```
or 

```
pip3 -V
```
```  
pip 25.0.1 from /root/.espressif/python_env/idf5.5_py3.10_env/lib/python3.10/site-packages/pip (python 3.10)
```

(thx)

TD-er, 2025-03-07:
Hmm that's odd, why should that be installed in the `/root/` folder?
I never used those espressif Python packages, but having the virtual env installed in the `/root/` folder doesn't look very promising.

uzi18, 2025-03-07:
Do not build anything as root, 
please use regular user instead.

uzi18, 2025-03-07:
@gitthnx you can use even browser to build target via gitpod or Codespace (cloud). 
Download bin and upload to your board. 
By the way this plugin could be incompatible with current ESPEasy plugins API.

gitthnx, 2025-03-07:
Thanks for all responses and information updates.

Yes, that's a bad habit from other sources requiring root access for hardware monitoring/firmware update (within the same terminal tab). 

For following steps everything compiles on regular user permissions with previously advised commands:
`git clone --depth 1 https://github.com/letscontrolit/espeasy EspEasy`
`cd EspEasy`
```
virtualenv -p python3.10  venv/python3.10/
source venv/python3.10/bin/activate
```

`taskset -c 0-1 platformio run -e custom_ESP32_4M316k`

and virtualenv path is now:
`/dev/shm/EspEasy/venv/python3.10/bin/python3`

Since i have to read and learn about the plugins, it will take some time for verifying if it's possible to build and how to implement ( being a tab within EspEasy or a device plugin?, maybe even with a tab being another tool like a wireless [Oscilloscope](https://github.com/BojanJurca/Esp32_oscilloscope/releases)? ).

( Custom.h not yet implemented. (wip) ) (thx)

uzi18, 2025-03-07:
Usually have used best set for me. 
You can use just normal build.
Maybe you need to define USES_P125, 
but this could be hardcoded in platformio.ini (for build/test time) 

How you want to use this plugin?

tonhuisman, 2025-03-07:
> Since i have to read and learn about the plugins, it will take some time for to verify, if it's possible to build and how to implement ( being a tab within EspEasy or a device plugin?, maybe even with a tab being another tool like a wireless [Oscilloscope](https://github.com/BojanJurca/Esp32_oscilloscope/releases)? ).

Have you read the PlatformIO/Development Guide pages? As to me it seems you mostly googled/AI-prompted stuff together, instead of reading what's already documented... 🤔 
Not sure why you can't use VSCode/PlatformIO for development though, as 99.9% of people are actually using an operating system with a GUI (Windows, Mac, Linux), these days.

The plugin from ESPEasyPluginPlayground you selected (P125) hasn't been updated for a long time, and isn't up to par with current coding standards (== requirements) for ESPEasy. To be able to use that with current `mega` code-base, a lot of changes (fixes) are needed to successfully compile it. And plugin ID 125 is already in use in the mega branch, so a new plugin ID is required (suggested ID: 179, the next available unused ID, request it in #3839 please)
NB: If the migration process seems to daunting, I can do that, if needed.

gitthnx, 2025-03-07:
> Have you read the PlatformIO/Development Guide pages I linked above? As to me it seems you mostly googled/AI-prompted stuff together, instead of reading what's already documented... 🤔 Not sure why you can't use VSCode/PlatformIO for development though, as 99.9% of people are actually using an operating system with a GUI (Windows, Mac, Linux), these days.

Sorry, i overlooked the part of 'Make a custom build using PlatformIO' and thought it's all mostly about building with VSCode and not from command line. 
Yes, nothings wrong about an excellent GUI (like VSCode, it did compile all individual selected targets fine, locally), but compiling from remote on a headless server is mostly easier without a GUI. 

> The plugin from ESPEasyPluginPlayground you selected (P125) hasn't been updated for a long time, and isn't up to par with current coding standards (== requirements) for ESPEasy. To be able to use [...]

Thanks for the information on that. Seems these might be interesting additions and tools? (thx)

gitthnx, 2025-03-07:
> Hmm that's odd, why should that be installed in the `/root/` folder? I never used those espressif Python packages, but having the virtual env installed in the `/root/` folder doesn't look very promising.

maybe, because i did a build task for 'esp-idf' within the same terminal tab (and kept root login, because of a former hw related requirement)
Thanks for the pointer. (thx)

gitthnx, 2025-03-07:
> [@gitthnx](https://github.com/gitthnx) you can use even browser to build target via gitpod or Codespace (cloud). Download bin and upload to your board. By the way this plugin could be incompatible with current ESPEasy plugins API.

Thanks for the advises. (thx)

gitthnx, 2025-03-07:
> Have you read the PlatformIO/Development Guide pages I linked above? As to me it seems you mostly googled/AI-prompted stuff together, instead of reading what's already documented... 🤔 Not sure why [...]

Sorry, i overlooked the part of '[Make a custom build using PlatformIO](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html#make-a-custom-build-using-platformio)' and thought it's all mostly about building with VSCode and not from command line. 
Yes, nothings wrong about an excellent GUI (like VSCode, it did compile all individually selected targets fine, locally), but compiling from remote on a headless server is mostly easier without a GUI. 

> The plugin from ESPEasyPluginPlayground you selected (P125) hasn't been updated for a long time, and isn't up to par with current coding standards (== requirements) for ESPEasy. To be able to use [...]

Thanks for the information on that. Seems these might be interesting additions and tools? (thx)

uzi18, 2025-03-08:
Here:
[image]

[image]

If you like terminal there is one :)