letscontrolit/ESPEasy #5278: CVE-2025-27840: Are our devices at risk?

mehdi-song, 2025-03-10:
https://nvd.nist.gov/vuln/detail/CVE-2025-27840 describes the existence of undocumented command on the ESP32 that potentially act as backdoors. There's an article here:
https://www.bleepingcomputer.com/news/security/undocumented-commands-found-in-bluetooth-chip-used-by-a-billion-devices/

TD-er, 2025-03-10:
Nope, it is completely blown out of context by calling it a 'backdoor'.

See also this Reddit post, which does explain it quite well:

[image]
Screenshot from this explainer on YouTube: https://www.youtube.com/watch?v=ndM369oJ0tk&ab_channel=LowLevel

See also here: https://www.esp32.com/viewtopic.php?t=44776
Especially the [post made by ESP_Sprite](https://www.esp32.com/viewtopic.php?p=145304#p145304). (He is someone who really knows what he's talking about. I know his hacking work for ages before he started working at Espressif)

tonhuisman, 2025-12-14:
This question has been answered completely, and can be closed.