letscontrolit/ESPEasy #5288: [FR] Add Bosch BMP58x (BMP580 - BMP581 - BMP585) Pressure Sensor

iz8mbw, 2025-03-25:
Ciao guys.
Please, in spare times, can you add the new Bosch BMP58x (BMP580 - BMP581 - BMP585) Pressure Sensor?

Libraries:
https://github.com/boschsensortec/BMP5_SensorAPI
https://docs.arduino.cc/libraries/sparkfun-bmp581-arduino-library/

BMP580 Boards can be found on AliExpress at less than 2 euros.

Thanks.

TD-er, 2025-03-25:
Just curious, what makes these better than the BMP38x versions?

iz8mbw, 2025-03-25:
The new sensors are more accurates.

TD-er, 2025-03-25:
OK, let's bring out the "accurate" vs "precise" charts...

[image]

It can only be accurate if it has been calibrated and also being processed exactly according to the manufacturer's specifications... and often in the opposite order of how I named them.

I don't believe those Chinese suppliers will perform a calibration after producing the boards.

Support for those will be added (not sure yet at what time, as sensors have to be ordered, etc)
However I am curious about the use case you plan on using those.
For example if you really need the higher sample rate, we might need to also add a SPI implementation of those.

TD-er, 2025-03-27:
Do you have a link to the "2$ boards" ?
I only see the sensor for around that price point. Any board with this sensor I can find is way overpriced.

closed 2025-09-16 by TD-er