letscontrolit/ESPEasy #5291: Wrong %unixday_sec% value after summer time change

fly74, 2025-03-30:
Hi Devs,

after change to summer time in Germany the value auf %unixday_sec% shows 1h less seconds in value.

Unix shell:

date -d "1970-01-01 $(date +%T)" +%s
65772
date
Sun Mar 30 19:16:17 CEST 2025

ESP:
[image]

NTP settings:

[image]

fly74, 2025-03-30:
The time on the ESP is shown correctly

[image]

tonhuisman, 2025-03-30:
The `unix*` time variables represent UTC, and are the basis for the other date/time variables that adhere to the Timezone configuration.

fly74, 2025-03-30:
But where is the local secs value stored ?

fly74, 2025-03-30:
Changed the linux part to UTC.

closed 2025-03-30 by fly74