letscontrolit/ESPEasy #5292: [BUG] missing P176 (VE.Direct) in the ESP8266 Energy build

sbrech, 2025-03-31:
**Describe the bug**
Can't find the 'Communication - Victron VE.Direct' Device

**To Reproduce**
Steps to reproduce the behavior:
1. flash ESP_Easy_mega_20241222_energy_ESP8266_4M1M.bin to a Wemos D1
2. try to find 'Communication - Victron VE.Direct' in the Device drop-down

**Expected behavior**
I found a 'Communication - Victron VE.Direct' entry

**Screenshots**

[image]

**Used platform (please complete the following information):**
 - ESP type: ESP8266
 - Build version: ESP_Easy_mega_20241222
 - Build set energy

**Platform Specifics (when applicable, please complete the following information):**
 - Flash size: 4M
 - Brand/Model: Wemos D1

TD-er, 2025-03-31:
It isn't included in ESP8266 builds, as we have a feature freeze on ESP8266.
We're constantly running into build size issues for ESP8266 and have spent lots of 100s of hours to make it all fit.
So new features and new plugins will only be included in official builds for ESP32.
It can still be used on ESP8266, but only on custom builds.

TD-er, 2025-12-14:
I'll close it now, please let me know if this needs to be re-opened

closed 2025-12-14 by TD-er