letscontrolit/ESPEasy #5294: Problems with DS18B20

Filicya, 2025-04-06:
The sensor used for parasite power in DS18B20 is showing abnormal values.
Temperature:86.81℃

The surrounding temperature is about 24 degrees.

Build:⋄ 20241222 - Mega
System Libraries:⋄ ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ mega-20241222
Plugin Count:⋄ 47 ["Normal","No Debug Log"]
Build Origin: GitHub Actions
Build Time:⋄ Dec 22 2024 23:45:09
Binary Filename:⋄ ESP_Easy_mega_20241222_normal_ESP8266_4M1M_VCC
Build Platform:⋄ Linux-6.5.0-1025-azure-x86_64-with-glibc2.35
Git HEAD:⋄ HEAD_8977d4d

tonhuisman, 2025-04-06:
Did you apply all tips provided in the [documentation](https://espeasy.readthedocs.io/en/latest/Plugin/P004.html#p004-page) ?
It states that parasitic power configuration isn't supported, and also warns for the quality of some cheap sensors from 'alternative' vendors.

Can you share a screenshot of the sensor configuration page?

tonhuisman, 2025-04-06:
That's only the overview,  I'd like to see the details 🤔

Filicya, 2025-04-06:
[image]
ID recognition is done.

tonhuisman, 2025-04-06:
Your configuration shows the sensor is parasite-powered, and also that it's unsupported... 👼.
IMHO, the only solution is to provide power to the sensor VDD pin, 3.3V is fine, and use a 4k7 pull-up resistor from VDD to the data pin of the DS18B20.

Filicya, 2025-04-06:
This issue occurred after I updated to a new firmware. 
I forgot what the old firmware version was....

TD-er, 2025-04-06:
That must have been quite an old version, as I did rewrite the code for this sensor a few years ago.
Maybe it can work with the new method when only using a single 1-wire sensor, though the pin might need a pull-up enabled.

Can you try adding a 4k7 resistor between the GPIO pin and 3V3 to see if that will make it work?
If so, then I could add a checkbox to enable GPIO pull-up.

Filicya, 2025-04-06:
The pull-up resistor is already connected.
I thought maybe there was a problem with a specific GPIO, so I swapped it out, but that didn't fix it.
As for the connection, the pull-up resistor is connected to the GPIO, and then the GPIO and GND are connected to one sensor in parasite mode with two 0.8mm solid wires.

So you rewrote the code.
I'm not sure why it gave me the answer 85°C.

TD-er, 2025-04-06:
If I remember correctly, 85C is some error code for this sensor.

tonhuisman, 2025-04-06:
85C is the power-on initial value. This seems to imply that parasite-power usage isn't working here...

TD-er, 2025-04-06:
I really wonder how it could have worked before...
See [datasheet](https://www.analog.com/media/en/technical-documentation/data-sheets/MAX31820PAR.pdf) page 7 and 11.
As mentioned, the sensor may take upto 1.5 mA, meaning your pull-up resistor should be no more than 2k, or else you may not be able to draw 1.5 mA at all and even then the voltage may drop significantly.
The internal pull-up resistor of an ESP is in the order of 50 - 100k, so very weak when compared to what is needed.

So I am really surprised it did work at all.
The datasheet also shows some FET to supply the data line with enough power while not reading and the 1-wire plugin in ESPEasy has no support for toggling an extra GPIO pin to address this.

Filicya, 2025-04-06:
I regret updating the firmware...

TD-er, 2025-04-06:
Do you have any rough idea of when you installed the setup?

TD-er, 2025-04-06:
I think I can make a small adjustment to the code to write a 'high' level to the pin right after the read bit.
Will make a test build, so you can test...
ESP8266 normal 4M1M with VCC as in the screenshot, right?

TD-er, 2025-04-06:
I made a [testbuild](https://www.dropbox.com/scl/fi/4hp73hl3mg84z6ec009gz/ESP_Easy_mega_20250406_normal_ESP8266_4M1M_VCC.bin?rlkey=9vem8ivmdqyzu16x9dgo12ida&dl=0).
Can you try it?

TD-er, 2025-04-06:
Made another [testbuild](https://www.dropbox.com/scl/fi/wdi3ywpslhhihvc7ussqr/ESP_Easy_mega_20250407_normal_ESP8266_4M1M_VCC.bin?rlkey=k91q755two7swf5kw5d92nox0&dl=0)

I simplified the code a bit and also try to make sure the GPIO pin is high when not being used.
Hopefully this will allow to have 1 parasitic powered sensor to work.

@Filicya 
Please let me know if it is working on your setup.

Filicya, 2025-04-07:
It still shows around 85°C.
I'll check the wiring again.

TD-er, 2025-04-07:
Can you try [this build](https://www.dropbox.com/scl/fi/ysmn5t6uxudv3154byr0u/ESP_Easy_mega_20250407_normal_ESP8266_4M1M_VCC_.bin?rlkey=j7gzp0f8yxnqcxv1uyacc60ht&dl=0) ?
I made the 'reset' longer, according to the datasheet

[image]

The 2nd "Master Rx" was not a minimum of 480 us, now it is...

[image]

Filicya, 2025-04-08:
Environment - 1-Wire Temperature | yane | 28-ff-55-4b-32-18-01-2e [DS18B20] | ❶ (9) | GPIO-14 (D5) | Temperature:86.19

It seems that the information is not being obtained properly from the sensor after all.

Thank you for all your help.

uzi18, 2025-04-08:
@Filicya please try older builds as well, every year step once you find working version

Filicya, 2025-04-08:
I tried it with a 3-wire system and it seems to work fine.

I'll make a 3-core wire and change it. . . .

Thank you.

Filicya, 2025-04-08:
When I connected it with a 3-core wire it worked fine, but when I tried connecting it in parasite mode on the microcontroller side, the temperature rose to 90°C.

I don't have an oscilloscope or anything so I can't see the signal, but it seems like something is wrong.

TD-er, 2025-04-08:
What is shown as statistics on the task config page?

For example:

Address: | 28-ff-2d-90-bb-22-02-2c [DS18B20]
-- | --
Resolution: | 12
Parasite Powered: | false
Samples Read Success: | 407
Samples Read Init Failed: | 0
Samples Read Retry: | 0
Samples Read Failed: | 0

And what happens if you set the resolution to 9 bits? (when in parasitic mode)

TD-er, 2025-04-08:
I have now wired up some sensors in parasitic mode and I can find them, but no valid read till now...

[image]

So I'm now able to reproduce it...

TD-er, 2025-04-08:
OK, with a single 4k7 pull-up resistor, it is possible to read upto 1 sensor using parasitic power:

[image]

Problem is the sensor may read `0x0550`, when the bus has too much load connected.
This value is the power-on reset value.
First I had connected 6 modules and an extra pull-up resistor connected on the last one, however the Vcc pin of the sensors isn't connected to anything.
After disconnecting the last 3 and the incorrectly mounted pull-up resistor, I can read some from a parasitic powered sensor.

TD-er, 2025-04-08:
[test build](https://www.dropbox.com/scl/fi/olfcjsl6vg93hx1wt26g2/ESP_Easy_mega_20250408_normal_ESP8266_4M1M_VCC__.bin?rlkey=9h09od205bzzxj4d0ld8zp6v2&dl=0)

This should support upto 1 sensor parasite powered.

Make sure to only have GND and data pin connected and the pull-up 4k7 resistor on the board side.

Filicya, 2025-04-09:
Thank you,
I'll take the time to try it out.

The sensor reading is lead success, but when I changed the wiring to use the parasite power supply method, it went up to around 90°C.

uzi18, 2025-04-09:
Maybe you got fake sensor, they encounter a lot of different troubles with parasite mode.

TD-er, 2025-04-09:
How is the sensor wired?

Pull-up should be on the board and only GND and data of the sensor should be connected.

uzi18, 2025-04-09:
Vcc of sensor should be wired to sensors gnd

Please add some photos of sensor and cable. 
How long is cable?

TD-er, 2025-04-09:
Hmm that's not how I wired my test setup yesterday.
Also some datasheet (which I cannot find again right now, it was from analog) stated to only wire the data line and GND...

Now that I search for it, I only see 'howto' pages mentioning to short Vdd and GND together.
I'm curious what the difference is in behavior.

uzi18, 2025-04-09:
@TD-er maybe for power supply detection module - to be sure it detects correctly low state on VDD.

Filicya, 2025-04-10:
Page 7 of the Analog Devices datasheet has a diagram showing how to connect Vdd and GND.

I twisted pins 1 and 3 so as not to short them to pin 2, and connected them to the ESP8266 through a 0.8mm (3M) two-core solid wire.

TD-er, 2025-04-10:
And there is a pull-up resistor of 4k7 on the board side?

Filicya, 2025-04-11:
A 2.2KΩ resistor is attached to the board side.

When I was testing the firmware, I noticed that noise was being introduced to the DHT11, although I'm not sure what the effect was.

[image]

uzi18, 2025-04-11:
Looks like fake sensors - genuine starts with 28-00-00-...

tonhuisman, 2025-04-11:
Task 4 is disabled, but it is configured for the same GPIO as the DHT11. Not sure if that was enabled, sometime during your testing, but possibly the DHT11 is somewhat 'upset' by the data stream that was meant for the DS18B20?
Best would be to reconfigure or delete task 4 to avoid confusion.
The DHT11 might need a power-cycle to recover from the invalid state.

TD-er, 2025-04-11:
> Looks like fake sensors - genuine starts with 28-00-00-...

Only older ones.
I have a few which are for sure genuine ones and these start with
* `28-ff-2a-43-bb-22-02-f4`
* `28-ff-2d-90-bb-22-02-2c`

Not sure how the numbering is done, but these are from the same batch of board-assembly, so very likely from the same reel (manufacturer handles stock, so might be a different reel)

Filicya, 2025-04-11:
I didn't realize it was a fake. 
I must have made the same IO by mistake.

uzi18, 2025-04-11:
My bad, note says: 
'It is a fake if its ROM address does not follow the pattern 28-xx-xx-xx-xx-00-00-xx'

https://github.com/cpetrich/counterfeit_DS18B20

uzi18, 2025-04-11:
Here you got more about internals: https://www.tinytransistors.net/2021/01/16/real-vs-fake-ds18b20/

Filicya, 2025-04-12:
Thank you for all your help.
For some reason, it seems to be reading fine with D4 (GPIO-2).

TD-er, 2025-04-12:
GPIO-2 does have a pull-up resistor.
See: https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp8266

And maybe the presence of a LED may also 'clean up' the signal.

tonhuisman, 2025-04-12:
> Thank you for all your help. For some reason, it seems to be reading fine with D4 (GPIO-2).
> 

Do you have the SPI interface enabled on the Hardware page? As that is using GPIO-14/12/13 (D5/D6/D7), and when enabled _will_ cause interference with other sensors when using the same pins. It's quite hard to configure those pins in a task if SPI is enabled, but it's not impossible🤔

Filicya, 2025-04-12:
Does this mean that it is not being used?
[image]

tonhuisman, 2025-04-12:
> Does this mean that it is not being used?

Correct.

TD-er, 2025-04-12:
Yep, with "init SPI" unchecked, those pins are not being used.
So it is likely the pull-up resistor and/or the LED on that pin might have a positive effect on the readings.
I am now adding some extra stats to show the rise time of the GPIO pin, so we can maybe see why it is not working on those pins, without the need of hooking up a logic analyzer to the pin and take a deep dive in your setup.
This value can then also be used to make the evaluation of logic level duration a bit less strict.

For example the shortest "low" you can expect from a sensor is 15 usec.
However if the rise time is close to this duration, a short "low" may be interpreted as a long "low".
Or the sensor may be without power too long and thus the sensor will reset.

TD-er, 2025-04-12:
Just a short update to show what the effect is on the parasite power pin voltage during read/scan

[image]
As can be seen, the voltage does get lower during read, which will trigger sensor faults during reading.
So I really think the extra pull-up on your GPIO-2 is helping out here.
This setup is with 4 sensors (3 actual fake ones) connected via really thin dupont wires barely touching the PCB through holes.
So about as bad as it can get, perfect to optimize for :)

TD-er, 2025-04-13:
I just added an extra compensation for poor RC rise time of the signal.
In this test I added an 1 nF capacitor over the data pin and GND, which would add about 4.7 usec to the rise time.
The measured rise time is 8 usec, which can clearly be seen here in the analog part of this measurement. (without the capacitor the measured rise time was 2 usec)

[image]

As soon as the ESP detects the voltage to rise over the threshold, it sets the pin to output a 'high' level.
The maximum low duration to signal a '1' is 15 usec.
As can be seen, the logic signal is very close to this, when having a poor RC rise time.

This will be shown in the ESPEasy user interface like this:

[image]

With this code, the parasite powered 1-wire sensors can still be read and used.

N.B. the rise time is measured at the start of each read/write operation.

TD-er, 2025-04-13:
@Filicya Can you test this test build?
https://www.dropbox.com/scl/fi/omw2bnktg778p01hgoyoq/ESP_Easy_mega_20250413_normal_ESP8266_4M1M_VCC.bin?rlkey=t3pz52hgwmvx4p168duq07xtn&dl=0

Filicya, 2025-04-13:
> [test build](https://www.dropbox.com/scl/fi/olfcjsl6vg93hx1wt26g2/ESP_Easy_mega_20250408_normal_ESP8266_4M1M_VCC__.bin?rlkey=9h09od205bzzxj4d0ld8zp6v2&dl=0)
>
> This should support upto 1 sensor [...]

This build can get numbers successfully.

TD-er, 2025-04-13:
I wired the sensors to an ESP8266 and noticed a few strange issues.

See the strange 1.75 level between readings when using GPIO-2:

[image]

When using GPIO-0 (which also has a pull-up resistor of roughly 10k present), there is another odd level at around 2.4V inbetween readings:

[image]

Both are below the absolute minimum of 2.8V, so it isn't odd the sensor will return bogus values.

TD-er, 2025-04-13:
I have no clue why the ESP8266 is reacting different compared to ESP32.
See:

[image]

It just seems like the output source strength of the GPIO pins (also tried GPIO-12/-13) is much lower compared to ESP32.

Also I have no idea where this 1.9V level may come from.

TD-er, 2025-04-13:
> This build can get numbers successfully.

Also tried this build and some older builds.
This build you told was working on your setup is absolutely not working on my (ESP8266) setup here.
Tried several sensors and several GPIO pins.

So I am beginning to believe the idea that it was working on older builds was merely a fluke and likely multiple retries in the background may have led to the idea it was somewhat working with parasite powered sensors.

TD-er, 2025-04-13:
OK, I'm really quite convinced parasite powered on ESP8266 is not going to work for some reason.

Here 2 samples...
The 'low' part of about 2.1V and 2.5V are while the sensor is taking the reading.

[image]

The first is with 4k7 pull-up resistor and the GPIO for the data set to output high.
The second is the same situation with a 1k8 resistor in parallel over the 4k7 pull-up resistor.

So it seems the ESP8266 is not capable of sourcing enough power when set as output to pull the level high strong enough to keep the internal capacitor of the sensor charged.

Filicya, 2025-04-14:
> [@Filicya](https://github.com/Filicya)このテストビルドをテストできますか? https://www.dropbox.com/scl/fi/omw2bnktg778p01hgoyoq/ESP_Easy_mega_20250413_normal_ESP8266_4M1M_VCC.bin?rlkey=t3pz52hgwmvx4p168duq07xtn&dl=0

This build is unable to detect parasite mode. Why?

TD-er, 2025-04-14:
I also noticed it may take a while for the sensor to detect parasite powered.
In the debug logs I also print a 'P', when this is detected and I noticed this flag may toggle every now and then. Typically the later queried sensors (when multiple are present) consistently show the correct state, but the first one being read is not always constant.

My theory is that when the sensor resets itself, the parasite flag also gets reset.
So that's why I changed the code to only check for parasite powered flag when this flag was not yet seen.
However in the web UI, I do check it again directly on the sensor. Maybe I should take into account the cached value if present.

uzi18, 2025-04-14:
@TD-er it toggles even wirh vdd+gnd connection?
Dallas had available sensors with -PAR suffix they were directly to use in parasite mode.
Basically you have there only 2 legs - Dx and GND

TD-er, 2025-04-14:
There seems to be something seriously wrong with the direct write to a pin on ESP8266.
Writing a 0 is working just fine, but writing a 1 isn't.

TD-er, 2025-04-14:
Problem is here....

[image]
Right after sending the `0x44` to start a measurement, the current increases.
However the voltage drops here to below 2.8V (1.9V in the screenshot)
Then the voltage rises and the sensor considers this as a 'reset' request and then signals a presence pulse, which is as long as other 'presence' pulses (+/- 1 msec difference)

Then after the 800 msec wait, the sensor reports `0x50` and `0x05`, which is a signal of the "after reset" state.

Also the fact the sensor can pull down the level is an indication of the pin being put to 'input' state and not output, even though I explicitly try to set it as such.
Also the curved edges on rising pulses show the GPIO isn't pulled high via the GPIO but rather via a resistor.

N.B. I also tried to power the 1-wire sensor from another GPIO pin, which is working perfectly fine. So the pin itself can source enough current for the sensor(s).
I found [here](http://bbs.espressif.com/viewtopic.php?t=139) the ESP8266 GPIO can source at least 12 mA, which should be more than enough.

TD-er, 2025-04-14:
OK, got it working...
Now I only need to make it fast again...
Anyway, problem is in switching the pin mode.
The pin was never switched to output mode.

This is in Parasite power mode.

[image]

As can be seen, the pins are swiftly pulled high and show no curved edge anymore...

[17 more replies]

PR #5295 (2025-04-16): [1wire] Pull up data pin  after reading a bit to support parasitic power
Only for 1 sensor possible

Fixes: #5294

* [ ] Testing by used
closed 2025-04-16 by TD-er