letscontrolit/ESPEasy #5309: [BUG] DS18B20 Not Detected with New Firmware (ESP_Easy_mega_20250430 on WT32-ETH01)

kaindjoh, 2025-05-02:
**Describe the bug**
After upgrading from ESP_Easy_mega_20241222 to ESP_Easy_mega_20250430, my DS18B20 temperature sensors are no longer detected.
In ESP_Easy_mega_20241222, I found that a 3.3kΩ pull-up resistor worked correctly, while 4.7kΩ did not. Based on this, I installed the 3.3kΩ resistor permanently in my sensor assembly, which was encapsulated in 2k gel for waterproofing and deployed in a rainwater cistern. Replacing the resistor is not an option due to this waterproofing :)
However, after updating to ESP_Easy_mega_20250430, none of my DS18B20 sensors are detected, even with 3.3kΩ pull-up. Downgrading to ESP_Easy_mega_20241222 restores functionality.

**To Reproduce**
Steps to reproduce the behavior:
Flash ESP_Easy_mega_20250430_normal_ESP32_4M316k_ETH.bin to WT32-ETH01.
Connect DS18B20 sensors as follows:
DATA → GPIO-4
GND → GND
VCC → 3.3V
Pull-up resistor → 3.3kΩ between VCC and DATA (sealed in 2k gel).
Open ESP Easy web interface and check sensor status.
Sensors are not detected.
Flash back to ESP_Easy_mega_20241222, sensors are detected correctly.

**Expected behavior**
DS18B20 sensors should be detected and provide temperature readings, as they do in ESP_Easy_mega_20241222.

**Screenshots**
no Screenshots

**Used platform (please complete the following information):**
ESP type: WT32-ETH01
Build version: ESP_Easy_mega_20250430_normal_ESP32_4M316k_ETH.bin
Build set: Normal

**Platform Specifics (when applicable, please complete the following information):**
Flash size: 4M
Brand/Model: WT32-ETH01
Browser + OS: Windows 11 / Chrome

**Additional context**
Were there any changes in the 1-Wire library that affect DS18B20 detection?
Could the new firmware require different pull-up resistor values?
Are there other reports of DS18B20 not working in ESP_Easy_mega_20250430?

tonhuisman, 2025-05-02:
- What length is the wire between the first sensor and the ESP?
- What length is the wire between the first and second DS18b20?

You can lower the pull-up resistance further to ca. 2k2 by adding a 6k8 resistor between VCC and GPIO-4, or 5k6 to go to ca. 2k pull-up.
This _doesn't_ take into account any LED (or other components, with or without serial resistors) that might also be connected to GPIO-4!

TD-er, 2025-05-02:
There have for sure been changes to the 1-wire code.
However this should not make it harder to detect sensors.
Rather the opposite!

What has changed:
- Actively set GPIO pin to output high, instead of just letting the pull-up resistor pull the signal high.
- Time rise-time before addressing a 1-wire sensor so we should take into account this rise time to make sure sensors without strong enough pull-up resistors (or extremely long cables) should not run into timing issues.
- Show rise-time in task config (can you show a screenshot of the page showing the rise time?)
- Keep 1-wire GPIO pin active high during measurement to make sure sensor will remain powered if a parasite powered sensor is used (only supported for 1 parasite powered device on the bus)

So can you show what the task config page shows as rise time?

kaindjoh, 2025-05-02:
What length is the wire between the first sensor and the ESP? -> 5m
What length is the wire between the first and second DS18b20? -> 7,5m

I've encapsulated the ESP, the connecting cables, and the resistor in a box with 2k gel due to the temperature/humidity conditions. Unless absolutely necessary, I don't want to disassemble it :)

kaindjoh, 2025-05-02:
> There have for sure been changes to the 1-wire code. However this should not make it harder to detect sensors. Rather the opposite!
>
> What has changed:
>
> * Actively set GPIO pin to output high, [...]

first sensor:
[image]

second sensor:
[image]

> * Keep 1-wire GPIO pin active high during measurement to make sure sensor will remain powered if a parasite powered sensor is used (only supported for 1 parasite powered device on the bus)

no parasit-mode used

> 
> So can you show what the task config page shows as rise time?

kaindjoh, 2025-05-02:
Link to sensor:
https://www.amazon.de/dp/B07KNQVWV2?ref_=ppx_hzsearch_conn_dt_b_fed_asin_title_1&th=1

Sorry, i have no ESP32 (WT32-ETH0) left

I use 3,3V

Everything works fine with ESP_Easy_mega_20241222 (the 4k7 didn't work, but the 3k3 does).

TD-er, 2025-05-02:
I do have a few of those sensors, though with a shorter cable.
But at least as 'fake' as those are you listed. (not original ones)

I don't think changing the boot state is going to make a difference, but you could select it to use an internal pull-up.

I will try it on an ESP32-classic chip (those WT32-ETH01 are a bit hard to wire up as they don't have an USB port) and see what happens.
I did test with lots of other sensors and flaky breadboard wires too to 'simulate' really poor setups and those worked just fine.

Is it possible for you to add another pull-up resistor, or is the ESP itself also covered in hotglue?

qwikus, 2025-05-02:
What value of rise time is considered acceptable? I have 3 non-original sensors on one port. Line length from 4 to 8 meters. Resistor 3k3. Works fine. Rise time 6 µs.

kaindjoh, 2025-05-02:
@TD-er 
Due to the temperature and humidity conditions, everything is completely encapsulated. I could potentially cut the wires between the ESP and DS18B20 as a last resort before opening the sealed box with the ESP. Then i could try a parallel resistor. Or i could try 5V instead of 3.3V

TD-er, 2025-05-02:
Can you disable this task (so you keep the sensor IDs stored) and add a new 1-wire task on the same GPIO?
Then select the checkbox "Auto select sensor" and see if it will pick one and try to read.

TD-er, 2025-05-02:
Hmm that sensor does seem to have quite a large capacitance.
I will try to fine-time it a bit more.

[image]

reported rise time is 1 usec here.

Dickesplumpel, 2025-05-07:
All DS18B20 are no name and original mixed together, some last years now, some died after months, they are all in a metal tube so I can't say which one last longer.

On 20250430_normal_ESP32_4M316k_ETH with 4 DS18B20 on GPIO-23 , no parasite power but device page shows true, functioning sometimes normal, sometimes strange,  boot Default, cable length 3-5m, 3.3V powered, pullup value forgotten.

[image]

Back to 20241222_normal_ESP32_4M316k_ETH

[image]

I will flash it tomorrow with LittleFS to see if the issue is gone then.

On 20250430_climate_ESP32_4M316k_LittleFS_ETH with 5 DS18B20 on GPIO-23, Default, no issues, 3.3V powered, pullup I think 3.3k, cable length 3-5m.

[image]

On 20250430_normal_ESP32c2_4M316k_LittleFS 3 DS18B20, GPIO-10, Input pullup, cable length 8m, pullup 3.7k, 5V powered, raise time 3µs, no issues.

Is it normal that from these values

[image]

this numbers are the same? Manufacturer?

[image]

TD-er, 2025-05-07:
OK, so it might be an issue with SPIFFS builds.
Thanks for testing...

Dickesplumpel, 2025-05-08:
20250430_normal_ESP32_4M316k_LittleFS_ETH no issues, so you might be on the right track.

[image]

TD-er, 2025-05-08:
@Dickesplumpel 
What board is used here?
The title of the issue mentioned WT32-ETH01
On that board an Ethernet chip is present, which is using GPIO-23. See: https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#pins-used-for-rmii-ethernet-phy
So just to be sure there is no hardware conflict.

Dickesplumpel, 2025-05-08:
[image]

kaindjoh, 2025-05-08:
FYI, i use this one:

[image]

TD-er, 2025-05-08:
Check!

By the way, I just came up with the following idea to implement for 1-wire...
Right now I just follow the defined timings as mentioned in the datasheets (several as there seem to be some differences per brand).
So if a minimum "low" time of 15 usec is defined, I accept 15 + the RC-rise time.

However, I did notice that on all 1-wire parts I did hook up to my logic analyzer, there is a clear distinction between a 0 and a 1 with regards to timing. Though between devices there is some fluctuation.

So my idea is not to focus on the strict timings, but rather the differences between high and low per device.

Let's assume these made up numbers in microseconds of how long a "low" period is for 3 different imaginary devices:

- `0` : 5 usec, `1` : 25 usec
- `0` : 10 usec, `1` : 35 usec
- `0` : 15 usec, `1` : 45 usec

As you can see, the last one might be a bit tricky when only considering the strict timings from the datasheet. You only need a small variation to no longer be able to distinguish a `0` from an `1`.

N.B. those made up numbers are not that far from the actual values.

So my idea is to not decide immediately per pulse whether it is a 0 or a 1, but rather per N bits to collect the observed timings in usec.
Then we can easily determine where the separation should be.
As a first approach I was thinking about collecting the 2 highest and 2 lowest values.
Just conceptually; Average those per 2 highest and 2 lowest and then the threshold is right in the middle.  (of course the actual calculation can be way simpler, but this is easier to grasp the idea and there are corner cases like what to do when from N samples you only have  a single `1` or `0`)

This way we don't need to rely on very strict timings and can also handle variations among sensors (e.g. fake vs. genuine, but also between brands)

Dickesplumpel, 2025-05-08:
I understood maybe half of it but this half sounds good.

uzi18, 2025-05-08:
@TD-er you think about store it per node or per task?

TD-er, 2025-05-08:
> [@TD-er](https://github.com/TD-er) you think about store it per node or per task?

I won't 'store' this as it is just intermediate runtime data.
It would be like 8 or 9 bytes in an array.
So it is just to capture the timings in usec and then after receiving the 8 bits it can be immediately analyzed and parsed into a single byte.
Then we can keep this threshold value for the rest of the communication to that sensor until the next "reset" call which is required before every communication sequence.

kaindjoh, 2025-05-25:
@TD-er , i get no device adress with the 20241222:

[image]

kaindjoh, 2025-05-25:
here the result with 20241222, same ds18b20, same wt32-eth0, same length of wire.

[image]

wutz-ESP, 2025-07-29:
Problem finding DS18b20 20241222 and newer 

**ESPEasy-mega-20240822/src/_P004_Dallas.ino**

`# define P004_SCAN_ON_INIT       PCONFIG(3)

boolean Plugin_004(uint8_t function, struct EventStruct *event, String& string)
{
  boolean success = false;

  switch (function)
  {
    case PLUGIN_DEVICE_ADD:
    {
      Device[++deviceCount].Number           = PLUGIN_ID_004;
      Device[deviceCount].Type               = DEVICE_TYPE_DUAL;
      Device[deviceCount].VType              = Sensor_VType::SENSOR_TYPE_SINGLE;
      Device[deviceCount].Ports              = 0;
      **Device[deviceCount].PullUpOption       = false;**
      **Device[deviceCount].InverseLogicOption = false;**
      Device[deviceCount].FormulaOption      = true;
      Device[deviceCount].ValueCount         = 1;
      Device[deviceCount].SendDataOption     = true;
      Device[deviceCount].TimerOption        = true;
      Device[deviceCount].GlobalSyncOption   = true;
      Device[deviceCount].OutputDataType     = Output_Data_type_t::Simple;
      Device[deviceCount].PluginStats        = true;
      break;
    }
`

**ESPEasy-mega-20241222/src/_P004_Dallas.ino**

two lines missing 
Device[deviceCount].PullUpOption       = false;
Device[deviceCount].InverseLogicOption = false;

`# define P004_SCAN_ON_INIT       PCONFIG(3)

boolean Plugin_004(uint8_t function, struct EventStruct *event, String& string)
{
  boolean success = false;

  switch (function)
  {
    case PLUGIN_DEVICE_ADD:
    {
      Device[++deviceCount].Number         = PLUGIN_ID_004;
      Device[deviceCount].Type             = DEVICE_TYPE_DUAL;
      Device[deviceCount].VType            = Sensor_VType::SENSOR_TYPE_SINGLE;
      Device[deviceCount].Ports            = 0;
      Device[deviceCount].FormulaOption    = true;
      Device[deviceCount].ValueCount       = 1;
      Device[deviceCount].SendDataOption   = true;
      Device[deviceCount].TimerOption      = true;
      Device[deviceCount].GlobalSyncOption = true;
      Device[deviceCount].OutputDataType   = Output_Data_type_t::Simple;
      Device[deviceCount].PluginStats      = true;
      Device[deviceCount].setPin2Direction(gpio_direction::gpio_output);
      break;
    }
`

tonhuisman, 2025-07-29:
> Problem finding DS18b20 20241222 and newer
> 
> **ESPEasy-mega-20240822/src/_P004_Dallas.ino**

@wutz-ESP:
Hmm, why are you using these older sources?
This issue is about DS18b20 not working as expected, but the trend in the comments is that this is most likely related to third-party sourced sensors, that seem to be less compatible to the original specs, and also don't seem to work with the already carefully tuned plugin code of ESPEasy.
This makes me wonder a) how your sensors are connected, and b) what exact make and model of sensor you are using?

TD-er, 2026-01-29:
And just curious, do you see some (many...??) read errors in the stats on the setup page of those DS18xx tasks?
Or is it just working fine now?

TD-er, 2026-02-04:
Great!

One more thing you could try is to set the resolution to 12 bit.
This will increase the read-time per sensor, thus increasing the possibility of something happening ;)
But on a more serious note, devices which cannot be set to (or kept at) 12 bits is a good indicator of fake modules.

wutz-ESP, 2026-02-05:
all tests are made on nodes 'on my desk' field node is still running 20240822

TD-er, 2026-02-05:
Great!
So the sensors you're having are either 'the real deal' or 'pretty good copies'.

TD-er, 2026-03-08:
Hmm I know the 1wire sensor doesn't mind the 5V and also some even may 'like' it more than 3V3.
But the ESP GPIO pins are not designed to handle 5V signal levels.
Also typically a 'high' level is defined as relative to the voltage on the power pin. (usually 0.7x Vcc)
But since the ESP's are designed for 3V3, their level may be 'clamped' to 3V3 (+0.3V) via a clamping diode.
Meaning the voltage of the signal wire may never exceed 3.6V, even though it is pulled-up via a resistor.

This probably does mess up the timings as received by the sensor as the ESP does see the signal rise more quickly compared to what the sensor receives.
For the sensor, the timing signals may appear different due to this 'clamping'

On older builds, the timing as sent by ESPEasy was just based on counting clock ticks.
The problem with that approach is that these timings may be wrong by the time they reach the sensor, or data sent back to the ESP is received on the ESP side as the capacitance of the cable depends on the type of cable and its length. (and also how many sensors are connected to the bus)
So I changed the approach by actually measuring the rise time of the signal and adjusting the timings based on this so it doesn't depend on the cable length.

But apparently there are still setups where this is counter productive. :(

wutz-ESP, 2026-09-16:
New test using 

ESP_Easy_mega_20260915_normal_ESP8266_4M1M

Device - Environment - DS18xxx/MAX31xxx/1-Wire Temperature 
address of two DS18B20 are shown correct on changing resolution from 9 to 10 and pressing Submit
Tab reloads an shows addresses as before

ESP_Easy_mega_20260915_display_B_ESP32_4M316k

Device - Environment - DS18xxx/MAX31xxx/1-Wire Temperature 
address of four DS18B20 are shown correct on pressing Submit

[image]

Tab reloads an shows addresses - None -

[image]

***  if you press here submit all devices in this task are deleted  ****

**on reloading Device and Edit this Task addresses of all four devices shown correct**

TD-er, 2026-09-16:
Can you press submit on that page while having the 'inspect' window open on the 'network' tab?

[image]
On the `devices?index=...` page, on the 'Payload' tab, you can see the values being submitted.
What does it mention at the "dallas_addrX" (X being 0 ... 3)  field?

What happens is that each found device is assigned an index number and after submitting, a new scan is performed to retrieve that address again.
I guess that's kinda error-prone, so I should actually use the address as index here. The potential problem is that when a different set of devices reply when submitting, you essentially assign the wrong address to that task value.

TD-er, 2026-09-16:
OK, I think I know what is happening here. (and already have a working fix on my desk)

The scan for 1wire devices may sometimes fail, as some reset is given, wait for 10 usec and then disable interrupts to do highly timing critical stuff.
However, if right before disabling interrupts, some interrupt is fired, the timing is out of range and thus the call to `Dallas_reset` fails and no other communication is done.

Since the scan is also performed when loading the page, any extra interrupt (which is very likely when loading or saving the page) can cause this scan to fail.

So in my fix I now try this 2 times (or I can set it to 3 attempts if needed) and have not been able to reproduce your issue since. (I could reproduce it here)

The 2nd issue is that the index used in the selector box was the index of the scan result.
The order of those found devices is always the same, as long as all devices reply on a scan.
However if you have 1 device which not always replies to a scan for whatever reason, the index may be wrong.

For example:
- index 0: aa:...:aa
- index 1:  bb:...:bb
- index 2:  cc:...:cc

If when loading the page all 3 devices respond, you pick device "bb:...:bb" (which is index '1') and then when saving this device does not respond, then index '1' is stored in the settings as device "cc:...:cc".

The fix is to not have those indices as value in the selection box, but the `uint64_t` representation of the address.
So when saving, I don't do a rescan, but rather store this `uint64_t` value directly instead of searching again for it.

I will wrap up my code and make a new build for it.

wutz-ESP, 2026-09-16:
Sorry, a was to late this morning and had not enough time give you a response.

closed 2026-09-16 by TD-er