letscontrolit/ESPEasy #5310: Could we have...

fly74, 2025-05-04:
Could we have localized names of this %sysweekday_s% instead of doing this, or is there another way?

```
 If %sysweekday%=1
   [7SEG2].7dtext,SON~%syshour_0%.%sysmin_0%
  Endif
  If %sysweekday%=2
    [7SEG2].7dtext,MON~%syshour_0%.%sysmin_0%
  Endif
  If %sysweekday%=3
    [7SEG2].7dtext,DIE~%syshour_0%.%sysmin_0%
  Endif
  If %sysweekday%=4
    [7SEG2].7dtext,MIT~%syshour_0%.%sysmin_0%
  Endif
  If %sysweekday%=5
    [7SEG2].7dtext,DON~%syshour_0%.%sysmin_0%
  Endif
  If %sysweekday%=6
    [7SEG2].7dtext,FRI~%syshour_0%.%sysmin_0%
  Endif
  If %sysweekday%=7
    [7SEG2].7dtext,SAM~%syshour_0%.%sysmin_0%
  Endif
```

TD-er, 2025-05-04:
Not sure we should add localized names of anything, as it will likely explode in build size.
However we could add string variables.

This way we could use something like `[str#%sysweekday%]` for example when you define your string variables at system boot.
Would that already be a good compromise?

fly74, 2025-05-04:
Maybe a function like the Oracle PL/SQL function:

DECODE(%sysweekday_s%, 1, 'SON',
                    2, 'MON',
                    'ELSE_STRING')

tonhuisman, 2025-05-04:
As a workaround/alternative you can use this code (uses the new named variables feature 😃)
```
  Let,day,(%sysweekday%-1)*3
  Let,day3,%v_day%+3
  [7SEG2].7dtext,{substring:%v_day%:%v_day3%:SONMONDIEMITDONFRISAM}~%syshour_0%.%sysmin_0%
```

fly74, 2025-05-04:
@tonhuisman Thx, works perfect and is short enough.

closed 2025-05-04 by fly74