letscontrolit/ESPEasy #5311: Backup TAR archive doesn't include all filesystem files

fly74, 2025-05-04:
The TAR archive don't save all files e.g.

[image]

TAR:

[image]

TD-er, 2025-05-04:
OK, I do see the use case for changing fonts to customize your UI, but this one in the last screenshot does have strong Day-Of-the-Tentacles vibes... ;)

tonhuisman, 2025-05-04:
> TAR:
> 
> [image]

That looks like the Save (config) function that only includes the 'basic' configuration stuff, including rules, the Backup should have all files.

fly74, 2025-05-04:
> That looks like the Save (config) function that only includes the 'basic' configuration stuff, including rules, the Backup should have all files.

This is the file what http://$dnshostname/download has produced. Can we make it to include all files to have a really full backup file?

tonhuisman, 2025-05-04:
The full backup is available at the `http://$dnshostname/backup` endpoint, that's not available for the minimal_ota builds, though.

fly74, 2025-05-04:
No same result as /download.

tonhuisman, 2025-05-04:
Just tested for completeness on an ESP32 NEOPIXEL build, but here it is working as intended.
NB: I'm testing from the UI using the Save and Backup buttons, not from a script, maybe your script is taking a different route than expected?

fly74, 2025-05-04:
Sorry, /backup is working. But why 2 ways to backup with different content ?

tonhuisman, 2025-05-04:
> why 2 ways to backup with different content

It is possible to disable the .tar feature of `/download`, for backward compatibility, and then you might still want to be able to create a full backup of the system, resulting in 2 functional endpoints...
And also to save only stuff configuration-related, especially on larger file-systems, you can store quite some stuff, like when using the Cache controller or images to show on a display, that you might not want to save with configuration.

When scripting, you could use the `/backup` endpoint, and when that fails with a 404 error for builds that don't have the .tar feature included, switch to `/download` to get the configuration only.

fly74, 2025-05-04:
@tonhuisman Thx for clarification.

closed 2025-05-04 by fly74

fly74, 2025-05-04:
@tonhuisman but then I would need the filelist via json for scripting for the OTA builds, when it fits in to get the rule files too.

tonhuisman, 2025-05-04:
Hm, it might be problematic to squeeze that into those builds, the limits are very tight already.

fly74, 2025-05-04:
But is it not only listing to json?

tonhuisman, 2025-05-04:
Yes, but it's additional code, so might expand the build too much 🤯

fly74, 2025-05-05:
But the webUI already lists all files.