letscontrolit/ESPEasy #5313: [Forum] letscontrolit down?

chromoxdor, 2025-05-07:
Just wanted to get my morning dose of the forum but it seems to be down. 
@TD-er  dis something happen?

tonhuisman, 2025-05-07:
It's just slow, looks like there are quite a lot of 'guests' using the forum (> 3000), usually indicating something close to a DDOS 'harassment'.

chromoxdor, 2025-05-07:
> quite a lot of 'guests'

Is there anything free on offer today?
Maybe software 😁

TD-er, 2025-05-07:
We're really getting slammed by those stupid bots from all kinds of AI companies, most of which I never heard of.

I already did add a lot of them to make sure the bots get a different UI which doesn't include a session ID in the URL.
Problem with those bots is that they work in parallel with lots of bot instances.
And each will have a different session ID thus each fetched URL is different.
So the list of links to fetch will approach infinity.
This in turn will make the site more attractive to fetch, so more bots are added and the length will approach infinity even faster.

I will try to add a wildcard match for bots.

chromoxdor, 2025-05-07:
Thats  really annoying 😳

TD-er, 2025-05-07:
Yep has already taken me quite some hours in the past few days.
Quite frustrating, especially since each click may take minutes and re-logins which seem to loop forever.

Anyway, I hopefully added some 'Unknown bot' wildcard signature, let's see if this will free up some CPU-cycles on the server.

chromoxdor, 2025-05-07:
The guests are gone and everything loads in an instant again. Thanks!

TD-er, 2025-05-07:
I asked Remco what he did, as we "lost" about 4000+ 'guest' concurrent visits.

[image]

When I have an update, I'll let you know.

TD-er, 2025-05-07:
> Maybe new file airobots.txt with content "deny" :) As robots.txt looks like is not supported by AI ;)

On one hand, I hope this is serious, but on the other hand I hope it is a joke...
If serious, then it would mean those AI companies are even bigger scum than I already thought.

Seriously I truly hate this AI-hype these days as it only kills creativity and costs such enormous amounts of energy. Not only the electricity to run those AI engines, but also to counter all the untruth they spread and brain-dead videos and blog texts which appear just about everywhere and which take a while to realize it is again AI-generated crap.
I really truly hate AI!

uzi18, 2025-05-07:
In fact it was introduced as a joke, because ai robots does not respect robots.txt so they need own file ;)

TD-er, 2025-05-07:
Here the wall of shame of the log lines from today containing "bot" (case insensitive grep)
```
      1  duckduckbot
      2  SiteAuditBot
      3  Discordbot
      4  EzoicBot
      6  MojeekBot
      9  SeznamBot
     15  Qwantbot
     18  duckassistbot
     23  Pinterestbot
     53  DataForSeoBot
     80  facebookexternalhit
    206  OAI-SearchBot
    402  PerplexityBot
   1134  YandexBot
   2174  Googlebot
   2655  bingbot
   3504  AwarioBot
   4188  Applebot
   4955  MJ12bot
   5514  ClaudeBot
   5658  DotBot
   8885  SemrushBot
  21493  AhrefsBot
  50798  Amazonbot
  83782  GPTBot
```

This month alone about 150 GByte from the forum alone, likely most of this in the last 3 days.

uzi18, 2025-05-07:
@TD-er on the other hand if you block all bots you should have in mind about all search engines like google,bing,duckduckgo,etc.

TD-er, 2025-05-07:
> [@TD-er](https://github.com/TD-er) on the other hand if you block all bots you should have in mind about all search engines like google,bing,duckduckgo,etc. no food - not found :D

I will only block those who ignore all robots.txt, like OpenAI does apparently.
They have 0 documentation on rate limiting, robots.txt, etc. When you search for it in their docs pages, they only mention rate limiting requests to them.

I already wasn't fan of AI tools, but these kind of low-life creatures make me dislike AI even more!

I see 0 use for GPT bot or Amazon bot to crawl the forum so extremely excessive.
There are currently 5300+ crawlers active at this moment.

Why???

TD-er, 2025-05-07:
OK, I added all bots which I could find in robots.txt and also deny based on user agents in the .htaccess.

So hopefully this will lessen the load soon as this is totally unusable.

TD-er, 2025-05-08:
This is so extremely utterly frustrating.
Just half an hour ago the forum had like 77 guests online and was snappy.
Now it takes upto 5 minutes to load a page to login and there are like 5000 guests online.

I did block just about any AI bot or crawler I could find via .htaccess, have them all added to the robots.txt and still they keep on hammering the server.

I noticed the bot from bytedance (Bytespider) alone was responsible for 30 GByte of traffic.
And there was also a 'slight increase' in number of visitors (unique IPs, 'sites')

[image]

TD-er, 2025-05-08:
Not only ignore the robots.txt, but also keep changing the user agent, so they can still get to pass the .htaccess.
And I can't add like over 300k IP-addresses to the list.
Even if you assume /24 subnets, then it is still 122 subnets for those with 'bot' in the user agent. So you still have those with 'crawler', 'spyder', 'spider' or what other agent string they come up with.

I've seen some of those bots to list 'official IPs', but these are not all /24 subnets.
So there is a big chance such a ban would also ban valid users.
Also if they just use like 322k different IPs to 'attack' the site in one day, why not 300k new IPs on the next day.

uzi18, 2025-05-08:
@TD-er hint: https://stackoverflow.com/questions/62540108/htaccess-rule-if-with-fuzzy-user-agent

TD-er, 2025-05-08:
> [@TD-er](https://github.com/TD-er) hint: https://stackoverflow.com/questions/62540108/htaccess-rule-if-with-fuzzy-user-agent

I have something quite similar already.

```
RewriteCond %{HTTP_USER_AGENT} (\.ai\ |Agentic|AI\ Article\ Writer|AI\ Content\ Detector|AI\ Dungeon|AI\ Search\ Engine|AI\ SEO\ Crawler|AI\ Writer|AI21\ Labs|AI2Bot|AIBot|AIMatrix|AISearchBot|AI\ Training|AITraining|Alexa|Alpha\ AI|AlphaAI|Amazon\ Bedrock|Amazon-Kendra) [NC,OR]
RewriteCond %{HTTP_USER_AGENT} (Amazon\ Lex|Amazon\ Comprehend|Amazon\ Sagemaker|Amazon\ Silk|Amazon\ Textract|AmazonBot|Amelia|AndersPinkBot|Anthropic|AnyPicker|Anyword|Aria\ Browse|Articoolo|Automated\ Writer|AwarioRssBot|AwarioSmartBot|Azure|BardBot|Brave\ Leo|ByteDance) [NC,OR]
RewriteCond %{HTTP_USER_AGENT} (Bytespider|CatBoost|CC-Crawler|CCBot|ChatGLM|Chinchilla|Claude|ClearScope|Cohere|Common\ Crawl|CommonCrawl|Content\ Harmony|Content\ King|Content\ Optimizer|Content\ Samurai|ContentAtScale|ContentBot|Contentedge|Conversion\ AI|Copilot) [NC,OR]
RewriteCond %{HTTP_USER_AGENT} (CopyAI|Copymatic|Copyscape|Cotoyogi|CrawlQ\ AI|Crawlspace|Crew\ AI|CrewAI|DALL-E|DataForSeoBot|DataProvider|DeepAI|DeepL|DeepMind|DeepSeek|Diffbot|Doubao\ AI|DuckAssistBot|FacebookBot|FacebookExternalHit) [NC,OR]
RewriteCond %{HTTP_USER_AGENT} (Falcon|Firecrawl|Flyriver|Frase\ AI|FriendlyCrawler|Gemini|Gemma|GenAI|Genspark|GLM|Goose|GPT|Grammarly|Grendizer|Grok|GT\ Bot|GTBot|Hemingway\ Editor|Hugging\ Face|Hypotenuse\ AI) [NC,OR]
RewriteCond %{HTTP_USER_AGENT} (iaskspider|ICC-Crawler|ImageGen|ImagesiftBot|img2dataset|INK\ Editor|INKforall|IntelliSeek|Inferkit|ISSCyberRiskCrawler|JasperAI|Kafkai|Kangaroo|Keyword\ Density\ AI|Knowledge|KomoBot|LLaMA|LLMs|magpie-crawler|MarketMuse) [NC,OR]
RewriteCond %{HTTP_USER_AGENT} (Meltwater|Meta\ AI|Meta-AI|Meta-External|MetaAI|MetaTagBot|Mistral|Narrative|NeevaBot|Neural\ Text|NeuralSEO|Nova\ Act|OAI-SearchBot|Omgili|Open\ AI|OpenAI|OpenBot|OpenText\ AI|Operator|Outwrite) [NC,OR]
RewriteCond %{HTTP_USER_AGENT} (Page\ Analyzer\ AI|PanguBot|Paperlibot|Paraphraser\.io|Perplexity|PetalBot|Phindbot|PiplBot|ProWritingAid|QuillBot|RobotSpider|Rytr|SaplingAI|Scalenut|Scraper|Scrapy|ScriptBook|SEO\ Content\ Machine|SEO\ Robot|Sentibot) [NC,OR]
RewriteCond %{HTTP_USER_AGENT} (Sidetrade|Simplified\ AI|Sitefinity|Skydancer|SlickWrite|Sonic|Spin\ Rewriter|Spinbot|Stability|StableDiffusionBot|Sudowrite|Super\ Agent|Surfer\ AI|Text\ Blaze|TextCortex|The\ Knowledge\ AI|Timpibot|Vidnami\ AI|Webzio|Whisper) [NC,OR]
RewriteCond %{HTTP_USER_AGENT} (WordAI|Wordtune|WormsGTP|WPBot|Writecream|WriterZen|Writescope|Writesonic|xAI|xBot|YouBot|Zero\ GTP|Zerochat|Zhipu|Zimm) [NC]
RewriteRule (.*) - [F,L]

RewriteCond %{HTTP_USER_AGENT} (Semrush|DotBot|MJ12bot|MojeekBot|GPTBot) [NC]
RewriteRule (.*) - [F,L]

#block Semrush
Deny from 85.208.96.0/24
Deny from 85.208.98.0/24
Deny from 185.191.171.0/24

#block MJ12bot
Deny from 42.200.230.0/24
Deny from 65.108.46.0/24
Deny from 89.117.56.0/24
Deny from 185.4.176.0/24
Deny from 192.99.36.0/24
Deny from 195.191.219.0/24
Deny from 195.201.199.0/24
Deny from 207.180.200.0/24
Deny from 217.182.134.0/24

#block Bytedance
Deny from 20.171.207.0/24

BrowserMatchNoCase "meta-externalagent" bad_bot
BrowserMatchNoCase "LieBaoFast" bad_bot
BrowserMatchNoCase "Mb2345Browser" bad_bot
BrowserMatchNoCase "zh-CN" bad_bot
BrowserMatchNoCase "MicroMessenger" bad_bot
BrowserMatchNoCase "zh_CN" bad_bot
BrowserMatchNoCase "Kinza" bad_bot
BrowserMatchNoCase "Bytespider" bad_bot
BrowserMatchNoCase "Baiduspider" bad_bot
BrowserMatchNoCase "Sogou" bad_bot
BrowserMatchNoCase "Datanyze" bad_bot
BrowserMatchNoCase "AspiegelBot" bad_bot
BrowserMatchNoCase "adscanner" bad_bot
BrowserMatchNoCase "serpstatbot" bad_bot
BrowserMatchNoCase "spaziodat" bad_bot
BrowserMatchNoCase "undefined" bad_bot
BrowserMatchNoCase "claudebot" bad_bot
BrowserMatchNoCase "facebook" bad_bot
BrowserMatchNoCase "Petalbot" bad_bot
BrowserMatchNoCase "YandexBot" bad_bot
BrowserMatchNoCase "Applebot/0.1" bad_bot
BrowserMatchNoCase "aiohttp" bad_bot
BrowserMatchNoCase "facebookexternalhit/1.1" bad_bot
BrowserMatchNoCase "facebookcatalog/1.0" bad_bot
BrowserMatchNoCase "aiohttp" bad_bot
BrowserMatchNoCase "GPTBot/1.2" bad_bot
BrowserMatchNoCase "Amazonbot/0.1" bad_bot
BrowserMatchNoCase "AhrefsBot/7.0" bad_bot
Order deny,allow
Deny from env=bad_bot
```

TD-er, 2025-05-08:
OK, Apache crashed again so the number of visitors and open connections is low now.
Let's see how long it lasts.

TD-er, 2025-05-08:
I added a few more IP-blocks to the .htaccess of those that still were able to access the server.
Remco also added some additional blocks to the firewall on the server.

Today alone (even with most of the .htaccess block active since 1am) we already have over 215k unique IP-addresses in the logs scaping the forum and wiki.

TD-er, 2025-05-08:
OK, we seem to be "under attack" by some botnet which is scraping from a prefetched list.
In the last 2 hours we had like 23k unique IPs fetching URLs.
These have normal looking user agent strings, though it is a bit odd that yesterday we had like close to 20k unique user agent strings. It is a bit unlikely there are so many actively used browser versions, so I think it is randomly generated.

Also the URLs fetched by these seem to have some session-ID in the URL, however it is a bit odd to see the same session-ID coming from requests made from different IP-addresses.

Since these requests are not made by logged in users, it is also strange to see `GET /forum/posting.php?mode=quote&p=...` in the requests.

When looking at the session IDs, there are over 30k sessions between 9:40 and 11:55

TL;DR
I have absolutely no idea how to block these.

TD-er, 2025-05-08:
Found an interesting pattern in the IP-addresses.
Since it is a bit 'interesting range', I will not publicly disclose it.

However I hope the forum will now get some rest to serve actual users.

TD-er, 2025-05-08:
Just to let y'all know, the forum software will now be updated and also the PHP version.
So the current downtime is expected... for once :)

sincze, 2025-05-08:
@TD-er Are we open to suggestions to put the forum behind cloudflare e.g. ? It provides a free plan but I can pull some strings for something different as this idea is non-profit.

TD-er, 2025-05-08:
phpBB forum update has been completed and also PHP updated to a version of this decade :)

@sincze 
I have no idea what the features are of the free plan.
Is it still 'free' if you get attacked by this many bots as we were in the last few days?
Or will Cloudflare rate limit access for free plans?

I know of stories where users of similar protection systems (no idea if those were Cloudflare or other similar products) were put on rather steep payment plans after such a surge in traffic.

Just to give some idea... we have had about 200 GB of traffic in just a few days and about 322k unique IPs.
So I have worked my way through about 1 GB of log files to see the patterns of what had to be blocked.

Right now the hosting of the forum and wiki is "free" as it is being hosted on one of the servers of Remco, for over a decade already.

sincze, 2025-05-08:
We should be fine on free plan first for non commercial usage. Have been
protecting several companies with free plan.

Only name servers need to be changed for it to start working.

Lock down firewall to only accept connection via cloudflare IP's. This will
avoid direct IP attacks.

Configuration we can do online or via code.

Some available tricks:

https://developers.cloudflare.com/bots/plans/free/

https://developers.cloudflare.com/turnstile/

TD-er, 2025-05-09:
Let's first see how the current changes into our own firewall and `.htaccess` will turn out to be.
Right now we're back to a few tens of concurrent visitors (bots included), which is perfectly fine.
I also reset the max. online counter to see how bad it may get during the nights.

We'll keep the Cloudflare idea in mind for when things don't appear to be 'fixed' which might be quite soon given how fast the army of AI-search-bots is growing. :(

TD-er, 2025-05-12:
Just an update...
This morning the forum was again hit by lots of requests.

By far, most of the IP-addresses used were from Vietnam, India and Indonesia.

In total 108k pageviews by 78515 unique IP-addresses between 6am and 7am local time.

As far as I could see, it looks like these IP-addresses are nearly all from ISPs in that region, so it looks like this was a botnet running on PCs from customers of these ISPs.

I have sent an email to 5 different abuse-addresses as it is just not feasible to block these IPs as that would result in blocking over 18'000 /24 subnets or simply block these countries.

Let's hope the subnet owners will actually do something about this.
I doubt many scripts will be able to block these kinds of attacks other than making the server unavailable which is probably the same result as we now likely had with over 11k active visitors.

TD-er, 2025-05-15:
This is really annoying...
Today already over 500k unique IPs, from subnets that weren't already blocked.
And all in short bursts of roughly 1 hour.

a few more from Vietnam and almost the entire 177.0.0.0/8 subnet from Brazil.

```
# Vietnam botnet
Deny from 203.210.128.0/20
Deny from 221.132.0.0/18
Deny from 222.252.0.0/14
Deny from 14.160.0.0/11
Deny from 14.192.0.0/11
Deny from 14.224.0.0/11
Deny from 113.160.0.0/11
Deny from 123.16.0.0/12

# India botnet
Deny from 223.0.0.0/8

# Brazil botnet
Deny from 177.0.0.0/8
```

No idea why this is happening. But this is absolutely not normal.

TD-er, 2025-05-17:
Forum is disabled right now, moving to another server which hopefully allows us to see more of what is happening.

TD-er, 2025-06-07:
I thought/hoped we had tackled it...
But in the last 24h there has been close to 700 GByte of traffic and since 7 am this morning the forum has been under serious attack.
We're working on it... when we hopefully can login...

sincze, 2025-06-07:
Thanks for the update—must be frustrating to keep battling this. It’s clear you’re putting in a lot of effort, and I r(and the community) really appreciate it.

Just as a friendly reminder: this is exactly the kind of situation where something like Cloudflare could take a big load off your plate. It won’t solve everything, but it can help absorb and mitigate this kind of attack traffic before it even hits the server.

Happy to jump in and help get that set up if you’d like—it might save us all some stress next time. Let me know!

TD-er, 2025-06-08:
This morning again the forum under attack.
But this one was different.
2 huge botnets from Mexico and Indonesia were making massive requests for password reset.

My theory...
They want to flood the mail system so our server gets banned by large mail providers and maybe even mentioned by other services like spamhaus.
This way it is impossible for new users to register as the activation mail doesn't arrive and also perhaps other mails to admin etc. gets lost so this might be a preparation for another attack.

So I wouldn't be surprised if some users did get emails about password resets.

sincze, 2025-06-08:
Wow.
Quite annoying indeed. https://www.cloudflare.com/en-gb/application-services/products/turnstile/ could be beneficial to detect automated vs real users.

TD-er, 2025-06-08:
OK, I didn't want to say this as I really don't want to make things political.
But I rather would not start working with some US company right now.
Actually the opposite, I personally would like to be less dependent on US companies, as with the current US government you really can't tell what will be happening tomorrow... or even the next few hours.
At home, I have been working hard to change my own 'online life' in such a way that I won't be put in an impossible position when the 'cloud services' I still use get blocked for me for whatever hypothetical reason.
I'm still using services offered by US companies, like Google and Microsoft, but I'm no longer dependent on them.  (not even for GitHub, though that will take quite some effort to change things, as also has been done with PioArduino which I'm also part of ...)
So after all this work I put into this, I don't think making my online presence more depending on a US-based service is the way to go.

N.B. This isn't just limited to US-based services, however those have become much less predictable due to US government pressure in the past few months.

IMHO one should always realize where dependencies are and evaluate what impact they might have when those are no longer available.
This has always been the idea behind ESPEasy, as you really should not allow yourself to be vendor locked-in.  (not even ESPEasy locked-in ;) )
And with the current US government, I'm not sure this way of thinking will be very much appreciated.

sincze, 2025-06-08:
@TD-er Thanks for sharing your vision.

Appreciated. Let me start looking for alternatives that fit your vision as stopping this alone in my opinion will cost you a day job. 

I also use Crowdsec that seems French.

TD-er, 2025-06-10:
We're looking into CrowdSec and also got Bunny.net suggested by TransIP.

Today we've had some extra plugins active in Apache, using quite a lot of rules via mod_security.

I already ran into issues when posting messages on the forum and got presented this message:

> Not Acceptable
> An appropriate representation of the requested resource could not be found on this server.

So the rule I triggered with my post is now disabled.
If someone else runs into issues posting, please let me know.