letscontrolit/ESPEasy #5317: playing with small string optimisation

uzi18, 2025-05-12:
@TD-er @tonhuisman puzzles here :)

Take a code from lines:
https://github.com/letscontrolit/ESPEasy/blob/cc2a7aecfb6c3e6b0330a02dbca89b7de80bc2d4/src/src/Helpers/Networking.cpp#L1406-L1418

pio run -e max_ESP32_16M1M_ETH

```
RAM:   [==        ]  19.2% (used 62768 bytes from 327680 bytes)
Flash: [======    ]  63.7% (used 2672485 bytes from 4194304 bytes)
```

Get rid of string and generate it in loop:

```
String getCNonce(const int len) {
  char alphanum[26+26+10+1];

  for (int i = 0; i < 26; ++i) {
    if (i<10) alphanum[i] = '0'+i;
    alphanum[i+10]= 'A'+i;
    alphanum[i+10+26]= 'a'+i;
  }

  String s;

  for (int i = 0; i < len; ++i) {
    s += alphanum[rand() % (sizeof(alphanum) - 1)];
  }

RAM:   [==        ]  19.2% (used 62768 bytes from 327680 bytes)
Flash: [======    ]  63.7% (used 2672469 bytes from 4194304 bytes)
```
So got -16 bytes of flash.

Now get rid with alphanum:

```
String getCNonce(const int len) {
  String s;

  const int numbers = '9'-'0'+ 1;
  const int alphas = 'Z'-'A'+ 1;
  const int allchars = 2*alphas+numbers; 
  char a;

  for (int i = 0; i < len; ++i) {
   int j = rand() % (allchars - 1);
   if (j<numbers) a = ('0'+j);
   else if (j<numbers+alphas) a = ('A'+j);
   else a = ('a'+j);
   s += a;
  }

  return s;
}

RAM:   [==        ]  19.2% (used 62768 bytes from 327680 bytes)
Flash: [======    ]  63.7% (used 2672453 bytes from 4194304 bytes)
```
So -32 bytes
add alphanum one more time (just for sizeof):
```
String getCNonce3(const int len) {
  static const char alphanum[] =   "0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz";
  String s;

  const int numbers = '9'-'0'+ 1;
  const int alphas = 'Z'-'A'+ 1;
  const int allchars = 2*alphas+numbers; 
  char a;

  for (int i = 0; i < len; ++i) {
   int j = rand() % (sizeof(alphanum) - 1);
   if (j<numbers) a = ('0'+j);
   else if (j<numbers+alphas) a = ('A'+j);
   else a = ('a'+j);
   s += a;
  }

  return s;
}

RAM:   [==        ]  19.2% (used 62768 bytes from 327680 bytes)
Flash: [======    ]  63.7% (used 2672429 bytes from 4194304 bytes)

```
So -56 bytes it is like a magic
how it is possible?

TD-er, 2025-05-12:
In your last piece of code you have the array `alphanum` declared as static, however it is no longer used.
`sizeof(alphanum)` is a `constexpr`, meaning the compiler does fill in a hard-coded integer value wherever this sizeof is used.
So for the linker the static declared array is never used and thus removed while linking.

If you remove the `static`, you will see the array will be left in, as the linker will not optimize away those variables in the scope of the function. (or `-O2` is quite more intelligent than expected)

uzi18, 2025-05-12:
have already removed static and no change in size, but didn't report it :)

TD-er, 2025-05-13:
```c++
  constexpr int numbers = '9'-'0'+ 1;
  constexpr int alphas = 'Z'-'A'+ 1;
  constexpr int allchars = 2*alphas+numbers; 
```
This might even reduce it a bit more, unless the compiler already found this optimization.

Not a reduction in size, but in execution speed is when you call `s.reserve(len);`

Further optimization:

```c++
String getCNonce3(const int len) {
  constexpr int numbers = '9'-'0'+ 1;
  constexpr int alphas = 'Z'-'A'+ 1;
  constexpr int allchars = 2*alphas+numbers; 
  constexpr int firstLowerCase = numbers+alphas;
  
  String s;
  s.reserve(len);

  for (int i = 0; i < len; ++i) {
   const int j = rand() % allchars;
   const char offsetChar = (j < numbers) ? '0' : ((j < firstLowerCase) ? 'A' : 'a');
   const char a = offsetChar + j;
   s += a;
  }
  return s;
}
```

Not tested, but I would be surprised if it would result in a larger build (`s.reserve(len)` might add 16 bytes I think)

uzi18, 2025-05-14:
In the end want almost similar version with offset+j at the end of function, but magic stopped me with head scratch ;) 
Still 8 bytes more than size-of version, when drop reserve method (this we want) it is new record 4 bytes less than size-of version (so not 16 but exactly 12 bytes) .

TD-er, 2025-05-14:
It is really hard to predict how much you'll gain in build size with some optimizations...
I've had it lots of times, where some change resulted in an unexpected large reduction in build size.
So searched for similar use cases which could benefit from the same or similar change.
Yay!, further reduction in build size.
Now find more occurrences... only to find the build size to be larger than before we began.

Also the compiler might suddenly decide to unroll a loop, thus increasing build size.

Sometimes declaring variables as const may help the compiler to strip those allocations and assignments, but it might also decide not to do this.

For example this part can be written in 2 lines of code:
```c++
    const int j = rand() % allchars;
    const char offsetChar = (j < numbers) 
      ? '0' 
      : ((j < firstLowerCase) ? 'A' : 'a');
    const char a = offsetChar + j;
    s += a;
```
However it makes it harder to read and it is completely unpredictable whether it may yield a smaller or even larger build size.

So I prefer better readable code as it will result in less bugs.

TD-er, 2025-05-14:
Please don't declare static strings in a header file.
This will result in either build errors/warnings if you include the header file more than once. Or it will result in multiple instances of the same static object.

uzi18, 2025-05-14:
Didn't said as static, but mean as global one.

Also there is small opportunity to short lots of strings in addLog functions by use of one more param like prepending which are almost the same in one plugin

TD-er, 2025-05-14:
> Also there is small opportunity to short lots of strings in addLog functions by use of one more param like prepending which are almost the same in one plugin

Yep, but I already did exactly that for those few plugins used in minimal OTA builds.

Probably this might still help for ESP8266 Display builds though.

N.B. You may also try to test `concat(...)` and `strformat(...)` to see which one results in smaller code. It isn't always predictable which one will be smaller.

I do know `.c_str()` is also rather expensive.
So it can also be useful to do stuff like this, when the same `String` needs to be used in a `strformat` call:

```c++
String foo;
// Something filling this string

// Make sure 'foo' will not be changed after we assign it to a const char*
const char* foo_c_str = foo.c_str();
addLog(LOG_LEVEL_INFO, strformat(
  F("%s %s %s"),
  foo_c_str,   foo_c_str,  foo_c_str));
```

uzi18, 2025-05-15:
@TD-er in Plugin_023_myFont every char font uses 8 bytes, but 1st and last 2 are always 0.

TD-er, 2025-05-15:
So that might save about 300 bytes.
On the other hand you need to add code to pad those while writing it to the screen.
Would be interesting to see if this would result in a net smaller build size.
And also would not hurt performance too much.
For example reading PROGMEM data has to be done per 32 bits.
If you try reading it per byte, then the internal code will actually perform a full read of 32 bits per read. It is absolutely not guaranteed to be handled in a smart way, even when reading in order.

TD-er, 2025-05-15:
Hmm took a quick look at the code where it is being called and this is already totally not speed-optimized :)

So the code can be easily adapted to something like this:
```c++
    for (i = 0; i < char_width && currentPixels + i < maxPixels; ++i) { // Prevent display overflow on the pixel-level
      if (i == 0 || i > 5) { sendChar(0x00); }
      else {
        sendChar(pgm_read_byte(Plugin_023_myFont[*string - 0x20] + i - 1));
      }
    }
```

uzi18, 2025-05-15:
@TD-er 
```
RAM:   [==        ]  19.2% (used 62752 bytes from 327680 bytes)
Flash: [======    ]  64.0% (used 2682385 bytes from 4194304 bytes)
Building .pio/build/max_ESP32_16M1M_ETH/ESP_Easy_mega_20250515_max_ESP32_16M1M_ETH.bin

```
vs :

[image]

RAM:   [==        ]  19.2% (used 62752 bytes from 327680 bytes)
Flash: [======    ]  64.0% (used 2682401 bytes from 4194304 bytes)

uzi18, 2025-05-15:
Your version
```
RAM:   [==        ]  19.2% (used 62752 bytes from 327680 bytes)
Flash: [======    ]  64.0% (used 2682397 bytes from 4194304 bytes)
```

TD-er, 2025-05-15:
OK, so with the stripping of the 3 bytes per character that would result in roughly 300 bytes less

TD-er, 2025-05-15:
Just for completion....
```c++
const char Plugin_023_myFont[][8] PROGMEM = {
  { 0x00, 0x00, 0x00, 0x00, 0x00 }, // SPACE
  { 0x00, 0x5F, 0x00, 0x00, 0x00 }, // !
  { 0x00, 0x07, 0x00, 0x07, 0x00 }, // "
  { 0x14, 0x7F, 0x14, 0x7F, 0x14 }, // #
  { 0x24, 0x2A, 0x7F, 0x2A, 0x12 }, // $
  { 0x23, 0x13, 0x08, 0x64, 0x62 }, // %
  { 0x36, 0x49, 0x55, 0x22, 0x50 }, // &
  { 0x00, 0x05, 0x03, 0x00, 0x00 }, // '
  { 0x1C, 0x22, 0x41, 0x00, 0x00 }, // (
  { 0x41, 0x22, 0x1C, 0x00, 0x00 }, // )
  { 0x08, 0x2A, 0x1C, 0x2A, 0x08 }, // *
  { 0x08, 0x08, 0x3E, 0x08, 0x08 }, // +
  { 0xA0, 0x60, 0x00, 0x00, 0x00 }, // ,
  { 0x08, 0x08, 0x08, 0x08, 0x08 }, // -
  { 0x60, 0x60, 0x00, 0x00, 0x00 }, // .
  { 0x20, 0x10, 0x08, 0x04, 0x02 }, // /
  { 0x3E, 0x51, 0x49, 0x45, 0x3E }, // 0
  { 0x00, 0x42, 0x7F, 0x40, 0x00 }, // 1
  { 0x62, 0x51, 0x49, 0x49, 0x46 }, // 2
  { 0x22, 0x41, 0x49, 0x49, 0x36 }, // 3
  { 0x18, 0x14, 0x12, 0x7F, 0x10 }, // 4
  { 0x27, 0x45, 0x45, 0x45, 0x39 }, // 5
  { 0x3C, 0x4A, 0x49, 0x49, 0x30 }, // 6
  { 0x01, 0x71, 0x09, 0x05, 0x03 }, // 7
  { 0x36, 0x49, 0x49, 0x49, 0x36 }, // 8
  { 0x06, 0x49, 0x49, 0x29, 0x1E }, // 9
  { 0x00, 0x36, 0x36, 0x00, 0x00 }, // :
  { 0x00, 0xAC, 0x6C, 0x00, 0x00 }, // ;
  { 0x08, 0x14, 0x22, 0x41, 0x00 }, // <
  { 0x14, 0x14, 0x14, 0x14, 0x14 }, // =
  { 0x41, 0x22, 0x14, 0x08, 0x00 }, // >
  { 0x02, 0x01, 0x51, 0x09, 0x06 }, // ?
  { 0x32, 0x49, 0x79, 0x41, 0x3E }, // @
  { 0x7E, 0x09, 0x09, 0x09, 0x7E }, // A
  { 0x7F, 0x49, 0x49, 0x49, 0x36 }, // B
  { 0x3E, 0x41, 0x41, 0x41, 0x22 }, // C
  { 0x7F, 0x41, 0x41, 0x22, 0x1C }, // D
  { 0x7F, 0x49, 0x49, 0x49, 0x41 }, // E
  { 0x7F, 0x09, 0x09, 0x09, 0x01 }, // F
  { 0x3E, 0x41, 0x41, 0x51, 0x72 }, // G
  { 0x7F, 0x08, 0x08, 0x08, 0x7F }, // H
  { 0x41, 0x7F, 0x41, 0x00, 0x00 }, // I
  { 0x20, 0x40, 0x41, 0x3F, 0x01 }, // J
  { 0x7F, 0x08, 0x14, 0x22, 0x41 }, // K
  { 0x7F, 0x40, 0x40, 0x40, 0x40 }, // L
  { 0x7F, 0x02, 0x0C, 0x02, 0x7F }, // M
  { 0x7F, 0x04, 0x08, 0x10, 0x7F }, // N
  { 0x3E, 0x41, 0x41, 0x41, 0x3E }, // O
  { 0x7F, 0x09, 0x09, 0x09, 0x06 }, // P
  { 0x3E, 0x41, 0x51, 0x21, 0x5E }, // Q
  { 0x7F, 0x09, 0x19, 0x29, 0x46 }, // R
  { 0x26, 0x49, 0x49, 0x49, 0x32 }, // S
  { 0x01, 0x01, 0x7F, 0x01, 0x01 }, // T
  { 0x3F, 0x40, 0x40, 0x40, 0x3F }, // U
  { 0x1F, 0x20, 0x40, 0x20, 0x1F }, // V
  { 0x3F, 0x40, 0x38, 0x40, 0x3F }, // W
  { 0x63, 0x14, 0x08, 0x14, 0x63 }, // X
  { 0x03, 0x04, 0x78, 0x04, 0x03 }, // Y
  { 0x61, 0x51, 0x49, 0x45, 0x43 }, // Z
  { 0x7F, 0x41, 0x41, 0x00, 0x00 }, // [
  { 0x02, 0x04, 0x08, 0x10, 0x20 }, // BACKSLASH
  { 0x41, 0x41, 0x7F, 0x00, 0x00 }, // ]
  { 0x04, 0x02, 0x01, 0x02, 0x04 }, // ^
  { 0x80, 0x80, 0x80, 0x80, 0x80 }, // _
  { 0x01, 0x02, 0x04, 0x00, 0x00 }, // `
  { 0x20, 0x54, 0x54, 0x54, 0x78 }, // a
  { 0x7F, 0x48, 0x44, 0x44, 0x38 }, // b
  { 0x38, 0x44, 0x44, 0x28, 0x00 }, // c
  { 0x38, 0x44, 0x44, 0x48, 0x7F }, // d
  { 0x38, 0x54, 0x54, 0x54, 0x18 }, // e
  { 0x08, 0x7E, 0x09, 0x02, 0x00 }, // f
  { 0x18, 0xA4, 0xA4, 0xA4, 0x7C }, // g
  { 0x7F, 0x08, 0x04, 0x04, 0x78 }, // h
  { 0x00, 0x7D, 0x00, 0x00, 0x00 }, // i
  { 0x80, 0x84, 0x7D, 0x00, 0x00 }, // j
  { 0x7F, 0x10, 0x28, 0x44, 0x00 }, // k
  { 0x41, 0x7F, 0x40, 0x00, 0x00 }, // l
  { 0x7C, 0x04, 0x18, 0x04, 0x78 }, // m
  { 0x7C, 0x08, 0x04, 0x7C, 0x00 }, // n
  { 0x38, 0x44, 0x44, 0x38, 0x00 }, // o
  { 0xFC, 0x24, 0x24, 0x18, 0x00 }, // p
  { 0x18, 0x24, 0x24, 0xFC, 0x00 }, // q
  { 0x00, 0x7C, 0x08, 0x04, 0x00 }, // r
  { 0x48, 0x54, 0x54, 0x24, 0x00 }, // s
  { 0x04, 0x7F, 0x44, 0x00, 0x00 }, // t
  { 0x3C, 0x40, 0x40, 0x7C, 0x00 }, // u
  { 0x1C, 0x20, 0x40, 0x20, 0x1C }, // v
  { 0x3C, 0x40, 0x30, 0x40, 0x3C }, // w
  { 0x44, 0x28, 0x10, 0x28, 0x44 }, // x
  { 0x1C, 0xA0, 0xA0, 0x7C, 0x00 }, // y
  { 0x44, 0x64, 0x54, 0x4C, 0x44 }, // z
  { 0x08, 0x36, 0x41, 0x00, 0x00 }, // {
  { 0x00, 0x7F, 0x00, 0x00, 0x00 }, // |
  { 0x41, 0x36, 0x08, 0x00, 0x00 }, // }
  { 0x02, 0x01, 0x01, 0x02, 0x01 }, // ~
  { 0x02, 0x05, 0x05, 0x02, 0x00 }  // DEL
};
```

```c++
    for (i = 0; i < char_width && currentPixels + i < maxPixels; ++i) { // Prevent display overflow on the pixel-level
      if (i == 0 || i > 5) { sendChar(0x00); }
      else {
        sendChar(pgm_read_byte(Plugin_023_myFont[*string - 0x20] + i - 1));
      }
    }
```

uzi18, 2025-05-15:
it was already stripped
one more approach: 
[image]

TD-er, 2025-05-15:
> it was already stripped one more approach: [image]

Not correct as the `charaddr` should not contain the result of `pgm_read_ptr`, but rather `Plugin_023_myFont[*string - 0x20];`

tonhuisman, 2025-05-15:
@uzi18 Can't yet see what gain you exactly achieve here, the numbers I see here hover around 2682397, can you show an original bin size vs. max gain you got so far?

uzi18, 2025-05-15:
2682385 is before any change in p023
But myFont is const... 
In fact it doesn't compile because pgm_read_ptr is macro, so it want more love.

uzi18, 2025-05-15:
Which one esp8266 env contains p023?

uzi18, 2025-05-15:
We should have some kind of info when build starts what we have in env included for sure. 
These define/undefine stuff masks sometimes true ;)

TD-er, 2025-05-15:
@uzi18 You mean [this](https://espeasy.readthedocs.io/en/latest/Plugin/_Plugin.html#plugins-per-build-set) isn't correct?

uzi18, 2025-05-15:
@TD-er  mixed idea:
```
    char b = *string - 0x20;
    const char * charaddr = Plugin_023_myFont[b];
    uint32_t bytes30 = pgm_read_dword(charaddr);
    uint32_t bytes74 = pgm_read_dword(charaddr+4);
    for (i = 0; i < char_width && currentPixels + i < maxPixels; ++i) { // Prevent display overflow on the pixel-level
      if (i == 0 || i > 5) {b=0;}
      else if (i<5){
        b = bytes30; 
        bytes30 >>= 8;
      } else {
        b = bytes74;
      }
      sendChar(b);
    }

RAM:   [==        ]  19.2% (used 62752 bytes from 327680 bytes)
Flash: [======    ]  64.0% (used 2682421 bytes from 4194304 bytes)
```

TD-er, 2025-05-15:
You can't read 32 bits like that, it needs to be 4-byte aligned.
Also I don't think it really will make the reading faster (even if you did properly align it) as you will do 2 reads here instead of 5 reads (worst case) when you apply your 300-byte reduction in font size. 
The code is quite a bit more complex and larger than earlier attempts.

uzi18, 2025-05-16:
@TD-er you are right, by the way didn't change row size from 8 to 5 so all build did not drop 3 bytes by row in table ;)
We have got original 2682385
vs :
```
    char b;
    for (i = 0; i < char_width && currentPixels + i < maxPixels; ++i) { // Prevent display overflow on the pixel-level
      if (i == 0 || i > 5) { b=0x00; }
      else {
        b=pgm_read_byte(Plugin_023_myFont[*string - 0x20] + i - 1);
      }
      sendChar(b);
    }

RAM:   [==        ]  19.2% (used 62752 bytes from 327680 bytes)
Flash: [======    ]  63.9% (used 2682113 bytes from 4194304 bytes)

```
Means -272 Bytes for esp32

uzi18, 2025-05-16:
```

    char b = *string - 0x20;
    const char * charaddr = Plugin_023_myFont[b];
    uint32_t b30 = pgm_read_dword(charaddr);
    uint32_t b74 = pgm_read_dword(charaddr+4);

    for (i = 0; i < char_width && currentPixels + i < maxPixels; ++i) { // Prevent display overflow on the pixel-level
      if (i<4){
        b = b30;
        b30 >>= 8;
      } else {
        b = b74;
        b74 >>= 8;
      }
      sendChar(b);
    }
```
this aligned version +64 for esp8266, +32 for esp32

uzi18, 2025-05-16:
> [@uzi18](https://github.com/uzi18) You mean [this](https://espeasy.readthedocs.io/en/latest/Plugin/_Plugin.html#plugins-per-build-set) isn't correct?

Nice to know about such list :)

TD-er, 2025-05-16:
Hopefully we can at some point generate that list in the docs and build ESPEasy using the same source file. Just to make sure the docs and builds are always in sync.

TD-er, 2025-05-16:
> ```
> 
>     char b = *string - 0x20;
>     const char * charaddr = Plugin_023_myFont[b];
>     uint32_t b30 = pgm_read_dword(charaddr);
>     uint32_t b74 = pgm_read_dword(charaddr+4);
> 
>     for (i = 0; i < char_width && currentPixels + i < maxPixels; ++i) { // Prevent display overflow on the pixel-level
>       if (i<4){
>         b = b30;
>         b30 >>= 8;
>       } else {
>         b = b74;
>         b74 >>= 8;
>       }
>       sendChar(b);
>     }
> ```
> 
> this aligned with 8B row version +64 for esp8266, +32 for esp32

This will not work, as you have no guarantee the PROGMEM data starts at a 32-bit aligned address.
So it will not even work when using the original 8-byte per character, and especially not when using the reduced 5-byte per character font data.
Please do not try to optimize it here for speed as it isn't the right location to do so.
The functions to get a single byte from PGM need to have a cache of the last read 32-bit value given an address. However I think the current pgm functions are implemented either via a macro or in a .h file. So adding such cache would significantly increase the build size.

You can have a look at the ESP8266 String implementation as that one does have an optimized function to construct a String object from a flash-string.
However this is quite platform-specific code, so it should not be used here in plugin code.

uzi18, 2025-05-16:
have added attribute to table:
```
-const char Plugin_023_myFont[][8] PROGMEM = {
+const __attribute__(( aligned(4))) char Plugin_023_myFont[][8] PROGMEM = {

```
but it was just a test

uzi18, 2025-05-16:
this one -264B 

```
    char b = 0x00;
    const char * baddr = Plugin_023_myFont[*string - 0x20];
    for (i = 0; i < char_width && currentPixels < maxPixels; ++i,++currentPixels) { // Prevent display overflow on the pixel-level
      if (i == 0 || i > 5) { b=0x00; }
      else { b=pgm_read_byte(baddr + i - 1);}
      sendChar(b); 
    }
    //currentPixels += char_width;
```

TD-er, 2025-05-16:
` char b = 0x00;` should be an `unsigned char` or `uint8_t`

```c++
    const char * baddr = Plugin_023_myFont[*string - 0x20];
    for (i = 0; i < char_width && currentPixels < maxPixels; ++i,++currentPixels) { // Prevent display overflow on the pixel-level
      unsigned char b = 0x00;
      if (i > 0 && i <= 5) { b=pgm_read_byte(baddr + i - 1); }
      sendChar(b); 
    }
```

uzi18, 2025-05-22:
@TD-er one more function:
https://github.com/letscontrolit/ESPEasy/blob/cc2a7aecfb6c3e6b0330a02dbca89b7de80bc2d4/src/src/PluginStructs/P025_data_struct.cpp#L15-L33
uses: 
```
RAM:   [==        ]  17.8% (used 58396 bytes from 327680 bytes)
Flash: [========  ]  80.7% (used 1533469 bytes from 1900544 bytes)

```

uzi18, 2025-05-22:
```
String Plugin_025_valuename(uint8_t value_nr, bool displayString) {
/*
  const __FlashStringHelper *strings[] { // just for reference
    F("AIN0 - AIN1 (Differential)"),     F("AIN0_1"),
    F("AIN0 - AIN3 (Differential)"),     F("AIN0_3"),
    F("AIN1 - AIN3 (Differential)"),     F("AIN1_3"),
    F("AIN2 - AIN3 (Differential)"),     F("AIN2_3"),
    F("AIN0 - GND (Single-Ended)"),      F("AIN0"),
    F("AIN1 - GND (Single-Ended)"),      F("AIN1"),
    F("AIN2 - GND (Single-Ended)"),      F("AIN2"),
    F("AIN3 - GND (Single-Ended)"),      F("AIN3")
  };
*/

  const int8_t inputs[] = {
    0,1,
    0,3,
    1,3,
    2,3,
    0,-1,
    1,-1,
    2,-1,
    3,-1
  };

  const size_t index         = (2 * value_nr);
  constexpr size_t nrinputs = NR_ELEMENTS(inputs);

  if (index + 1 >= nrinputs) return F("");

  String s;
  s += strformat(F("%s%d"), F("AIN"), inputs[index]);
  int input2 = inputs[index+1]; 
  if (input2<0) {
    if (!displayString) { 
      s += F(" - GND (Single-Ended)");
    }
  } else {
    if (displayString) {
      s += strformat(F("%s%s%d%s"), F(" - "), F("AIN"), input2, F(" (Differential)"));
    } else {
      s += strformat(F("%s%d"), F("_"), input2);
    }
  }
  return s;
};

RAM:   [==        ]  17.8% (used 58396 bytes from 327680 bytes)
Flash: [========  ]  80.7% (used 1533805 bytes from 1900544 bytes)
```

So takes more ...

tonhuisman, 2025-05-22:
> s += strformat(F("%s%d"), F("_"), input2);

`s += strformat(F("%c%d"), '_', input2);`

uzi18, 2025-05-22:
> > s += strformat(F("%s%d"), F("_"), input2);
> 
> `s += strformat(F("_%d"), input2);`

This creates another string in flash, want use duplicated string formats here

TD-er, 2025-05-22:
Better use `%c%d` then if you need to use a char to be inserted in a `strformat` call.
`String(F("_")).c_str()` is quite 'expensive' regarding build size.

TD-er, 2025-05-22:
@uzi18 
Just a tip...
When a function needs to return a `String` object, don't use it like this:
```c++
  if (index + 1 >= nrinputs) return F("");
```
Rather use the `EMPTY_STRING` static const declared object we have for this.
This saves about 16 bytes as the compiler would otherwise turn `return F("");` into `return String(F(""));`

uzi18, 2025-05-22:
this builds on esp8266 and esp32:

```
diff --git a/src/_P025_ADS1115.ino b/src/_P025_ADS1115.ino
index 274d627f5..419badb87 100644
--- a/src/_P025_ADS1115.ino
+++ b/src/_P025_ADS1115.ino
@@ -171,7 +171,7 @@ boolean Plugin_025(uint8_t function, struct EventStruct *event, String& string)
             String log;
 
             if (loglevelActiveFor(LOG_LEVEL_DEBUG)) {
-              log = strformat(F("ADS1x15 : Analog value: %.2f / Channel: %d"), value, P025_MUX(i));
+              log = strformat(F("%s : Analog value: %.2f / Channel: %d"), F("ADS1x15"), value, P025_MUX(i));
             }
             # endif // ifndef BUILD_NO_DEBUG
 
diff --git a/src/src/PluginStructs/P025_data_struct.cpp b/src/src/PluginStructs/P025_data_struct.cpp
index c9e3783fd..a62bfb989 100644
--- a/src/src/PluginStructs/P025_data_struct.cpp
+++ b/src/src/PluginStructs/P025_data_struct.cpp
@@ -11,6 +11,14 @@
 
 P025_VARIOUS_BITS_t::P025_VARIOUS_BITS_t(int16_t value) : _regValue(value) {}
 
+//RAM:   [==        ]  17.8% (used 58396 bytes from 327680 bytes)
+//Flash: [========  ]  80.7% (used 1533469 bytes from 1900544 bytes)
+//valname
+//RAM:   [==        ]  17.8% (used 58396 bytes from 327680 bytes)
+//Flash: [========  ]  80.7% (used 1533805 bytes from 1900544 bytes)
+//strformat
+//RAM:   [==        ]  17.8% (used 58396 bytes from 327680 bytes)
+//Flash: [========  ]  80.7% (used 1533521 bytes from 1900544 bytes)
 
 const __FlashStringHelper* Plugin_025_valuename(uint8_t value_nr, bool displayString) {
   const __FlashStringHelper *strings[] {
@@ -126,7 +134,7 @@ bool P025_data_struct::read(float& value, taskVarIndex_t index) const {
 
     if (loglevelActiveFor(LOG_LEVEL_DEBUG)) {
       addLog(LOG_LEVEL_DEBUG,
-             concat(F("ADS1x15: Start measurement failed"),
+             strformat(F("%s: Start measurement failed"),F("ADS1x15"),
                     reg.toString()));
     }
 # endif // ifndef BUILD_NO_DEBUG
@@ -141,7 +149,7 @@ bool P025_data_struct::read(float& value, taskVarIndex_t index) const {
 
     if (loglevelActiveFor(LOG_LEVEL_DEBUG)) {
       addLog(LOG_LEVEL_DEBUG,
-             concat(F("ADS1x15: Not Ready after start measurement"),
+                   strformat(F("%s: Not Ready after start measurement%s"),F("ADS1x15"),
                     reg.toString()));
     }
 # endif // ifndef BUILD_NO_DEBUG
@@ -153,7 +161,7 @@ bool P025_data_struct::read(float& value, taskVarIndex_t index) const {
   else {
     if (loglevelActiveFor(LOG_LEVEL_DEBUG)) {
       addLog(LOG_LEVEL_DEBUG,
-             concat(F("ADS1x15: Detected SPS: "), sds));
+             strformat(F("%s: Detected SPS: %d"),F("ADS1x15"), sds));
     }
   }
 # endif // ifndef BUILD_NO_DEBUG
@@ -166,7 +174,7 @@ bool P025_data_struct::read(float& value, taskVarIndex_t index) const {
 
     if (loglevelActiveFor(LOG_LEVEL_DEBUG)) {
       addLog(LOG_LEVEL_DEBUG,
-             concat(F("ADS1x15: Cannot read from conversion register"),
+             strformat(F("%s: Cannot read from conversion register"),F("ADS1x15"),
                     reg.toString()));
     }
 # endif // ifndef BUILD_NO_DEBUG
@@ -175,7 +183,7 @@ bool P025_data_struct::read(float& value, taskVarIndex_t index) const {
 
   value = _fullScaleFactor * raw;
 
-  addLog(LOG_LEVEL_INFO, strformat(F("ADS1x15: RAW value: %d, output value: %f"), raw, value));
+  addLog(LOG_LEVEL_INFO, strformat(F("%s: RAW value: %d, output value: %f"), F("ADS1x15"), raw, value));
   return true;
 }
 
@@ -224,7 +232,7 @@ bool P025_data_struct::startMeasurement(uint8_t i2cAddress, uint16_t configRegis
 
   if (loglevelActiveFor(LOG_LEVEL_DEBUG)) {
     addLog(LOG_LEVEL_DEBUG,
-           concat(F("ADS1x15: Start measurement failed"),
+           strformat(F("%s: Start measurement failed"),F("ADS1x15"),
                   reg.toString()));
   }
 # endif // ifndef BUILD_NO_DEBUG
@@ -263,7 +271,7 @@ long P025_data_struct::waitReady025(uint8_t i2cAddress)
 
       if (loglevelActiveFor(LOG_LEVEL_DEBUG)) {
         addLog(LOG_LEVEL_DEBUG,
-               concat(F("ADS1x15: waitReady OK, Config_reg: "),
+               strformat(F("%s: waitReady OK, Config_reg: "),F("ADS1x15"),
                       reg.toString()));
       }
 # endif // ifndef BUILD_NO_DEBUG
@@ -280,7 +288,7 @@ long P025_data_struct::waitReady025(uint8_t i2cAddress)
 
   if (loglevelActiveFor(LOG_LEVEL_DEBUG)) {
     addLog(LOG_LEVEL_DEBUG,
-           concat(F("ADS1x15: waitReady timeout, Config_reg: "),
+           strformat(F("%s: waitReady timeout, Config_reg: "),F("ADS1x15"),
                   reg.toString()));
   }
 # endif // ifndef BUILD_NO_DEBUG

```
But also gives more code:
Flash: [========  ]  80.7% (used 1533521 bytes from 1900544 bytes)

TD-er, 2025-05-22:
It may build, but it will crash.
```c++
         addLog(LOG_LEVEL_DEBUG,
               strformat(F("%s: waitReady OK, Config_reg: "),F("ADS1x15"),
                       reg.toString()));
       }
```
You can't put a flash string here to be used as something to replace `%s`. (at least not on ESP8266, as Ton mentioned)
Also a `String` class object needs to be converted to a C-string using `.c_str()` when used in `strformat` to replace a `%s`.
So that's why `concat` was added, as it does the conversion from flash-string for you.

TD-er, 2025-05-22:
The return value of `concat` is a `String`.
And `String` does have an `operator+` for lots of other types.
So you could do things like this:
```c++
addLog(LOG_LEVEL_DEBUG,
  concat(
    F("ADS1x15"), 
    F(": waitReady OK, Config_reg: ")) 
  + reg.toString()));
```

tonhuisman, 2025-05-22:
Still looking at the code for the output selector:

```
> String s;
>   s += strformat(F("AIN%d"), inputs[index]);
>   const int input2 = inputs[index+1]; 
>   if (input2<0) {
>     if (!displayString) { 
>       s += F(" - GND (Single-Ended)");
>     }
>   } else {
>     if (displayString) {
>       s += strformat(F(" - AIN%d (Differential)"), input2);
>     } else {
>       s += strformat(F("_%d"), input2);
>     }
>   }
>   return s;
```

Compared your code above (applied some changes to avoid 'expensive' F() macro to String conversions) with my proposal:
```
  const int input2 = inputs[index + 1];

  if (input2 < 0) {
    return strformat(displayString ? F("AIN%d - GND (Single-Ended)") : F("AIN%d"), inputs[index]);
  }
  return strformat(displayString ? F("AIN%d - AIN%d (Differential)") : F("AIN%d_%d"), inputs[index], input2);
```
On ESP8266 this reduces a NORMAL build with 96 bytes, but on ESP32 NORMAL build _increases_ with 96 bytes...

Your code above _increases_ both builds, ESP8266 with 76 bytes, and ESP32 with 192 bytes.

Changing the returntype of the function of course requires a few small changes in other places, but that's the same for both variants.

TD-er, 2025-05-22:
> On ESP8266 this reduces a NORMAL build with 96 bytes, but on ESP32 NORMAL build _increases_ with 96 bytes...

I think it is a very clean change.
However I can't see why it would increase the bin size on ESP32 builds.

```c++
String Plugin_025_valuename(uint8_t value_nr, bool displayString) {
/*
  const __FlashStringHelper *strings[] { // just for reference
    F("AIN0 - AIN1 (Differential)"),     F("AIN0_1"),
    F("AIN0 - AIN3 (Differential)"),     F("AIN0_3"),
    F("AIN1 - AIN3 (Differential)"),     F("AIN1_3"),
    F("AIN2 - AIN3 (Differential)"),     F("AIN2_3"),
    F("AIN0 - GND (Single-Ended)"),      F("AIN0"),
    F("AIN1 - GND (Single-Ended)"),      F("AIN1"),
    F("AIN2 - GND (Single-Ended)"),      F("AIN2"),
    F("AIN3 - GND (Single-Ended)"),      F("AIN3")
  };
*/

  const int8_t inputs[] = {
    0,1,
    0,3,
    1,3,
    2,3,
    0,-1,
    1,-1,
    2,-1,
    3,-1
  };

  const size_t index         = 2 * value_nr;
  constexpr size_t nrinputs = NR_ELEMENTS(inputs);

  if (value_nr >= (nrinputs / 2)) return EMPTY_STRING;

  const int8_t input1 = inputs[index];
  const int8_t input2 = inputs[index + 1];

  if (input2 < 0) {
    return strformat(displayString 
      ? F("AIN%d - GND (Single-Ended)")
      : F("AIN%d")
      , input1 );
  }
  return strformat(displayString 
      ? F("AIN%d - AIN%d (Differential)")
      : F("AIN%d_%d")
      , input1, input2);
};
```
IMHO is this a very clean solution.

TD-er, 2025-05-22:
Hmm this makes it slightly smaller:
```c++
String P025_data_struct::Plugin_025_valuename(uint8_t value_nr, bool displayString) {
  /*
     const __FlashStringHelper *strings[] { // just for reference
      F("AIN0 - AIN1 (Differential)"),     F("AIN0_1"),
      F("AIN0 - AIN3 (Differential)"),     F("AIN0_3"),
      F("AIN1 - AIN3 (Differential)"),     F("AIN1_3"),
      F("AIN2 - AIN3 (Differential)"),     F("AIN2_3"),
      F("AIN0 - GND (Single-Ended)"),      F("AIN0"),
      F("AIN1 - GND (Single-Ended)"),      F("AIN1"),
      F("AIN2 - GND (Single-Ended)"),      F("AIN2"),
      F("AIN3 - GND (Single-Ended)"),      F("AIN3")
     };
   */

  constexpr uint32_t inputs_1 = 0x32102100;
  constexpr uint32_t inputs_2 = 0xFFFF3331;

  if (value_nr >= 8) { return EMPTY_STRING; }

  const size_t shift = 4 * value_nr;

  const int input1 = (inputs_1 >> shift) & 0xF;
  const int input2 = (inputs_2 >> shift) & 0xF;

  if (input2 == 0xF) {
    return strformat(displayString
        ? F("AIN%d - GND (Single-Ended)")
        : F("AIN%d")
                     , input1);
  }
  return strformat(displayString
        ? F("AIN%d - AIN%d (Differential)")
        : F("AIN%d_%d")
                   , input1, input2);
}
```

I also tried to get rid of 8 calls of this function to show the output selectors:
```c++
void P025_data_struct::webform_LoadOutputSelector(struct EventStruct *event)
{
  const int nrOptions = 8;
  String    valOptions[nrOptions];

  for (int i = 0; i < nrOptions; ++i) {
    valOptions[i] = Plugin_025_valuename(i, true);
  }

  const uint8_t nrValues = P025_NR_OUTPUT_VALUES;

  for (uint8_t i = 0; i < nrValues; i++) {
    sensorTypeHelper_loadOutputSelector(event,
                                        P025_PCONFIG_INDEX(i),
                                        i,
                                        nrOptions,
                                        valOptions);
  }
}
```
Doesn't seem to make a lot of difference.
So maybe the compiler decided to unroll the loop.

uzi18, 2025-05-23:
disable unroll in this function does nothing:
```
    case PLUGIN_WEBFORM_LOAD_OUTPUT_SELECTOR:
    {
  const int nrOptions = 8;
  String    valOptions[nrOptions];
#pragma GCC unroll 0
  for (int i = 0; i < nrOptions; ++i) {
    valOptions[i] = Plugin_025_valuename(i, true);
  }
      //constexpr int nrOptions = NR_ELEMENTS(valOptions);

...

```