letscontrolit/ESPEasy #5318: TFT Question

fly74, 2025-05-13:
Sorry for ask here, but I can't get this [Link](https://www.amazon.de/dp/B0CDPYYQ74) thing to work. The labels are looking for i2c but it should be an SPI. Tried all from [here](https://espeasy.readthedocs.io/en/latest/Plugin/P116.html) but it is not showing everything. Maybe somebody has a working wiring to share. Backlight is turned on.

tonhuisman, 2025-05-13:
SCL = CLK (Hardware page)
SDA = MOSI (Hardware page)
DC = DC (Plugin)
CS = CS (Plugin)
RES = RES (Plugin)
BLK = Backlight (Plugin)

TFT Display model = ST7789 240x280

Most likely you haven't configured the TFT Display model correct...

fly74, 2025-05-13:
Thx, as I tried, maybe broken.

closed 2025-05-13 by fly74