letscontrolit/ESPEasy #5322: [BUG] st7789v3 170x320

pasi-konik, 2025-05-22:
**Expected behavior**
in plugin 116 - TFT display mode missing mode 170x320

**Used platform (please complete the following information):**
 -ESP32 devkit v1
 - Build set display

tonhuisman, 2025-05-22:
You are using the latest ESPEasy release? (20250430)

NB: This feature was [merged](https://github.com/letscontrolit/ESPEasy/pull/5263) on March 15th

pasi-konik, 2025-05-22:
correct resolution is 170x320, st7789v3 chip. I tried with new version of espeasy, it is better, but there are irregularities. The largest rectangle is st7789,r,0,1,319,169,white or st7789,r,1,0,169,319,white.

tonhuisman, 2025-05-22:
I tested and tuned it on a ST7735 using the full resolution rectangle. Don't have a ST7789 with that resolution (not sure if/where that's actually available).
There are some small differences in the ST77xx series of display controllers, but documentation isn't very clear on that. Using the ST7735 configuration on a ST7789 could explain the differences you experience.

What exact display are you using, and where can it be bought?

pasi-konik, 2025-05-23:
[image]
[image]

pasi-konik, 2025-05-23:
Temu, link on QR cide

tonhuisman, 2025-05-23:
I've ordered a similar display using a ST7789 controller, that's due to arrive in about a week, so then I'll have a look at what needs to be adjusted (will probably need a new entry in the settings).

pasi-konik, 2025-06-11:
is there any progress

tonhuisman, 2025-06-11:
Just connected it to an ESP, and using the ST7735 172x320 setting I can indeed use the specified 170x320 pixels, so most likely the display actually has 170x320 pixels, but needs a small adjustment as there's a white band at one side, and the max coordinates have to be limited somewhat.
I'll add another ST7735 config, as that seems to work fine (ST7735 probably uses the most tolerant/compatible setup for most of these ST77xx controllers)

PR #5329 (2025-06-12): [P116] Add support for ST7735/ST7789 170x320 displays
Resolves #5322 

Features:
- [P116] Add resolution 170x320 for ST7735 (and ST7789) display controllers
- Update documentation

TODO:
- [x] Testing by requester
- [x] Testing locally
closed 2025-06-12 by TD-er