letscontrolit/ESPEasy #5327: esp32 s2 multipoint processing

Dickesplumpel, 2025-06-06:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Used platform (please complete the following information):**
 - ESP type: [e.g. ESP8266, ESP32, ESP32-S2, ESP32-S3, ESP32-C3]
 - Build version: [e.g. self built, mega-YYYYMMDD]
 - Build set [e.g. normal, energy, display, climate, collection-X, max]

**Platform Specifics (when applicable, please complete the following information):**
 - Flash size: [e.g. 4M]
 - Brand/Model: [e.g. Sonoff Basic r2, NodeMCU, Wemos D1]
 - Browser + OS [e.g. Windows 11/Chrome, IOS/safari]

**Additional context**
Add any other context about the problem here.

Dickesplumpel, 2025-06-06:
esp32-s2 multipoint processing  is not working on ADC1 ch8 (T9) GPIO-9 and ADC1 ch9 (T10) GPIO-10, gives out digital value or voltage if factory calibration is enabled, two point calibration works. ADC1 ch0 to ADC1 ch7 works great. 
ESP_Easy_mega_20250430_normal_ESP32s2_4M316k_LittleFS_CDC_ETH build.

TD-er, 2025-06-06:
Hmm sounds like the calibration may not be present in the chip?

Can you post a screenshot of the calibration part of the ADC task?
I would like to know what kind of (factory) calibration is found on the chip.

Dickesplumpel, 2025-06-06:
Sorry, acidently blown it up 5 minutes ago, 24V was a bit tooo much. On the new chip I had this issue when I use "use current sample", when I switch to oversampling everything is o.k., I think that was the same "mistake" I had on the other chip (didn't recognize). So this behavior is normal?

TD-er, 2025-06-06:
I meant this section:

[image]

But then from your board

Dickesplumpel, 2025-06-07:
[image]

TD-er, 2025-06-07:
Hmm that's weird... I would expect the range at max. attenuation to go from 0 to a little over 3V
Have to check with other ESP32-S2 devices here.
Maybe there is an issue with the bit-depth not being set for all ports?

I may have to check later this evening when I'm at my desk again.

Dickesplumpel, 2025-06-07:
Yes, I have expected that too but it shows 1786mV for 5584, so 13bit resolution. The calibration curve shows from 0-4096 12bit. It seems that different IDF versions support different max attenuations (11 and 12) and different bit-depths (12 and 13).

TD-er, 2025-06-16:
Hmm another thing you might want to test...
If you change ADC pins, please also reboot.
It seems like I don't release the pin from the GPIO MUX, so I have no idea whether it still may have the other pins connected to the ADC input.

I'm not sure what's up with the ESP32-S2 and its actual ADC resolution.
The datasheet and TRM seem to have been updated about a year ago to mention 12 bit ADC for the ESP32-S2.

I did change the resolution for the S2 in a test build just to see what the ADC calibration functions do with the higher ADC values and it seems the calibration table isn't linear, so it actually is a calibration table.

12-bit ADC calibration curves:

[image]

Forcing the ADC resolution to 13 bit:

[image]

As you can see, the range of 0 dB attenuation is simply doubled, but the others are not an exact double.

The TRM states:

[image]

So 1100 mV is full range
Meaning with 0 dB attenuation you would expect 1100 mV to be the max (give or take some due to actual calibration)

However it looks like we're still "missing" about 2.5 dB...

Research continues...

Dickesplumpel, 2025-06-17:
Disable factory calibration does it. RuecklaufWZirkulation still has factory calibration enabled therefore jump down and up around 4 o'clock, 34°C.

[image]

Dickesplumpel, 2025-06-17:
I don't know what you mean. Adc1 ch0-9 are working correct when factory calibration is disabled and when I use my "own" calibration. When enabled they have that value jump at the "end of the calibration table" at 4096 bit as shown in the pictures with the marks so I thought it has something to do with 12/13bit resolution.
The Board is an Wemos-mini-S2

[image]

TD-er, 2025-06-17:
As far as I understand it and have tested, the resolution differences only limit the range of the ADC input.

Unless.... there is something where the ADC actually is reading in 13-bit mode and then only the last 12 bits are being used.

Here is the actual build for the S2 I have been using this morning: https://www.dropbox.com/scl/fi/elb4i6izezsr2mfsat05u/ESP_Easy_mega_20250617_normal_ESP32s2_4M316k_LittleFS_CDC_ETH.factory.zip?rlkey=fhk0qzz3f3tlihpbq5orkmxsy&dl=0

Zip file containing both the factory image (to be flashed via serial) and the non-factory image (upgrade via web page on ESPEasy)

N.B. please do not use this in production use as it is based on the code I'm working on for refactoring the WiFi/netwerk code.
But for WiFi only it is working quite well...

Dickesplumpel, 2025-06-17:
As far as I understand this behaviour is that the adc reads 13 bits but uses only 12 bit factory calibration which is no problem as far as the 13th bit is 0 (so same as 12 bit reading), what happens when the 13th bit changes from 0 to 1 ( from 4097 to 8192) I don't know, but this change seems to produce these jumps, but maybe I'm complete wrong, its only guessing.
I will test your new firmware on another s2.

TD-er, 2025-06-17:
Can you flash again the previous build from the web flasher and then -if the ESP can connect to WiFi again- update the firmware via the web interface of ESPEasy using the bin file without 'factory' in the name?

Dickesplumpel, 2025-06-17:
This works, internal adc 13bit, will take a closer look tomorrow, now I go for a bike ride. I think I have read somewhere that reading voltages lover than 150mV and higher than 2450mV at 12dB are not usable.

TD-er, 2025-06-18:
OK, I will add it later this evening as a separate PR so we will have it in a new build and it doesn't have to wait till I have finalized the WiFi rewrite

Dickesplumpel, 2025-06-18:
👍

PR #5332 (2025-06-21): [ESP32-S2] Change ADC for ESP32-S2 back to 13 bit (#5327)
Fixes: #5327
closed 2025-06-21 by Dickesplumpel