letscontrolit/ESPEasy #5358: [FR] Prometheus Metrics

fly74, 2025-07-22:
It seems that the metrics (/metrics) don't provide the cpu temperature. Can we add this?

tonhuisman, 2025-07-22:
@fly74 What (ESP32) build do you need for testing this?

fly74, 2025-07-22:
> That could look like this:
> 
> ```
> ...
> # HELP espeasy_cpu_temperature Level of CPU temperature in Celcius
> # TYPE espeasy_cpu_temperature gauge
> espeasy_cpu_temperature 28.40
> ...
> ```

not with mine:

espeasy_free_ram 216600
espeasy_free_stack 6544
espeasy_load 13.99
espeasy_uptime 30711
espeasy_wifi_connected 456311000
espeasy_wifi_reconnects 2
espeasy_wifi_rssi -50

and some espeasy_device_*

with ESP_Easy_mega_20250430_normal_ESP32_4M316k_ETH on ESP32-D0WDQ5-V3 - build the reason ?

tonhuisman, 2025-07-22:
> with ESP_Easy_mega_20250430_normal_ESP32_4M316k_ETH on ESP32-D0WDQ5-V3 - build the reason ?

Well, I added that new `cpu_temperature` value about an hour ago, so it's not in that build 🤣 🤣 🤣 
That why I asked what build you need to test 😉 
⚠️ Warning: we don't have SPIFFS builds for ESP32 anymore, only LittleFS ⚠️

[ESP_Easy_mega_20250722_normal_ESP32_4M316k_LittleFS_ETH.zip](https://github.com/user-attachments/files/21374095/ESP_Easy_mega_20250722_normal_ESP32_4M316k_LittleFS_ETH.zip)

> found some space lately, so maybe it fits in ?

No, it's a management decision that ESP8266 is feature frozen (as Espressif declared the ESP8266 platform deprecated about 5 years ago 😱), and also, I'm not the manager...
These 'extra bytes' are needed dearly so we can fix bugs from time to time.

fly74, 2025-07-23:
So I hope it is in next official build. thx @tonhuisman

fly74, 2025-07-23:
I thought you don't want to have to many issues open. :)

tonhuisman, 2025-08-17:
@fly74 
I've opened a new feature request for adding a new setting, it lists the current default time-outs available (with/without wait for ack).

fly74, 2025-09-06:
> > with ESP_Easy_mega_20250430_normal_ESP32_4M316k_ETH on ESP32-D0WDQ5-V3 - build the reason ?
> 
> Well, I added that new `cpu_temperature` value about an hour ago, so it's not in that build 🤣 🤣 🤣 That why I asked what build you need to test 😉 ⚠️ Warning: we don't have SPIFFS builds for ESP32 anymore, only LittleFS ⚠️ ([Migration guide](https://espeasy.readthedocs.io/en/latest/Tools/Tools.html#migrate-from-spiffs-to-littlefs-esp32))
> 
> [ESP_Easy_mega_20250722_normal_ESP32_4M316k_LittleFS_ETH.zip](https://github.com/user-attachments/files/21374095/ESP_Easy_mega_20250722_normal_ESP32_4M316k_LittleFS_ETH.zip)
> 
> > found some space lately, so maybe it fits in ?
> 
> No, it's a management decision that ESP8266 is feature frozen (as Espressif declared the ESP8266 platform deprecated about 5 years ago 😱), and also, I'm not the manager... These 'extra bytes' are needed dearly so we can fix bugs from time to time.

@tonhuisman can make this one for [20250430_neopixel_ESP32_4M316k_ETH] too?

PR #5359 (2025-07-25): [Metrics] Add internal CPU temperature
Resolves #5358 

Features:
- Add internal CPU temperature (when available) in Open Prometheus metrics
- Update documentation

TODO:
- [x] Testing by requester ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/5358#issuecomment-3104700275))