letscontrolit/ESPEasy #5384: [BUG]ESP32-s3 NR16... - Generic - MQTT Import (p037) not working correct in 30042025 and 20122024

HarryHase4, 2025-08-20:
**Describe the bug**
decimal "cut" in values is not working

**To Reproduce**
Generic - MQTT Import - values send are interger / string single value e.g. 0
log:
IMPT : [mqtt1#start] : 0.000
28881: EVENT: mqtt1#start=0.00

** tests: changing decimal to1 or 2 and back to 0, no change!

**Expected behavior**
0.00 not expected, expect only 0

**Screenshots**
https://files.fm/u/n9djj89rbn

**Used platform (please complete the following information):**
 - ESP32-S3 dev-board
 - Build version:  30042025 and 20122024
 - Build set: max
 - [ESP_Easy_mega_20250430_custom_ESP32s3_8M1M_LittleFS_CDC_ETH Apr 30 2025]
 
*Platform Specifics (when applicable, please complete the following information):**
 - Flash size: 16M

tonhuisman, 2025-08-21:
Hmm, your are rather incomplete,  can you shown them completely,  also including the 'on ... do' and 'endon' lines?

HarryHase4, 2025-08-21:
STOP: now it's getting spooky ... moment please

HarryHase4, 2025-08-21:
sorry, I don't know what happened, it is also working on mega-20250430

PR #5385 (2025-09-07): [P037] Use configured decimals when generating events
Resolves #5384 

Features:
- [P037] now uses the configured Decimals setting per Value when generating events

TODO:
- [ ] Testing by reporter
- [x] Update documentation