letscontrolit/ESPEasy #5389: [BUG] [P036] Spaces are getting ignored when font centered

chromoxdor, 2025-08-30:
I don't know if it is a feature or a bug, but when alignment is set to center, spaces in text are getting ignored.
I tried to add them manually and also by "d" transformation.

chromoxdor, 2025-08-30:
Edit: of course not the spaces in between text but leading spaces!

chromoxdor, 2025-08-30:
> Is that from the `Lines` fields in the Device Configuration, or when putting text on there via Rules?

Text via rules.. but I think it is the same for the lines field

tonhuisman, 2025-08-30:
> Text via rules

I assume you used quotes around the content?

TD-er, 2025-08-31:
Not sure how that can me made to work, as quite a lot of functions used to replace (system)variables do call `.trim()` on the string they processed.
The reason for this trim call is to get rid of variables which would be replaced with an empty string. For example not set variables.

If we were to change all those calls, then I think quite a lot of other stuff would break.

There was something added in the past (or was it in some PR- or issue-discussion?)  to allow for more verbose text processing.
A bit like adding some wrapper token which would then only be replaced when being displayed or in command/rules argument parsing.

Just an idea as a (temporary) work-around for the P036 display.
Maybe you can add such a token for which there is no character in the font?
Like an ASCII value < 32 and not too low as we use those as temporary placeholder when parsing.

chromoxdor, 2025-09-01:
> I wonder why you use Center to align stuff?

I am using it to center a temp value that varies between 2 and 3 digits. 

> so with uneven length content 2 lines will never be placed exactly below each other, that's what left- and right-align are for.

That doesn't matter for my use case.

I think my issue is so specific that it can probably be ignored. It was actually meant more as a question than a bug report. :)

tonhuisman, 2025-09-01:
Shouldn't that be `[var#temp#d3]` ?

chromoxdor, 2025-09-01:
But it is really not an issue since i found a workaround and with this little information the display has to show i can also use a left or right alignment and work then with spaces...

chromoxdor, 2025-09-08:
> The only way to use a Centered display without a shift on those values is to prefix the 99° with a 0; it's the number of characters that changes, if that didn't cause a shift when centered, then that would be a bug 😱

You are right…fortunately, for customisations, there is always the mighty rules editor 🙂

I ended up with a different layout and added some animation. Therefore, I created a custom "coffee" font. 

[image]
[image]

tonhuisman, 2025-12-14:
This seems too be quite specific, and has found a re-worked solution.

@chromoxdor Can you close this please?