letscontrolit/ESPEasy #5396: [BUG] multiple on System#Boot do + timerSet

agroszer, 2025-09-14:
**Describe the bug**
Somehow on System#Boot misbehaves?

**To Reproduce**
I got here Rule set1:
```
on System#Boot do
 Let,1,1
 Let,14,0
 Let,12,0

 event,onChange
endon

on vezpwr#State do
 event,onChange
endon
etc....
```

Rule set 3:
```
On System#Boot Do
  TimerSet,8,10
  Pulse,5,0,50
  Pulse,5,0,50
  Pulse,5,0,50
Endon

On Rules#Timer=8 Do
  TimerSet,8,10
  Pulse,5,0,50
  Pulse,5,0,50
  Pulse,5,0,50
Endon
```
On reboot WIFI connects, LED blinks 3x. But that's it no more blinks. Timer 8 does not fire? See nothing regarding timer 8 in the logs.
On System#Boot Do in rule set 1 executes fine, because the latter logic there works fine. (Uses a bunch of timers), logs events fine.

What's going on? Any chance to debug (running) timers?

**Used platform (please complete the following information):**
 - Build: ESP_Easy_mega_20250430_collection_A_ESP8266_4M1M Apr 30 2025

tonhuisman, 2025-09-14:
TL;DR; Not a bug, but works as designed.

2 concepts here that work somewhat different from your interpretation:
- The Rules 1..4 files are handled like a single set of rules, not 4 separate sets. For file-size reasons they where created separate, and for manageability that's still helpful, but they are processed as a single 'file'.
- An event handler (`On System#Boot Do`) is handled only once when triggered, starting from Rules 1 and continued for Rules 2, 3 and 4 if any Rules code is there. Any other event handler with the same trigger specificity is ignored.
 
When having multiple event handlers they are handled sequentially and in specificity-order, so if `On Sensor#Temp>=20 Do` is handled then both `On Sensor#Temp Do` and `On Sensor Do` are skipped, assuming the most specific rule is first in the file. When `On Sensor Do` is listed first, all events for `Sensor` are passed to that event-handler!

It's all [documented here](https://espeasy.readthedocs.io/en/latest/Rules/Rules.html), but that page has quite a lot of information...

TD-er, 2025-09-15:
Just to clarify the use of multiple event blocks acting on very similar events, you have to make sure the most specific one is listed first.

So in this order (as listed by Ton):
- `On Sensor#Temp>=20 Do` 
- `On Sensor#Temp Do` 
- `On Sensor Do`

agroszer, 2025-09-17:
"single set of rules": yeah I get it.

What I don't get is,
```on System#Boot do
 Let,1,1
```
in rules set 1 will execute for sure, but then why does the LED blink 3x (most probably from Rule set 3: On System#Boot Do)?
Timer 8 is definitely not running/firing.
Or IOW what the heck makes the LED blink 3x?

tonhuisman, 2025-09-17:
> makes the LED blink 3x?

You have to share more of your rules to be able to say something about that, f.e. what's going on in the `OnChange` event that's called multiple times and from multiple other events? (and should it possibly be triggered as `AsyncEvent` ?)

TD-er, 2025-09-17:
Just a few ideas to look for:
- Use `asyncevent` instead of `event`, as this will not keep the rules parsing occupied
- Make sure the ESP isn't rebooting for some reason. (using `event` may cause a stack overflow or out of memory if you have blocks calling `event`, which also calls `event`)
- Double check to see if each `on ... do` ends with an `endon`
- Add `logEntry` lines in your rules to see when calls are being made (see: https://espeasy.readthedocs.io/en/latest/Reference/Command.html )
- Is the GPIO connected to the LED also used for other purposes? (e.g. set as network status led on the Hardware tab)

closed 2025-09-21 by agroszer