letscontrolit/ESPEasy #5404: [BUG] MQTT - Auto Discovery setting missing in normal 20250430 ESP82xx

TungstenE2, 2025-09-25:
MQTT - Auto Discovery setting missing in normal 20250430 ESP82xx

https://espeasy.readthedocs.io/en/latest/Controller/C005.html

[image]

[image]

[image]

tonhuisman, 2025-09-28:
The documentation suggests to use `homeassistant/%devclass%/%unique_id%` for the Auto Discovery Topic setting, so it can provide the correct attributes per Device that's using the controller.
If you fix that setting (the rest seems to be correct), and use the Submit button, the Auto Discovery will be re-triggered automatically, (and start after a short, random, delay) and your devices should become visible in HA.

TungstenE2, 2025-09-28:
great, this did the magic! ;-)

many thx!

[image]

[image]

TungstenE2, 2025-10-04:
Hi all,

I was brave and updated the Chicken house.

ESP_Easy_mega_20250929_normal_ESP8266_4M1M.bin

Update was fine, but Controllers - MQTT auto discovery is still missing for ESP8266 while available for ES32C6.

I thought this was included in release 20250923. Was this wrong assumption?

[image]

https://td-er.nl/ESPEasy/latest/

TD-er, 2025-10-04:
MQTT Discovery is a new feature and we decided a while ago that ESP8266 will not get new features in regular builds.
It is still possible to use most of the new features, however this requires a custom build.

Espressif declared ESP8266 end-of-life about 5 years ago.

TungstenE2, 2025-10-04:
oh, that was not clear to me and also not from documentation.
thought this is in normal build.

may be you can mention in documentation as well?

https://espeasy.readthedocs.io/en/latest/Controller/C005.html

Was thinking to migrate to ESP32, so have to do it now for this feature.... ;-)

Only problem is the access to the chicken house and wiring done some years ago...

TD-er, 2025-10-04:
I don't know how much resources (RAM) the discovery feature uses.
We can make a custom build, however you should test this on a separate ESP8266 node first, before installing it in a hard-to-reach location like your chicken house.

TungstenE2, 2025-10-04:
could you help me with this custom build? that would be great.

I need only motorshield plugin, 1wire temp sensor, switch input for reed sensors and dummy switch.
Rules are also needed and mqtt to send data to controller.

[image]

TD-er, 2025-12-14:
Solved by updating docs

closed 2025-12-14 by TD-er