letscontrolit/ESPEasy #5409: [BUG] - switch inversed logic does not work

TungstenE2, 2025-09-28:
I am using a 4x Relay Modul Board with ESP32C6 and latest release.

[ESP_Easy_mega_20250925_normal_ESP32c6]

I am able to control the relay, but the logic needs to inverted, as it becomes active when state is 0 and inactive when state is 1.

I assumed the Sensor Setting 'Inversed Logic' is doing this, but is seems not to change anything.

Please check.

[image]

Thx for you support!

TungstenE2, 2025-10-06:
after a couple of reboots ESP32 is now inverted by HA is mixed up / inverted, need to check. Maybe delete HA swich and create again?

TungstenE2, 2025-10-08:
are you sure?

this is from a Tasmota plug using also Auto discovery:

[image]

tonhuisman, 2025-10-08:
Yes, I'm quite sure, I did quite some investigation, and when providing `switch` for Device Class, it is rejected by a current release of HA. Maybe it was accepted by older release, not sure.
Tasmota has their own plugin in HA, ESPEasy uses the common MQTT Auto Discovery (that's documented 'rudimentary', i.o.w. if you know _what_ to do, and _how_ to do it, then it'll 'just work', but if you don't..., it was quite a hair-pulling experience 🙀)

TD-er, 2025-10-15:
I uploaded the build with this fix here: https://td-er.nl/ESPEasy/latest/