letscontrolit/ESPEasy #5410: [FR] update 1wire Temp sensor limitation

TungstenE2, 2025-09-29:
Hi all,

I am moving from FHEM on Rasp Pi to HA OS on mini PC

On Rasp Pi I had 6 DS18B20 Temp Sensors hard wired using 1wire.

Now I would like to use the same hardwire attached to an ESP32C6 on GPIO-1.

In general this is working, but I can max use 4 of my 6 sensors.

Also I see ESPEasy mixing up sensor data after saving.

I would like to request to increase max sensors from 4 to 6 or more.

Sensor data will be send using MQTT to HA.

[image]

[image]

thx a lot

TD-er, 2025-09-29:
FYI...
There is a difference in reading the sensors in the same task and in separate task(s).

All 1wire sensors in the same task will start the measurement at "the same time" as in the "start measurement" command is given to all sensors in the same task and 750 msec later (assuming max. resolution is set) all sensors of the same task will be read.

This means all 1wire sensors in the same task will have their measurement done at (nearly) the same time. There is only a slight shift of about 20-ish msec in the moment the measurement is taken.

This can be important to know if you want to compare readings of related values.
For example 2 sensors on the same pipe to get some indication of flow speed or whatever you like to measure.

TungstenE2, 2025-09-29:
ah, so you can have several tasks with the same GPIO? This was unclear to me. Will check.

TungstenE2, 2025-10-01:
I powered them on 5v from the ESP32 directly with resistor, as recommended.

But seems to be mqtt issue now.

Sensor data at ESPeasy is updated, but HA stopped receiving yesterday afternoon and not picking up again.

[image]

[image]

[image]

TungstenE2, 2025-10-01:
just one observation I just made. ESP32C6 has internal and external antenna. I just attached external antenna and wifi signal was improved. will check further if this was the issue

TungstenE2, 2025-10-01:
will try beta, backup done

here is sysinfo before update.

[image]

TungstenE2, 2025-10-01:
no soldering. I think on ESP32C6 this can be done by config. Not sure how this is currently set in ESPeasy.

I only see better wifi connectivity using external antenna.

#####

The ESP32-C6 supports switching between an internal PCB antenna and an external antenna using onboard RF switches controlled by specific GPIO pins, usually via software configuration or firmware settings. For many boards (such as Xiao ESP32C6), the default is the internal antenna, but the external antenna can be activated by setting relevant GPIOs (commonly GPIO14 HIGH and GPIO3 LOW) to toggle the RF switch.[1][2][3][4][5]

### Typical GPIO Control for Switching
- To enable the external antenna, set GPIO3 to LOW and GPIO14 to HIGH in your firmware:
  ```
  pinMode(3, OUTPUT);     // RF switch power on
  digitalWrite(3, LOW);
  pinMode(14, OUTPUT);    // select external antenna
  digitalWrite(14, HIGH);
  ```
- To revert to the internal antenna, set GPIO14 to INPUT and GPIO3 to INPUT:
  ```
  pinMode(14, INPUT);     // select on-board antenna
  pinMode(3, INPUT);      // RF switch power off
  ```
This will trigger the onboard RF switch to use the external antenna port (typically IPEX/U.FL connector) if one is present.[4][5][1]

### Firmware/Platform Support
- For ESP-IDF: Multi-antenna functionality can be managed through the `esp_phy_set_ant_gpio()` and `esp_phy_set_ant()` APIs, allowing you to map antenna selection signals to specific GPIOs and toggle antennas in software.[2]
- ESPHome: Activation is possible by configuring outputs for the relevant GPIOs—most boards use GPIO3 and GPIO14.[5]
- Arduino: Recent core updates have improved software support for handling the switch (as of v3.0.4 for Xiao ESP32C6), so using the correct initialization and control code will ensure the switch triggers the correct antenna.[4]

### Hardware Details
- On boards with both antennas, an RF switch (such as FM8625H) is used to route antenna signals.[1]
- Some boards require a physical resistor adjustment or soldering to select the external antenna instead of software control, so check your board's schematic or documentation.[6]

### Practical Tips
- Only one antenna can be active at a time—external and internal cannot operate in parallel.[6]
- Board-specific instructions (schematics, API, GPIO mapping) are essential, as some may differ in hardware wiring.

If using a preconfigured board (like Xiao ESP32C6), always consult the board’s documentation for the correct GPIO mapping and switching method before modifying firmware or hardware.[2][1][4]

[1](https://forum.seeedstudio.com/t/xiao-esp32c6-switching-between-builtin-and-external-antenna/276374)
[2](https://docs.[espressif](https://docs.espressif.com/projects/esp-idf/en/stable/esp32c6/api-guides/phy.html).com/projects/esp-idf/en/stable/esp32c6/api-guides/phy.html)
[3](https://wiki.seeedstudio.com/xiao_esp32c6_getting_started/)
[4](https://www.sigmdel.ca/michel/ha/xiao/xiao_esp32c6_intro_en.html)
[5](https://community.[home-assistant](https://community.home-assistant.io/t/enable-external-wifi-antenna-on-xiao-esp32c6/910303).io/t/enable-external-wifi-antenna-on-xiao-esp32c6/910303)
[6](https://randomnerdtutorials.com/esp32-cam-connect-external-antenna/)
[7](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/user_guide.html)
[8](https://community.home-assistant.io/t/how-to-add-an-external-antenna-to-an-esp-board/131601)
[9](https://www.espressif.com/sites/default/files/documentation/esp32-c6_datasheet_en.pdf)
[10](https://www.youtube.com/watch?v=o4mtrueU6eM)
[11](https://www.mouser.com/pdfDocs/esp32-c6_hardware_design_guidelines_en.pdf)
[12](https://github.com/espressif/esp-csi/issues/184)
[13](https://github.com/espressif/esp-matter/issues/892)
[14](https://www.reddit.com/r/Esphome/comments/1lj3xb8/connecting_esp32_c6_to_wifi/)
[15](https://www.reddit.com/r/esp32/comments/1i007vz/options_for_switching_between_firmware_builds/)
[16](https://docs.platformio.org/en/latest/boards/espressif32/esp32-c6-devkitc-1.html)